openwrt/packages #11990: `travelmate:` appears to add erroneous domain to /etc/config/dhcp

karlic, 2020-04-28:
luci-app-travelmate - git-20.115.52331-39a8290-1
travelmate - 1.5.4-2

@dibdot, when trying to connect to BTWifi-with-FON, travelmate adds:
`list rebind_domain '/home'`
to /etc/config/dhcp.
It used to add 
`list rebind_domain 'www.btwifi.com'`

With '/home', option rebind_protection has to be '0' for travelmate to work, which doesn't seem right.

dibdot, 2020-04-28:
Please post the output of this command (if you sit behind the captive portal):
```
uclient-fetch --timeout=10 "http://captive.apple.com" -O /dev/null 2>&1 | tail -n 1
```

Thanks!

dibdot, 2020-04-28:
Edit: please with and without the tail command - thanks again!

karlic, 2020-04-28:
This is without piping onto tail (there's only 3 lines)
I logged out of the ISP and then:
```
root@OpenWrt:~# service travelmate stop
root@OpenWrt:~# uclient-fetch --timeout=10 "http://captive.apple.com" -O /dev/null 2>&1
Downloading 'http://captive.apple.com'
Connecting to 17.253.35.202:80
Failed to redirect to /home?CPURL=http://home.bt.com on www.btwifi.com
root@OpenWrt:~# 
```

closed 2020-04-29 by dibdot

dibdot, 2020-04-29:
Thanks, should work now!

karlic, 2020-04-29:
Confirmed it's now fixed. :-)

Thank you for the quick response.

commit 1aa4518d (2020-04-29): travelmate: fix captive portal detection

* made captive portal detection more bullet proof, fixes #11990
commit e92ac8ab (2020-04-29): travelmate: fix captive portal detection

* made captive portal detection more bullet proof, fixes #11990

(cherry picked from commit 1aa4518dfc57e7cee0bac0b8dadaf7718ab4a6e9)