openwrt/packages #12008: pulseaudio: Build failure for arm targets without neon support

jefferyto, 2020-04-29:
Maintainer: @tripolar (also ping @neheb and @dhewg)
Environment: Many arm targets (e.g. arm_cortex-a9, bcm53xx), OpenWrt master

Description:
I believe these build failures began after the conversion to meson in #11877.

Faillogs:
* [arm_arm926ej-s](https://downloads.openwrt.org/snapshots/faillogs/arm_arm926ej-s/packages/pulseaudio/avahi/compile.txt)
* [arm_cortex-a7](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7/packages/pulseaudio/avahi/compile.txt)
* [arm_cortex-a9](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9/packages/pulseaudio/avahi/compile.txt)
* [arm_fa526](https://downloads.openwrt.org/snapshots/faillogs/arm_fa526/packages/pulseaudio/avahi/compile.txt)
* [arm_mpcore](https://downloads.openwrt.org/snapshots/faillogs/arm_mpcore/packages/pulseaudio/avahi/compile.txt)
* [arm_xscale](https://downloads.openwrt.org/snapshots/faillogs/arm_xscale/packages/pulseaudio/avahi/compile.txt)

From [arm_cortex-a9](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9/packages/pulseaudio/avahi/compile.txt):

```
...
Compiler supports mmx: NO
Compiler supports sse: NO
Compiler supports neon: YES
...
[94/381] Compiling C object 'src/pulsecore/e9dd7ca@@libpulsecore_simd_neon@sta/remap_neon.c.o'
FAILED: src/pulsecore/e9dd7ca@@libpulsecore_simd_neon@sta/remap_neon.c.o 
arm-openwrt-linux-muslgnueabi-gcc -Isrc/pulsecore/e9dd7ca@@libpulsecore_simd_neon@sta -I. -I.. -Isrc -I../src -I/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/usr/include -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include/fortify -I/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include -I/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libiconv-stub/include -I/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libintl-stub/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=gnu11 -Os -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -ffile-prefix-map=/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-avahi/pulseaudio-13.0=pulseaudio-13.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIC -DHAVE_CONFIG_H -D_GNU_SOURCE -mfpu=neon -MD -MQ 'src/pulsecore/e9dd7ca@@libpulsecore_simd_neon@sta/remap_neon.c.o' -MF 'src/pulsecore/e9dd7ca@@libpulsecore_simd_neon@sta/remap_neon.c.o.d' -o 'src/pulsecore/e9dd7ca@@libpulsecore_simd_neon@sta/remap_neon.c.o' -c ../src/pulsecore/remap_neon.c
In file included from ../src/pulsecore/remap_neon.c:29:
/builder/shared-workdir/build/sdk/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/include/arm_neon.h:31:2: error: #error "NEON intrinsics not available with the soft-float ABI.  Please use -mfloat-abi=softfp or -mfloat-abi=hard"
 #error "NEON intrinsics not available with the soft-float ABI.  Please use -mfloat-abi=softfp or -mfloat-abi=hard"
  ^~~~~
...
```

PulseAudio [uses the Meson simd module](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/blob/v13.0/src/pulsecore/meson.build#L173-183) to detect support for mmx/sse/neon, then compile the right files.

I tried building locally, here is the part of meson-log.txt where it does this detection:

```
../../../../../../../../../media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/src/pulsecore/meson.build:173: WARNING: Module unstable-simd has no backwards or forwards compatibility and might not exist in future releases.
Running compile:
Working directory:  /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/openwrt-build/meson-private/tmpeber38jt
Command line:  arm-openwrt-linux-muslgnueabi-gcc -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include/fortify -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libiconv-stub/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libintl-stub/include /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/openwrt-build/meson-private/tmpeber38jt/testfile.c -o /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/openwrt-build/meson-private/tmpeber38jt/output.obj -pipe -c -Os -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -ffile-prefix-map=/media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0=pulseaudio-13.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -D_FILE_OFFSET_BITS=64 -O0 -mmmx 

Code:
 int i;

Compiler stdout:
 
Compiler stderr:
 arm-openwrt-linux-muslgnueabi-gcc: error: unrecognized command line option '-mmmx'

Compiler supports mmx: NO
Running compile:
Working directory:  /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/openwrt-build/meson-private/tmpfd6gqzox
Command line:  arm-openwrt-linux-muslgnueabi-gcc -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include/fortify -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libiconv-stub/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libintl-stub/include /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/openwrt-build/meson-private/tmpfd6gqzox/testfile.c -o /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/openwrt-build/meson-private/tmpfd6gqzox/output.obj -pipe -c -Os -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -ffile-prefix-map=/media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0=pulseaudio-13.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -D_FILE_OFFSET_BITS=64 -O0 -msse 

Code:
 int i;

Compiler stdout:
 
Compiler stderr:
 arm-openwrt-linux-muslgnueabi-gcc: error: unrecognized command line option '-msse'; did you mean '-fdse'?

Compiler supports sse: NO
Running compile:
Working directory:  /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulseaudio-noavahi/pulseaudio-13.0/openwrt-build/meson-private/tmp70pv9zra
Command line:  arm-openwrt-linux-muslgnueabi-gcc -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/usr/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include/fortify -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libiconv-stub/include -I/media/jeff/Jekyll/Files/openwrt/testing/b/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib/libintl-stub/include /media/jeff/Jekyll/Files/openwrt/testing/b/build_dir/target-arm_cortex-a9_musl_eabi/pulsea [... cut ...] ITS=64 -O0 -mfpu=neon 

Code:
 int i;

Compiler stdout:
 
Compiler stderr:
 
Compiler supports neon: YES
```

Basically it tries passing `-mfpu=neon` to gcc to see if gcc will complain (it doesn't).

This has been filed with Meson (mesonbuild/meson#5973); that issue has been closed in favour of a larger tracking bug (mesonbuild/meson#6361) but it doesn't look like it will be finished anytime soon.

I don't mind helping but I'm not familiar enough with Meson to know how this should be fixed :joy:

neheb, 2020-04-29:
I don't either. I submitted an issue with pulseaudio but no answer.

jefferyto, 2020-04-29:
For reference: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/872

neheb, 2020-05-08:
Fix was merged: https://github.com/openwrt/packages/commit/91e80e5442484e5bbb8515e686631c7e937f3a10 . Closing.

closed 2020-05-08 by neheb

i-garrison, 2021-06-19:
Workaround for Meson `unstable-simd` neon detection added in pulseaudio-14.99.2
https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/574

neheb, 2021-06-19:
Great.