openwrt/packages #12033: samba4-server: files shown as directories when macos=1

jacklovell, 2020-05-01:
Maintainer: @Andy2244 
Environment: BT Home Hub 5, MIPS, OpenWrt 19.07.2, r10947-65030d81f3, samba4-server 4.11.6-1.

Description:
I have an external hard drive with 3 ext4 partitions shared with Samba 4. Some mp4 video files (DVD rips) on one of the partitions I'm sharing appear as directories on connected clients. This occurs with Windows 10, MacOS 10.15 and Linux Mint 19 clients.

My config is as follows:
```
config samba
	option workgroup 'WORKGROUP'
	option description 'Samba on OpenWrt'
	option charset 'UTF-8'
	option macos '1'

config sambashare
	option path '/mnt/sda1'
	option create_mask '0666'
	option dir_mask '0777'
	option read_only 'no'
	option users 'lovells'
	option name 'Lovells'
	option guest_ok 'no'

config sambashare
	option path '/mnt/sda2'
	option name 'TimeMachine'
	option timemachine '1'
	option create_mask '0666'
	option dir_mask '0777'
	option read_only 'no'
	option guest_ok 'no'
	option users 'lovells'

config sambashare
	option path '/mnt/sda3'
	option name 'Win10Backup'
	option guest_ok 'no'
	option create_mask '0666'
	option dir_mask '0777'
	option read_only 'no'
	option users 'lovells'
```
De-selecting the "Enable macOS compatible shares" option in LUCI (i.e. removing `option macos 1`) fixes the problem: the files are correctly displayed as MP4 files and not directories. Unfortunately, this prevents me using the second share for my Time Machine backups.

Andy2244, 2020-05-01:
mhh need to see if i can reproduce this and if its still a issue in my upcoming 4.12.2 version.

In the meantime, if you feel like it you can try&error some of the vfs_fruit options in the `/etc/init.d/samba4` MacOs section. https://www.samba.org/samba/docs/current/man-html/vfs_fruit.8.html

I used one of the basic "recommend" defaults, but there are quite some variants out there.

jacklovell, 2020-05-02:
I think I've managed to find a pattern which may help you to reproduce the problem. All the MP4 files which were incorrectly shown as directories had the following xattr (examined using `getfattr -d` on the openwrt router itself): 
```
user.DOSATTRIB=""
```
Note that value is two quotation marks.

All the other MP4 files which were correctly shown as files rather than directories had a different value for this xattr, e.g.
```
user.DOSATTRIB=0sAAAEAAQAAABRAAAAIAAAANKCao7OH9YBADtzV8sf1gE=
```

And most interestingly, if I used `setfattr -n user.DOSATTRIB -v "" <file>` on a file which previously was fine, it would then be displayed as a directory. Removing this xattr using `setfattr -x user.DOSATTRIB`, or setting it to a value without the quotation marks (even a silly value like `something`) would return to the correct display of the file.

If you can reproduce this behaviour by doing `setfattr -n user.DOSATTRIB -v "" <file>` then this would point to the cause of the problem. I've only tested it on a Mac client though: I'll try with Windows and Linux clients next and report back if I see any different behaviour.

There seems to be a pretty simple workaround: remove the `user.DOSATTRIB` xattr from any problem files. But I don't know enough about vfs_streams_xattr, samba or extended attributes to know whether this has any other unwanted side-effects. And it's also not clear how this dodgy value of the xattr got created in the first place: is this a general samba/vfs_fruit bug or does it only show up in an openwrt build?

Also, disabling vfs_fruit prevents the Mac from seeing extended attributes, which reinforces my theory that it's this dodgy xattr which is causing the problem.

jacklovell, 2020-05-03:
I can confirm the same behaviour with Windows and Linux guests: when `user.DOSATTRIB = ""` the files are shown as directories. Also, in the Window file properties dialogue, the "Archive" attribute check box is ticked. Upon removing the `user.DOSATTRIB` xattr the Archive option is un-checked.

Andy2244, 2020-05-03:
mhh ok, so somehow this invalid `""` attribute gets stored via vfs_streams_xattr. You have any idea when/how this happens, since it seems only a few files got this invalid value for you? You remember what might be different about those?

The general fix seems to-be to reset the attributes and than disable them in the smb.conf.
see: https://www.ixsystems.com/community/threads/smb-share-has-folders-that-are-invisible-fixed.56454/

The culprit seems to-be the read-only attribute on some files, which than gets somehow interpreted as folder.

So just reset them and than add this to the template config:
```
store dos attributes = no
ea support = no
```

jacklovell, 2020-05-05:
I'm afraid I can't pin down exactly what causes this invalid `""` value. I've tried copying the same file over again, and it had a valid base64(?) `user.DOSATTRIB` string according to `getfattr`. It's possible the original transfers were interrupted by the network going down or some other issue, but it's too long ago for me to remember exactly what happened.

I've found I can reset the DOS attributes to valid values from a Windows client, by going into the file (or directory) properties and changing the Archive property. Interestingly, the files and folders with the invalid attribute value have a small cross in the bottom left corner of their icon. Setting to true the Archive property on these folders removes this cross and sets `user.DOSATTRIB` to a valid base64 string.

I'm going to leave it like this for a while, with `store dos attributes = yes` and `ea support = yes`, and see whether the problem returns. If it turns out to just be a one-off then it shouldn't be necessary to change these options and lose extended/DOS attribute support, now that we know of a simple fix.

Andy2244, 2020-05-05:
ok just report back

Neustradamus, 2021-04-13:
@jacklovell: It has been solved?

jacklovell, 2021-04-18:
I haven't noticed the problem reapparing. So let's consider it soved.

closed 2021-04-18 by jacklovell