openwrt/packages #12048: seafile-server: incorrect PKG_BUILD_DEPENDS

dl12345, 2020-05-02:
[commit 13d843](13d843fec1cd3281cc469fc74de302c1f793b723) is incorrect. PKG_BUILD_DEPENDS should contain directory names, not package names.

libevent2-openssl and libevent2-pthreads are both contained within the Makefile found in the libevent2 package directory, so just specifying libevent2 is good enough. There is no libevent2-openssl and libevent2-pthreads directory.

This commit results in the following error when processing dependencies....

```
[ian@centos openwrt]$ make menuconfig
WARNING: Makefile 'package/feeds/packages/seafile-server/Makefile' has a build dependency on 'libevent2-openssl', which does not exist
WARNING: Makefile 'package/feeds/packages/seafile-server/Makefile' has a build dependency on 'libevent2-pthreads', which does not exist
```

the PKG_BUILD_DEPENDS line should read

```
PKG_BUILD_DEPENDS:=vala/host libevent2 libevhtp
```

The actual DEPENDS:= line should contain the package names, which is correct.

neheb, 2020-05-08:
ping @jefferyto

jefferyto, 2020-05-08:
This was fixed by #12007.

closed 2020-05-08 by neheb