openwrt/packages #12074: wsdd2: high CPU usage

wongsyrone, 2020-05-04:
Maintainer: @Andy2244 
Environment: x86_64, Openwrt trunk, packages truck

Description:
wsdd2 always consumes high CPU usage, load decreases to 0.00 when terminates wsdd2.
```
Load average: 1.23 0.36 0.13 3/109 3919
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 3350     1 root     R      740   0%  25% /usr/bin/wsdd2 -i br-lan -N OpenWrt -G WORKGROUP -b
 3200     1 root     S    29196   1%   0% /usr/sbin/smbd -F
 3366  3200 root     S    27628   1%   0% {smbd-notifyd} /usr/sbin/smbd -F
 3367  3200 root     S    27620   1%   0% {cleanupd} /usr/sbin/smbd -F
```
After:
```
Load average: 0.02 0.18 0.10 2/119 4529
```

samba4 starts automatically without manual configuration.

Andy2244, 2020-05-05:
mhh never happened to me, need to test trunk in the upcoming days.

wongsyrone, 2020-05-05:
I checked the running options passed in, I found `-b` requires several parameters

```
Usage: wsdd2 [opts]
       -4  IPv4 only
       -6  IPv6 only
       -L  LLMNR debug mode (incremental level)
       -W  WSDD debug mode (incremental level)
       -d  go daemon
       -h  This message
       -l  LLMNR only
       -t  TCP only
       -u  UDP only
       -w  WSDD only
       -i "interface"  Listening interface (optional)
       -N  set NetbiosName manually
       -G  set Workgroup manually
       -b "key1:val1,key2:val2,..."  Boot parameters
           vendor:
           model:
           serial:
           sku:
           vendorurl:
           modelurl:
           presentationurl:
```
I execute `/usr/bin/wsdd2 -i br-lan -N OpenWrt -G WORKGROUP` without `-b` from the terminal, it looks everything is ok.
```
 3083  3072 root     S      744   0%   0% /usr/bin/wsdd2 -i br-lan -N OpenWrt -G WORKGROUP
```

wongsyrone, 2020-05-05:
To reproduce, try `/usr/bin/wsdd2 -i br-lan -N OpenWrt -G WORKGROUP -b ""`

wongsyrone, 2020-05-05:
Sent the PR [fixes infinite loop](https://github.com/Andy2244/wsdd2/pull/7)

BTW, the init script should check if `$BI_PARM` is an empty string, do `procd_append_param` only if `$BI_PARM` is valid.

wongsyrone, 2020-05-05:
init script fix sent: https://github.com/openwrt/packages/pull/12092

wongsyrone, 2020-05-06:
@Andy2244 Can you update the package and use the latest commit from https://github.com/Andy2244/wsdd2

Andy2244, 2020-05-07:
Sure, will update my packages on the weekend.

neheb, 2020-05-08:
Fix was merged. Close?

wongsyrone, 2020-05-08:
> Sure, will update my packages on the weekend.

Waiting for the package upgrade.

wongsyrone, 2020-05-11:
@Andy2244 Any update?

closed 2020-05-15 by wongsyrone