openwrt/packages #12082: symlink python to python3

lucize, 2020-05-05:
@commodo Hi, what is the implication of having python symlinked to python3.8, there are some packages  that are searching first for python and they find it in host/bin instead of using hostpkg/bin so any modules needed are searched in host/lib and not found

commodo, 2020-05-05:
python being symlinked to python3 should only be important on the target;
i'm a bit puzzled as to why it could be an issue on the host-side/build

lucize, 2020-05-05:
so for mesa I need python3/mako/host
python3/mako will install in ~/hostpkg/lib/python3.8/
mesa and xcb-proto are searching for python, they find it in ~/host/bin (python linked to host python3.7)
they are happy that there is a python3.7 but when searching for aditional modules (mako) in this case they search for lib of python3.7 /usr/lib not ~/hostpkg/lib/python3.8

hope you understand the issue

commodo, 2020-05-05:
maybe try `HOST_PYTHON3_PACKAGE_BUILD_DEPENDS:="mako"` before 
```
include ../pypi.mk
include $(INCLUDE_DIR)/package.mk
include ../python3-package.mk
```

installing python3 packages on the host build is always messy;
this mechanism was introduced to help with host issues;
it should also guarantee that the proper Python3 is being used;

lucize, 2020-05-05:
I think that the configure.ac for the apps is outdated, mesa seems to have been patched it on newer sources, but the question still stands for maybe others, it hurts us to have ~/hostpkg/pyhon symlinked to ~hostpkg/python3.8, hostpkg being searched first

jefferyto, 2020-05-05:
The host install of the python3 package does not install a `python` symlink to `$(STAGING_DIR_HOSTPKG)/bin`. I don't have a code reference to link to here but if you look in your `$(STAGING_DIR_HOSTPKG)/bin` you will see a `python3` symlink pointing to `python3.8` but no `python` symlink.

(If you are thinking of #11793, that added a symlink for the [target package](https://github.com/openwrt/packages/blob/38235de9c3a6d0574f3293178617549822fbc738/lang/python/python3/Makefile#L286), not host install.)

Also, `$(STAGING_DIR)/host/bin` is ahead of `$(STAGING_DIR_HOSTPKG)/bin` in the path when compiling packages. You can see this in [`TARGET_PATH_PKG`](https://github.com/openwrt/openwrt/blob/0fb5962e62596e14897a7ea9066ca17564f47d2f/rules.mk#L225), which is [exported](https://github.com/openwrt/openwrt/blob/0fb5962e62596e14897a7ea9066ca17564f47d2f/include/package.mk#L172) as `PATH` during package build.

There is a `python` symlink in `$(STAGING_DIR)/host/bin`, that links to your system Python.

You have already described this in https://github.com/openwrt/packages/issues/12082#issuecomment-623967829: mesa is using system Python (`python` in `$(STAGING_DIR)/host/bin)`, that's why it is Python 3.7 and wants to use `/usr/lib`), but the mako dependency is installed by host Python (`python3` in `$(STAGING_DIR_HOSTPKG)/bin`) and is installed inside buildroot.

The solution is to tell mesa to use host Python; this is usually done by setting the `PYTHON` environment variable during configure and/or make (e.g. [samba4](https://github.com/openwrt/packages/blob/38235de9c3a6d0574f3293178617549822fbc738/net/samba4/Makefile#L248), [uwsgi](https://github.com/openwrt/packages/blob/master/net/uwsgi/Makefile#L103), [libgpiod](https://github.com/openwrt/packages/blob/38235de9c3a6d0574f3293178617549822fbc738/libs/libgpiod/Makefile#L37)).

lucize, 2020-05-05:
yes, but letting it to use the system python would imply that we have to install the python-dev on the system, this is not available on the build bots, as we don't have python2 anymore is not ok for us to have the link for python as it is now for python3 ?
the old configure.ac is searching first for python and then for python3 so like #11793 but for the host install

jefferyto, 2020-05-05:
The `python` symlink in `$(STAGING_DIR)/host/bin` is added by the [prereq-build.mk](https://github.com/openwrt/openwrt/blob/0fb5962e62596e14897a7ea9066ca17564f47d2f/include/prereq-build.mk#L136), not us. I presume it exists because scripts in the main repo use `python` to run their own Python scripts; it doesn't exist to support compiling packages.

As I said, the solution is to tell configure to use host Python, not system Python.

lucize, 2020-05-05:
is ok for system to be in $(STAGING_DIR)/host/bin , having the host install in $(STAGING_DIR_HOSTPKG)/bin will let autotools to find the correct version (the host python, not system)

jefferyto, 2020-05-05:
If host Python is ahead of system Python in the path, a broken host Python would affect main repo builds.

lucize, 2020-05-05:
ok, I understand

closed 2020-05-05 by lucize