openwrt/packages #12083: ddns-scripts_cloudflare.com-v4: Feature Request / Support for API Tokens

DaTurret, 2020-05-05:
ddns-scripts_cloudflare.com-v4 : 2.7.8-12

Maintainer: @lantis1008
Environment: IPQ806X | Netgear Nighthawk X4S R7800 | OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.057.55219-13dd17f
OpenWrt 19.07.1 r10911-c155900f66 / LuCI openwrt-19.07 branch git-20.048.32085-85a6f07

Description:
I was wondering if there are plans to support Cloudflares more locked down "API Tokens" instead of API Keys. This would only give Access to a certain Zone, very Useful if you have multiple Zones that are not relevant to DDNS. It could be as simple as interpreting the Key as a Token if the Username is empty or something like that, although there is certanly a more elegant solution.

Here is my proposal in some script blocks (untested):

https://github.com/openwrt/packages/blob/9c95fedca3d22b01c28dda7c72e70dfec9ca0f2e/net/ddns-scripts/files/update_cloudflare_com_v4.sh#L25
Replace with:
```sh
[ -z "$username" ] && write_log 4 "Username missing, assuming Password is an API Token."
```

https://github.com/openwrt/packages/blob/9c95fedca3d22b01c28dda7c72e70dfec9ca0f2e/net/ddns-scripts/files/update_cloudflare_com_v4.sh#L127-L130
Replace with:
```sh
# set headers
if [-z "$username"]; then
    __PRGBASE="$__PRGBASE --header 'X-Auth-Email: $username' "
    __PRGBASE="$__PRGBASE --header 'X-Auth-Key: $password' "
else
    __PRGBASE="$__PRGBASE --header 'Authorization: Bearer $password' "
fi
__PRGBASE="$__PRGBASE --header 'Content-Type: application/json' "
```

lantis1008, 2020-05-05:
I'm not the maintainer. Feel free to submit whatever patch you like. @neheb reviewed mine, they may have some opinion.

DaTurret, 2020-05-05:
I have done some more research and have found out that either 2 Tokens would be needed or the Token would not be scoped, which would defy Token support in the first place...
Therefore closing.

closed 2020-05-05 by DaTurret