openwrt/packages #12085: PHP file_get_contents(): SSL operation failed

VladimirKalachikhin, 2020-05-05:
OpenWrt 18.06
OpenWrt 19.07

file_get_contents() break if server need the TLS 1.2 

with message:
***
**Warning**: file_get_contents(): SSL operation failed with code 1. OpenSSL Error messages: error:1409442E:SSL routines:ssl3_read_bytes:tlsv1 alert protocol version in /mnt/SDcard/www/tileproxy/testSSL.php on line 8

**Warning**: file_get_contents(): Failed to enable crypto in /mnt/SDcard/www/tileproxy/testSSL.php on line 8

**Warning**: file_get_contents(https://b.tile.opentopomap.org/14/8451/5349.png): failed to open stream: operation failed in /mnt/SDcard/www/tileproxy/testSSL.php on line 8
***
The cURL work fine.
On Ubuntu this problem is not present.

The code implements this issue:
```
<?php
echo "Get by file_get_contents:<br><br>\n";
$ctx = stream_context_create([
    'ssl' => [
	'verify_peer' => FALSE,
	'verify_peer_name' => FALSE,
        'crypto_method' => STREAM_CRYPTO_METHOD_TLSv1_2_CLIENT
    ],
]);
$img = file_get_contents('https://b.tile.opentopomap.org/14/8451/5349.png', false, $ctx);
echo "http_response_header:<pre>\n"; print_r($http_response_header); echo "</pre><br><br>\n";

echo "Get by cURL:<br><br>\n";
$ch = curl_init();
curl_setopt($ch, CURLOPT_URL, 'https://b.tile.opentopomap.org/14/8451/5349.png');
curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);
$img = curl_exec($ch);
echo "cURL info:\n<pre>"; print_r(curl_getinfo($ch)); echo "</pre>\n";
curl_close($ch); 
?>
```

mhei, 2020-05-10:
Sorry, I cannot reproduce your issue. Maybe I misunderstood anything?
I did:

```
# opkg update
# opkg install php7-cli php7-mod-curl

# php-cli test.php
Get by file_get_contents:<br><br>

Warning: file_get_contents(): Unable to find the wrapper "https" - did you forget to enable it when you configured PHP? in /tmp/test.php on line 10

Warning: file_get_contents(https://b.tile.opentopomap.org/14/8451/5349.png): failed to open stream: No such file or directory in /tmp/test.php on line 10
http_response_header:<pre>
</pre><br><br>
Get by cURL:<br><br>
cURL info:
<pre>Array
(
    [url] => https://b.tile.opentopomap.org/14/8451/5349.png
    [content_type] => image/png
    [http_code] => 200
    [header_size] => 281
    [request_size] => 77
    [filetime] => -1
    [ssl_verify_result] => 0
    [redirect_count] => 0
    [total_time] => 0.80087
    [namelookup_time] => 0.037317
    [connect_time] => 0.08197
    [pretransfer_time] => 0.537438
    [size_upload] => 0
    [size_download] => 15697
    [speed_download] => 19621
    [speed_upload] => 0
    [download_content_length] => 15697
    [upload_content_length] => -1
    [starttransfer_time] => 0.757009
    [redirect_time] => 0
    [redirect_url] =>
    [primary_ip] => 131.188.76.144
    [certinfo] => Array
        (
        )

    [primary_port] => 443
    [local_ip] => 192.168.8.52
    [local_port] => 50716
    [http_version] => 2
    [protocol] => 2
    [ssl_verifyresult] => 0
    [scheme] => HTTPS
    [appconnect_time_us] => 536447
    [connect_time_us] => 81970
    [namelookup_time_us] => 37317
    [pretransfer_time_us] => 537438
    [redirect_time_us] => 0
    [starttransfer_time_us] => 757009
    [total_time_us] => 800870
)
</pre>

# opkg install php7-mod-openssl

# php-cli test.php
Get by file_get_contents:<br><br>
http_response_header:<pre>
Array
(
    [0] => HTTP/1.1 200 OK
    [1] => Date: Sun, 10 May 2020 14:08:04 GMT
    [2] => Server: Apache/2.4.29 (Ubuntu)
    [3] => ETag: "38a8c6a2248f48cf2d5129dcd8ddefb1"
    [4] => Content-Length: 15697
    [5] => Cache-Control: max-age=435227
    [6] => Expires: Fri, 15 May 2020 15:01:51 GMT
    [7] => Access-Control-Allow-Origin: *
    [8] => Connection: close
    [9] => Content-Type: image/png
)
</pre><br><br>
Get by cURL:<br><br>
cURL info:
<pre>Array
(
    [url] => https://b.tile.opentopomap.org/14/8451/5349.png
    [content_type] => image/png
    [http_code] => 200
    [header_size] => 281
    [request_size] => 77
    [filetime] => -1
    [ssl_verify_result] => 0
    [redirect_count] => 0
    [total_time] => 0.629435
    [namelookup_time] => 0.072032
    [connect_time] => 0.111172
    [pretransfer_time] => 0.550656
    [size_upload] => 0
    [size_download] => 15697
    [speed_download] => 24955
    [speed_upload] => 0
    [download_content_length] => 15697
    [upload_content_length] => -1
    [starttransfer_time] => 0.590173
    [redirect_time] => 0
    [redirect_url] =>
    [primary_ip] => 131.188.76.144
    [certinfo] => Array
        (
        )

    [primary_port] => 443
    [local_ip] => 192.168.8.52
    [local_port] => 50724
    [http_version] => 2
    [protocol] => 2
    [ssl_verifyresult] => 0
    [scheme] => HTTPS
    [appconnect_time_us] => 549785
    [connect_time_us] => 111172
    [namelookup_time_us] => 72032
    [pretransfer_time_us] => 550656
    [redirect_time_us] => 0
    [starttransfer_time_us] => 590173
    [total_time_us] => 629435
)
</pre>
#
```

VladimirKalachikhin, 2020-05-10:
I use custom firmware. May be I forgot somethings.
What should be in the system for mod-openssl to work well?

mhei, 2020-05-10:
The package should have the correct dependencies set, just install/select it with your custom firmware.

VladimirKalachikhin, 2020-05-10:
Of course I did. But I see this issue.

mhei, 2020-05-12:
On which platform do you test? I have only tested 19.07, you're facing the issue on both release version... I don't really have an idea at the moment...

VladimirKalachikhin, 2020-05-12:
My (diff) config for make:
[diffconfig.txt](https://github.com/openwrt/packages/files/4618449/diffconfig.txt)

I finding issue on OpenWrt 18.06 - based firmware, and build fresh.

mhei, 2020-05-16:
I failed to rebuild a 18.06 from scratch and I don't have a Ralink/Mediatek device at hand, but I installed 18.06 from downloads.openwrt.org on an ARM device:

```root@OpenWrt:/etc/config# opkg update
Downloading http://downloads.openwrt.org/releases/18.06.8/targets/mxs/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/releases/18.06.8/targets/mxs/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/telephony/Packages.sig
Signature check passed.
root@OpenWrt:/etc/config# opkg install php7-cli php7-mod-openssl php7-mod-curl
Installing php7-cli (7.2.28-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/php7-cli_7.2.28-1_arm_arm926ej-s.ipk
Installing libpcre (8.41-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/libpcre_8.41-2_arm_arm926ej-s.ipk
Installing zlib (1.2.11-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/base/zlib_1.2.11-2_arm_arm926ej-s.ipk
Installing libxml2 (2.9.9-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/libxml2_2.9.9-1_arm_arm926ej-s.ipk
Installing zoneinfo-core (2018e-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/zoneinfo-core_2018e-1_arm_arm926ej-s.ipk
Installing php7 (7.2.28-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/php7_7.2.28-1_arm_arm926ej-s.ipk
Installing libstdcpp (7.3.0-2) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/targets/mxs/generic/packages/libstdcpp_7.3.0-2_arm_arm926ej-s.ipk
Installing php7-mod-openssl (7.2.28-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/php7-mod-openssl_7.2.28-1_arm_arm926ej-s.ipk
Installing libopenssl (1.0.2u-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/base/libopenssl_1.0.2u-1_arm_arm926ej-s.ipk
Installing php7-mod-curl (7.2.28-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/packages/php7-mod-curl_7.2.28-1_arm_arm926ej-s.ipk
Installing libmbedtls (2.16.4-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/base/libmbedtls_2.16.4-1_arm_arm926ej-s.ipk
Installing ca-bundle (20190110-1) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/base/ca-bundle_20190110-1_all.ipk
Installing libcurl (7.60.0-4) to root...
Downloading http://downloads.openwrt.org/releases/18.06.8/packages/arm_arm926ej-s/base/libcurl_7.60.0-4_arm_arm926ej-s.ipk
Configuring libpcre.
Configuring zlib.
Configuring libxml2.
Configuring zoneinfo-core.
Configuring php7.
Configuring libmbedtls.
Configuring ca-bundle.
Configuring libcurl.
Configuring php7-mod-curl.
Configuring libopenssl.
Configuring php7-mod-openssl.
Configuring libstdcpp.
Configuring php7-cli.
root@OpenWrt:/etc/config# cd /tmp
root@OpenWrt:/tmp# vi test.php
root@OpenWrt:/tmp# php-cli test.php
[... 28 lines cut ...]
    [total_time] => 0.622784
    [namelookup_time] => 0.037069
    [connect_time] => 0.076032
    [pretransfer_time] => 0.534817
    [size_upload] => 0
    [size_download] => 15697
    [speed_download] => 25236
    [speed_upload] => 0
    [download_content_length] => 15697
    [upload_content_length] => -1
    [starttransfer_time] => 0.574269
    [redirect_time] => 0
    [redirect_url] =>
    [primary_ip] => 131.188.76.144
    [certinfo] => Array
        (
        )

    [primary_port] => 443
    [local_ip] => 192.168.8.52
    [local_port] => 33904
)
</pre>
root@OpenWrt:/tmp#
```

mhei, 2020-05-28:
Closing since I cannot reproduce. In case the problem still exists, please reopen and provide a detailed log/step-by-step instruction.

closed 2020-05-28 by mhei

VladimirKalachikhin, 2020-12-27:
Let's start over.

I create OpenWRT-x86 virtual machine as described in https://openwrt.org/docs/guide-user/virtualization/virtualbox-vm
Image to machine - https://downloads.openwrt.org/releases/19.07.5/targets/x86/64/openwrt-19.07.5-x86-64-combined-ext4.img.gz

On the freshly started machine:
```
root@OpenWrt:/# opkg install php7-cli php7-mod-curl
Installing php7-cli (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7-cli_7.2.34-1_x86_64.ipk
Installing libpcre (8.43-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/libpcre_8.43-1_x86_64.ipk
Installing libxml2 (2.9.9-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/libxml2_2.9.9-3_x86_64.ipk
Installing zoneinfo-core (2020c-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/zoneinfo-core_2020c-1_x86_64.ipk
Installing php7 (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7_7.2.34-1_x86_64.ipk
Installing libstdcpp6 (7.5.0-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/targets/x86/64/packages/libstdcpp6_7.5.0-2_x86_64.ipk
Installing php7-mod-curl (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7-mod-curl_7.2.34-1_x86_64.ipk
Installing ca-bundle (20200601-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/base/ca-bundle_20200601-1_all.ipk
Installing libcurl4 (7.66.0-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/base/libcurl4_7.66.0-3_x86_64.ipk
Configuring libpcre.
Configuring libxml2.
Configuring zoneinfo-core.
Configuring php7.
Configuring ca-bundle.
Configuring libcurl4.
Configuring php7-mod-curl.
Configuring libstdcpp6.
Configuring php7-cli.
root@OpenWrt:/www# php-cli testSSL.php 
Get by file_get_contents:<br><br>

Warning: file_get_contents(): Unable to find the wrapper "https" - did you forget to enable it when you configured PHP? in /www/testSSL.php on line 8

Warning: file_get_contents(https://b.tile.opentopomap.org/14/8451/5349.png): failed to open stream: No such file or directory in /www/testSSL.php on line 8
http_response_header:<pre>
</pre><br><br>
Get by cURL:<br><br>
http_response_info:
<pre>Array
(
    [url] => https://b.tile.opentopomap.org/14/8451/5349.png
    [content_type] => image/png
    [http_code] => 200
    [header_size] => 279
    [request_size] => 77
    [filetime] => -1
    [ssl_verify_result] => 0
    [redirect_count] => 0
    [total_time] => 0.595267
    [namelookup_time] => 0.138891
    [connect_time] => 0.20573
    [pretransfer_time] => 0.416388
    [size_upload] => 0
    [size_download] => 15625
    [speed_download] => 26260
    [speed_upload] => 0
    [download_content_length] => 15625
    [upload_content_length] => -1
    [starttransfer_time] => 0.527365
    [redirect_time] => 0
    [redirect_url] => 
    [primary_ip] => 131.188.76.144
    [certinfo] => Array
        (
        )

    [primary_port] => 443
    [local_ip] => 192.168.10.102
    [local_port] => 60018
)
</pre>

root@OpenWrt:/www# opkg install php7-mod-openssl
Installing php7-mod-openssl (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7-mod-openssl_7.2.34-1_x86_64.ipk
Installing libopenssl1.1 (1.1.1i-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/base/libopenssl1.1_1.1.1i-1_x86_64.ipk
Configuring libopenssl1.1.
Configuring php7-mod-openssl.
root@OpenWrt:/www# php-cli testSSL.php 
Get by file_get_contents:<br><br>

Warning: file_get_contents(): SSL operation failed with code 1. OpenSSL Error messages:
error:1409442E:SSL routines:ssl3_read_bytes:tlsv1 alert protocol version in /www/testSSL.php on line 8

Warning: file_get_contents(): Failed to enable crypto in /www/testSSL.php on line 8

Warning: file_get_contents(https://b.tile.opentopomap.org/14/8451/5349.png): failed to open stream: operation failed in /www/testSSL.php on line 8
http_response_header:<pre>
</pre><br><br>
Get by cURL:<br><br>
http_response_info:
<pre>Array
(
    [url] => https://b.tile.opentopomap.org/14/8451/5349.png
    [content_type] => image/png
    [http_code] => 200
    [header_size] => 278
    [request_size] => 77
    [filetime] => -1
    [ssl_verify_result] => 0
    [redirect_count] => 0
    [total_time] => 0.39876
    [namelookup_time] => 0.000385
    [connect_time] => 0.070127
    [pretransfer_time] => 0.252536
    [size_upload] => 0
    [size_download] => 15625
    [speed_download] => 39258
    [speed_upload] => 0
    [download_content_length] => 15625
    [upload_content_length] => -1
    [starttransfer_time] => 0.324439
    [redirect_time] => 0
    [redirect_url] => 
    [primary_ip] => 131.188.76.144
    [certinfo] => Array
        (
        )

    [primary_port] => 443
    [local_ip] => 192.168.10.102
    [local_port] => 60026
)
</pre>

```
No other packages were installed.