openwrt/packages #12089: Best practices regarding use of chown and chmod

micmac1, 2020-05-05:
Hi all,

A few days ago somebody from Suse posted to oss-security about making sure that pre/post install scripts in packages are sane ([link](https://www.openwall.com/lists/oss-security/2020/04/30/1)). Later somebody involved in Gentoo replied and added more details.

TL;DR: Lax use of chmod/chown has serious security implications. A local user can setup symbolic links and hard links in a directory where he/she has write access and if root calls chown/chmod (either manually, in init scripts or in pre/post install scripts) these may follow the links and alter the permissions of the target, providing access to it to the local user, resulting in local privilege escalation.

To summarize the suggestions that can be found when following the links in the mail threat:

- don't use chown/chmod if you can help it
- use "su -" to switch to the user before running commands altering permissions/ownership where possible
- use special tools like "checkpath" and "fowners" when possible

This summary does not do justice to what the Gentoo person wrote on his website/blog, admittedly.

The two tools mentioned are Gentoo and OpenRC specific and we don't have them in OpenWrt as far as I know. I think we have no special tools at hand, really.

I know OpenWrt is not your typical multi-user OS. But we do have daemons running as non-root and need to adapt some permissions. So making sensible choices is a good thing.

My own list so far is this:

1. **Never** change permissions/ownership on live filesystem on files that the user may have edited.
2. **Never** change permissions/ownership of any file if you can avoid it.
3. To limit access to a certain user, a directory may be created and ownership of this directory handed to the user. But this shall only be done when the directory does not exist yet. **No** changes to existing directories.
4. Don't use "chmod" on a directory you created. Instead call "mkdir -m MODE ...".
5. The macro "$(INSTALL_CONF)" should not be used in case the files need to be readable by the daemon when it runs non-root. Instead use "$(INSTALL_DATA)" in this case. I take it as a given that in the files that the packages install there are no secrets that need hiding. If users puts something of value into these files they can change the permissions afterward. Note: this does not affect files installed to /etc/config, those can be installed with "$(INSTALL_CONF)".
6. If for convenience you want the init script to create directories that the user has the ability to configure (example: /etc/config/freeswitch sets db directory to '/tmp/freeswitch/db' but users can change it to anything else) , use "su -s /bin/sh -c "mkdir -m <mode> ..." to create these directories. Add a dependency like "+!BUSYBOX_DEFAULT_SU:shadow-su" to your package for "su" to be available. The reasoning here is that when the users can specify anything they want this could mean all kinds of strange directory paths and treachery. The "mkdir" may now fail, as we're not calling it as root anymore, but better to err on the side of caution.
7. Same applies when you want to run certain commands for the user (maybe using openssh to create a key or something). Use "su" to do that as the user, don't run it as root.

Regarding 3. What if the package installs a SUID-root binary into a directory and you want that directory to be only accessible by root and the daemon user, so say 750 root:daemon? You could probably use Package/daemon/install to at least set the dir to 750. But root:daemon can only be achieved on the device. And you can't use preinstall phase to create the directory (in case it doesn't exist already) the way you want it, because at this stage the user might not have been created. Meh :/

I think there's also some differentiation to be made regarding directories that are under root's control and others that are under a user's control. I'm just not sure how to phrase that yet.

I would be happy for your input/ideas/corrections.

Thanks!

Best regards,
Seb

(pinging @jslachta)

dangowrt, 2020-05-05:
I've run into this when porting GNUnet to OpenWrt and didn't find a better way than changing file ownership in the init script.
This is because OpenWrt currently only supports files owned by root:root in packages, as we simply set ownership for all files when generating the filesystem image.
Most other distributions use Debian's 'fakeroot' to emulate ownership of files installed and have them retained in opkg packages or squashfs.
This also recently became an issue for Thomas in his attempt to add SELinux support. He suggested adding fakeroot as build-tool, but only made use of it for setting SELinux labels.

https://patchwork.ozlabs.org/project/openwrt/patch/20191122095541.688125-4-thomas.petazzoni@bootlin.com/

The proper approach afaik is to use fakeroot to already have correct ownerships in packages and squashfs -- however, this obviously doesn't include files or folders needed at runtime in a tmpfs, for that we should probably add some safe wrappers to /etc/rc.common or /etc/functions.sh making sure that chown or chmod operations are carried out on 'stable ground'...

Just my 2 cents...

micmac1, 2020-05-06:
Hello @dangowrt 

Thank you for your feedback!

> I've run into this when porting GNUnet to OpenWrt and didn't find a better way than changing file ownership in the init script.

I looked at the init script and that indeed looks tricky.

> This is because OpenWrt currently only supports files owned by root:root in packages, as we simply set ownership for all files when generating the filesystem image. Most other distributions use [...]

That would be nice to have indeed. But at the moment I look at the way I do things with the tools at our disposal right now.

> The proper approach afaik is to use fakeroot to already have correct ownerships in packages and squashfs -- however, this obviously doesn't include files or folders needed at runtime in a tmpfs, for [...]

Yes, agree as well that if there could be some routines available for all that try to avoid at least the worst pitfalls that would be great. At the moment I have a hard time coming up with an idea of what these should look like, though :)

Again, thank you for your reply!

Kind regards,
Seb

brada4, 2020-05-07:
There is a need for something less racy (go-wt) than general use /tmp, something like /run/{package} , even /tmp can be fit under there and symlinked in default place.

micmac1, 2020-05-07:
> There is a need for something less racy (go-wt) than general use /tmp,
> something like /run/{package} , even /tmp can be fit under there and
> symlinked in default place.

Hi there,

Forgive my ignorance, can you explain what "go-wt" means?

Thanks!

champtar, 2020-05-07:
maybe the chmod `mode` ?

micmac1, 2020-05-07:
>maybe the chmod `mode` ?
 Yeah, makes sense. I see what you're getting at now.

brada4, 2020-05-08:
> There is a need for something less racy (go-wt) than general use /tmp, something like /run/{package} , even /tmp can be fit under there and symlinked in default place.
> Hi there, Forgive my ignorance, can you explain what "go-wt" means? Thanks!

Not world writable like /tmp

pprindeville, 2020-05-08:
> Most other distributions use Debian's 'fakeroot' to emulate ownership of files installed and have them retained in opkg packages or squashfs.

+1 on using `fakeroot`?

neheb, 2020-05-08:
The init scripts running as root do make this tricky. I just noticed the config file for gerbera is owned by root...

The INSTALL_CONF thing is quite annoying:

```
git grep /etc/config | grep INSTALL_DATA | wc -l
47
```

Not good.

micmac1, 2020-05-09:
> The init scripts running as root do make this tricky. I just noticed the config file for gerbera is owned by root...

I think for stuff in /etc/config INSTALL_CONF is fine, because those files get used by init scripts only.
 
> edit2: ubootenv in /etc/config is not owned by root on my system for some reason...

root here.

neheb, 2020-05-09:
Ah I misspoke. I meant 644 permissions.

micmac1, 2020-05-10:
Edited first post, added 8. and 9. to the list.

neheb, 2020-05-13:
I'll pin this.

As I mentioned above, several packages need to be fixed.

aparcar, 2020-08-03:
Hi team, based on Daniels comment I looked briefly into fakeroot and would suggest a patch that uses fakeroot both for image and package creation. The image created is pretty simple, however I'm struggling with the package creation.

```diff
diff --git a/include/image.mk b/include/image.mk
index e8c2cf7100..a1ee5e8d75 100644
--- a/include/image.mk
+++ b/include/image.mk
@@ -235,8 +235,8 @@ $(eval $(foreach S,$(JFFS2_BLOCKSIZE),$(call Image/mkfs/jffs2/template,$(S))))
 $(eval $(foreach S,$(NAND_BLOCKSIZE),$(call Image/mkfs/jffs2-nand/template,$(S))))
 
 define Image/mkfs/squashfs
-	$(STAGING_DIR_HOST)/bin/mksquashfs4 $(call mkfs_target_dir,$(1)) $@ \
-		-nopad -noappend -root-owned \
+	$(FAKEROOT) $(STAGING_DIR_HOST)/bin/mksquashfs4 $(call mkfs_target_dir,$(1)) $@ \
+		-nopad -noappend \
 		-comp $(SQUASHFSCOMP) $(SQUASHFSOPT) \
 		-processors 1
 endef
diff --git a/include/package-ipkg.mk b/include/package-ipkg.mk
index 622cbf3223..3bd2958c7d 100644
--- a/include/package-ipkg.mk
+++ b/include/package-ipkg.mk
@@ -11,7 +11,7 @@ endif
 
diff --git a/rules.mk b/rules.mk
index e734b4a582..5a3bda5412 100644
--- a/rules.mk
+++ b/rules.mk
@@ -272,6 +272,7 @@ ESED:=$(STAGING_DIR_HOST)/bin/sed -E -i -e
 CP:=cp -fpR
 LN:=ln -sf
 XARGS:=xargs -r
+FAKEROOT:=$(STAGING_DIR_HOST)/bin/fakeroot
 
 BASH:=bash
 TAR:=tar
```

The interesting parts for packages seem to be in [package-ipkg.mk](https://github.com/openwrt/openwrt/blob/master/include/package-ipkg.mk#L203) and [package-defaults.mk](https://github.com/openwrt/openwrt/blob/master/include/package-defaults.mk#L153).

It looks like package-ipkg.mk runs the install steps and does some `CONTROL` file stuff and [lastly](https://github.com/openwrt/openwrt/blob/master/include/package-ipkg.mk#L265) run the [`ipkg-build`](https://github.com/openwrt/openwrt/blob/master/scripts/ipkg-build) script to convert a folder into a package (tar.gz). So instead of passing the folder we could pass a script with install steps to `ipkg-build` and run it as fakeroot. 

To create the `CONTROL` file we either run the install-script once in `ipkg-build` and once as before, to have a folder (with wrong permissions) OR we integrate these steps into the install-script.

I'm trying to create such script, but Makefiles meta levels give me a hard time

```diff
+define PackageInstall
+  $(call Package/$(1)/install,$$(IDIR_$(1)))
+  $(if $(Package/$(1)/install-overlay),mkdir -p $(PACKAGE_DIR) $$(IDIR_$(1))/rootfs-overlay)
+  $(call Package/$(1)/install-overlay,$$(IDIR_$(1))/rootfs-overlay)
+endef
+
 ifneq ($(PKG_NAME),toolchain)
   define CheckDependencies
        @( \
@@ -200,9 +206,8 @@ $(_endef)
        @rm -rf $$(IDIR_$(1)); \
                $$(call remove_ipkg_files,$(1),$$(call opkg_package_files,$(call gen_ipkg_wildcard,$(1))))
        mkdir -p $(PACKAGE_DIR) $$(IDIR_$(1))/CONTROL $(PKG_INFO_DIR)
-       $(call Package/$(1)/install,$$(IDIR_$(1)))
-       $(if $(Package/$(1)/install-overlay),mkdir -p $(PACKAGE_DIR) $$(IDIR_$(1))/rootfs-overlay)
-       $(call Package/$(1)/install-overlay,$$(IDIR_$(1))/rootfs-overlay)
+       echo '$$(call PackageInstall,$(1))' > /tmp/package-$(1)
+       $(call PackageInstall,$(1))
        -find $$(IDIR_$(1)) -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| $(XARGS) rm -rf
        @( \
                find $$(IDIR_$(1)) -name lib\*.so\* -or -name \*.ko | awk -F/ '{ print $$$$NF }'; \
```

If that's fully implemented it allows any modification of user user/mode within the package creation.

aparcar, 2020-08-07:
After talking with @jow- we came up with a different implementation using `PKG_FILE_MOES` in combination with `fakeroot`. Please have a look at [my patch series](https://patchwork.ozlabs.org/project/openwrt/list/?series=194718).

aparcar, 2020-08-07:
> if ! [ -L "$dir" ] && [ -d "$dir" ] && [ -O "$dir" ] && [ -G "$dir" ]; then
    chown :newgroup "$dir"
fi

Is it actually necessary to check for symlink and directory? I'd say it's "save" when the file is owned by root.

micmac1, 2020-08-08:
> > if ! [ -L "$dir" ] && [ -d "$dir" ] && [ -O "$dir" ] && [ -G "$dir" ]; then
> > chown :newgroup "$dir"
> > fi
> 
> Is it actually necessary to check for symlink and directory? I'd say it's "save" when the file is owned by root.

Hi @aparcar 

It's just that I expect a directory, not a symlink. So I'm trying to make sure that this is not a symlink. As you know if you have symlink to a dir and run "[ -d $symlink ]" it'll return true.

Thanks by the way for the patches! I hope to find some time to test them.

Kind regards,
Seb

aparcar, 2020-08-08:
> It's just that I expect a directory, not a symlink. So I'm trying to make sure that this is not a symlink. As you know if you have symlink to a dir and run "[ -d $symlink ]" it'll return true.

We follow the idea that whatever is root owned can be trusted. Therefore, if root decides to have a symlink rather than a folder, this should be accepted. Different when a non-root user does so, in this case chown should fail.

My point, a check with -O and -G should be enough.

micmac1, 2020-08-08:
I'd argue that if root makes changes to the layout (dir to symlink for instance) then postinstall script making more changes on top is probably not a good idea. So to me the symlink check still is fine.

Anyway, if fakeroot gets added we wouldn't have to worry about this anymore anyway.

micmac1, 2020-08-08:
Well, actually we'd still need to think about this. I'm thinking init scripts.

aparcar, 2020-08-08:
There are multiple ideas, using LSB scripts or unit files which allow to define a running user. @dangowrt can you share some ideas?

dangowrt, 2020-08-08:
On Sat, Aug 08, 2020 at 01:47:01PM -0700, Paul Spooren wrote:
> There are multiple ideas, using LSB scripts or unit files which allow to define a running user. @dangowrt can you share some ideas?

You can alreay define the running user using `procd_set_param user $user` in procd-based initscripts.

EricLuehrsen, 2020-08-21:
@dangowrt using procd user parameter works for some minor packages. However external service daemons need to start off as "root" to bind to privileged resources such as well known ports and later drop to a lesser user once bound.

micmac1, 2020-09-08:
I finally got around to try fakeroot. It looks neat, but I ran into a limitation. When setting PKG_FILE_MODES it will call the ipkg build script with '-m "$(PKG_FILE_MODES)"' on any package install. But if the directory doesn't have the file contained in PKG_FILE_MODES it'll fail.

Take for instance mariadb. I add

PKG_FILE_MODES:=/usr/lib/mariadb/plugin/auth_pam_tool_dir:root:376:0750

but only need this for a single sub package ("auth-pam"). But whatever comes before fails when there is no such file:

```
home/sk/tmp/openwrt/staging_dir/host/bin/fakeroot -l /home/sk/tmp/openwrt/staging_dir/host/lib/libfakeroot.so -f /home/sk/tmp/openwrt/staging_dir/host/bin/faked /home/sk/tmp/openwrt/scripts/ipkg-build -m "/usr/lib/mariadb/plugin/auth_pam_tool_dir:root:376:0750" /home/sk/tmp/openwrt/build_dir/target-mips_24kc_musl/mariadb-10.4.13/ipkg-mips_24kc/mariadb-server-plugin-disks /home/sk/tmp/openwrt/bin/packages/mips_24kc/packages
+chown: cannot access '/home/sk/tmp/openwrt/build_dir/target-mips_24kc_musl/mariadb-10.4.13/ipkg-mips_24kc/mariadb-server-plugin-disks//usr/lib/mariadb/plugin/auth_pam_tool_dir': No such file or directory
```

Not complaining, just saying :)

aparcar, 2020-09-08:
That's a good finding, thanks for checking! I initially had the plan to run `install` via `fakeroot` so each would be handled separately, however it was requested *to have all mode settings in a single variable*. I see this as a limitation and could implement another approach or allow the file mode setting to skip in cases where the file is not existent, the easier solution.

micmac1, 2020-09-09:
Hi @aparcar 

I quickly looked at your patch. It's a bit different to what I had in mind. I was thinking on something along the lines of PKG_MAINTAINER vs. MAINTAINER. include/package-defaults.mk defines Package/Default, which does this:

`MAINTAINER:=$(PKG_MAINTAINER)`

So the default package define has a MAINTAINER set (can be "" of course). So you can set the user globally with PKG_MAINTAINER or set a MAINTAINER variable for each package.

The same can be easily implemented for the fakeroot setup:

```
diff --git a/include/package-defaults.mk b/include/package-defaults.mk
index 2fed72b1a4..2a04bc17e9 100644
--- a/include/package-defaults.mk
+++ b/include/package-defaults.mk
@@ -59,6 +59,7 @@ define Package/Default
   ALTERNATIVES:=
   LICENSE:=$(PKG_LICENSE)
   LICENSE_FILES:=$(PKG_LICENSE_FILES)
+  FILE_MODES:=$(PKG_FILE_MODES)
 endef

 Build/Patch:=$(Build/Patch/Default)
diff --git a/include/package-ipkg.mk b/include/package-ipkg.mk
index 62cda5b936..0bca8ae84d 100644
--- a/include/package-ipkg.mk
+++ b/include/package-ipkg.mk
@@ -260,7 +260,7 @@ $(_endef)
     endif

        $(INSTALL_DIR) $$(PDIR_$(1))
-       $(FAKEROOT) $(SCRIPT_DIR)/ipkg-build -m "$(PKG_FILE_MODES)" $$(IDIR_$(1)) $$(PDIR_$(1))
+       $(FAKEROOT) $(SCRIPT_DIR)/ipkg-build -m "$(FILE_MODES)" $$(IDIR_$(1)) $$(PDIR_$(1))
        @[ -f $$(IPKG_$(1)) ]

     $(1)-clean:
```

I tried this, it works. When set PKG_FILE_MODES in the Makefile this variable is used for all packages (except when a particular package has FILE_MODES set, of course). When on the other hand this is not set but a particular package has FILE_MODES set, then this doesn't affect the other packages.

I think this behavior is easy to understand, because the same behavior is already implemented for many other variables in OpenWrt:

PKG_MAINTAINER vs MAINTAINER
PKG_SOURCE_SUBDIR vs SUBDIR
PKG_LICENSE vs LICENSE
...

aparcar, 2020-09-09:
LGTM, mind creating a patch?

micmac1, 2020-09-09:
Don't mind at all :)

micmac1, 2020-09-13:
I sent two PRs to use fakeroot instead of postinst script, one for apache and the other for mariadb. Thanks to @aparcar and @dangowrt for implementing this!

closed 2024-02-25 by dangowrt