openwrt/packages #12094: zerotier: Removing network from configuration doesn't dejoin it from zerotier

alpharde, 2020-05-06:
Maintainer: @mwarning
Environment: Orange Pi R1, OpenWrt 19.07.2 r10947-65030d81f3 

Description: Removing a network from the configuration file (_/etc/config/zerotier_) doesn't dejoin or even remove it from the de-facto configuration files which now seem to be static at _/etc/config/zero_. The unwanted network will persist until manually removed with _zerotier-cli leave_.

mwarning, 2020-05-06:
Hi,

so the problem is that you use 'option config_path' in /etc/zerotier and if you remove a network (list join) from /etc/zerotier' and restart zerotier, then it does not take into effect because zerotier uses /etc/zerotier. Is that what you mean?

alpharde, 2020-05-06:
Whoops, didn't notice some luci package I was trying did replace the init script.
Not a problem on your end, sorry!

closed 2020-05-06 by alpharde