openwrt/packages #12095: shadowsocks-libev: ss-rules fails to start

amezin, 2020-05-06:
Maintainer: @yousong 
Environment: ZyXEL NBG6817, OpenWrt SNAPSHOT r13136+15-ad27c133eb / LuCI Master git-20.125.70452-f0a0d28 (built from master today). Maybe the problem was present for a long time, I haven't been using ss-redir recently

Description:
ss-rules fails at line `ip route flush table 100` in `ss_rules_flush` with message `Flush terminated`.
Does not create any iptables chains or ipsets

yousong, 2020-05-06:
With `ip-tiny` installed, I can reproduce the issue

```
root@OpenWrt:/# ip route flush table 100
Error: ipv4: FIB table does not exist.
Flush terminated
```

Likely it was caused by iproute2 package.  Will take a look anyway

yousong, 2020-05-06:
Hi @amezin , I pushed a one line fix to address the behavior change in iproute2 and kernel.  Please check if it works for you.  Thank you.

closed 2020-05-06 by yousong

amezin, 2020-05-07:
Yes, it works. Actually I did the same change as a "quick fix" locally. I'm only a bit worried that ignoring all errors may be not the best approach in the long term.

yousong, 2020-05-07:
> Yes, it works. Actually I did the same change as a "quick fix" locally. I'm only a bit worried that ignoring all errors may be not the best approach in the long term.

I had the same concern but go ahead with current change, hoping that if the `errexit` way of shell scripting will catch most if not all unexpected errors later in the "ip route add" command.