openwrt/packages #12108: libssh2: compile error

stintel, 2020-05-08:
Maintainer: @jslachta 
Environment: OpenWrt master r13174-73fa1aba94 on octeon

Description:
Building an image for my ERL fails, due to a compile error in libssh2, which is not even enabled in my .config.

```
stijn@taz ~/Development/LEDE/source $ grep ssh2 .config
# CONFIG_PACKAGE_libssh2 is not set
```

stintel, 2020-05-08:
[compile.txt](https://github.com/openwrt/packages/files/4596616/compile.txt)

neheb, 2020-05-08:
https://github.com/openwrt/packages/issues/11796

stintel, 2020-05-08:
I can apparently work around by adding this to the Makefile:
```
PKG_BUILD_DEPENDS:=libmbedtls
```
Don't know if that's an acceptable solution, or why libssh2 is being built in the first place.

neheb, 2020-05-08:
It is not. And no clue.

libssh2 can use either mbedtls or openssl.

micmac1, 2020-05-08:
I would say the problem is that some package has a "lazy" dependency on libssh2. For instance nmap. It has "+libssh2" for nmap-full. If instead it had "+PACKAGE_nmap-full:libssh2" then libssh2 wouldn't be built before nmap while nmap-full is not selected while, for instance, ncat is selected.

micmac1, 2020-05-08:
For cases where you have something like "+ARIA2_SFTP:libssh2" it seems not possible to avoid this. In that case adding PKG_BUILD_DEPENDs to libssh2 might be best. Although I would probably try to make PKG_BUILD_DEPENDs dependent on the abscence of CONFIG_LIBSSH2_MBEDTLS and CONFIG_LIBSSH2_OPENSSL.

cotequeiroz, 2020-05-08:
> I would say the problem is that some package has a "lazy" dependency on libssh2. For instance nmap. It has "+libssh2" for nmap-full. If instead it had "+PACKAGE_nmap-full:libssh2" then libssh2 wouldn't be built before nmap while nmap-full is not selected while, for instance, ncat is selected.

Why fixing it at the source by adding `+PACKAGE_nmap-full:libssh2` to nmap?  It tackles the problem right at the origin.  The generated dependencies can be seen at `tmp/.packagedeps`.  The packages that depend on libssh2 unconditionally can be seen with:
```
$ grep -E '.*=.*libssh2/compile($|[^)])' tmp/.packagedeps
$(curdir)/feeds/packages/nmap/compile += $(curdir)/feeds/packages/libssh2/compile $(curdir)/feeds/packages/pcre/compile $(curdir)/feeds/packages/python3/compile $(curdir)/libs/libpcap/compile $(curdir)/libs/openssl/compile $(curdir)/libs/toolchain/compile $(curdir)/libs/zlib/compile $(curdir)/utils/lua5.3/compile $(if $(CONFIG_GCC_LIBSSP),$(curdir)/libs/toolchain/compile) $(if $(CONFIG_USE_GLIBC),$(curdir)/libs/toolchain/compile)
```

The problem with adding mbedtls to PKG_BUILD_DEPENDS, and not tacking nmap is that you're always forcing a build of libssh2 and mbedtls even if you don't need them (you're not selecting nmap-full.

On its own, adding mbedtls to libssh2 PKG_BUILD_DEPENDS will case, in an eventual failure to build mbedtls, libssh2 to fail as well, even in cases where mbedtls is not used.

Besides, it is a good to not have to build both libssh2 and mbedtls, if you're not going to build nmap-full.  In the times of python2 and python3, this strategy was more widely used, as the annoyance of having to needlessly build another version of python was far greater.

micmac1, 2020-05-08:
> The problem with adding mbedtls to PKG_BUILD_DEPENDS, and not tacking nmap is that you're always forcing a build of libssh2 and mbedtls even if you don't need them (you're not selecting nmap-full. [...]

I agree with you. But fixing libssh2's PKG_BUILD_DEPENDS _as well_ would allow it to be used in other packages PKG_BUILD_DEPENDS, without libssh2 being selected in menuconfig.

cotequeiroz, 2020-05-08:
There are no other packages in this situation.
```
$ grep -E 'libssh2/compile\)' tmp/.packagedeps
$(curdir)/feeds/packages/aria2/compile += $(curdir)/libs/toolchain/compile $(curdir)/libs/zlib/compile $(if $(CONFIG_ARIA2_ASYNC_DNS),$(curdir)/feeds/packages/c-ares/compile) $(if $(CONFIG_ARIA2_COOKIE),$(curdir)/feeds/packages/sqlite3/compile) $(if $(CONFIG_ARIA2_EXPAT),$(curdir)/feeds/packages/expat/compile) $(if $(CONFIG_ARIA2_GMP),$(curdir)/libs/gmp/compile) $(if $(CONFIG_ARIA2_GNUTLS),$(curdir)/feeds/packages/gnutls/compile) $(if $(CONFIG_ARIA2_LIBGCRYPT),$(curdir)/feeds/packages/libgcrypt/compile) $(if $(CONFIG_ARIA2_LIBXML2),$(curdir)/feeds/packages/libxml2/compile) $(if $(CONFIG_ARIA2_NETTLE),$(curdir)/libs/nettle/compile) $(if $(CONFIG_ARIA2_OPENSSL),$(curdir)/libs/openssl/compile) $(if $(CONFIG_ARIA2_SFTP),$(curdir)/feeds/packages/libssh2/compile) $(if $(CONFIG_GCC_LIBSSP),$(curdir)/libs/toolchain/compile) $(if $(CONFIG_USE_GLIBC),$(curdir)/libs/toolchain/compile)
$(curdir)/network/utils/curl/compile += $(curdir)/libs/toolchain/compile $(curdir)/system/ca-certificates/compile $(if $(CONFIG_GCC_LIBSSP),$(curdir)/libs/toolchain/compile) $(if $(CONFIG_LIBCURL_GNUTLS),$(curdir)/feeds/packages/gnutls/compile) $(if $(CONFIG_LIBCURL_LDAP),$(curdir)/feeds/packages/openldap/compile) $(if $(CONFIG_LIBCURL_LIBIDN2),$(curdir)/feeds/packages/libidn2/compile) $(if $(CONFIG_LIBCURL_MBEDTLS),$(curdir)/libs/mbedtls/compile) $(if $(CONFIG_LIBCURL_NGHTTP2),$(curdir)/libs/nghttp2/compile) $(if $(CONFIG_LIBCURL_OPENSSL),$(curdir)/libs/openssl/compile) $(if $(CONFIG_LIBCURL_SSH2),$(curdir)/feeds/packages/libssh2/compile) $(if $(CONFIG_LIBCURL_THREADED_RESOLVER),$(curdir)/libs/toolchain/compile) $(if $(CONFIG_LIBCURL_WOLFSSL),$(curdir)/libs/wolfssl/compile) $(if $(CONFIG_LIBCURL_ZLIB),$(curdir)/libs/zlib/compile) $(if $(CONFIG_USE_GLIBC),$(curdir)/libs/toolchain/compile)
$(curdir)/feeds/packages/mc/compile += $(curdir)/feeds/packages/glib2/compile $(curdir)/libs/ncurses/compile $(curdir)/libs/toolchain/compile $(curdir)/utils/util-linux/compile $(if $(CONFIG_BUILD_NLS),$(curdir)/libs/libiconv-full/compile) $(if $(CONFIG_BUILD_NLS),,$(curdir)/libs/gettext/compile) $(if $(CONFIG_BUILD_NLS),,$(curdir)/libs/libiconv/compile) $(if $(CONFIG_GCC_LIBSSP),$(curdir)/libs/toolchain/compile) $(if $(CONFIG_MC_VFS),$(curdir)/feeds/packages/libssh2/compile) $(if $(CONFIG_MC_VFS),$(curdir)/feeds/packages/libtirpc/compile) $(if $(CONFIG_USE_GLIBC),$(curdir)/libs/toolchain/compile)
```
- aria2 will build libssh2 only if `ARIA2_SFTP` is selected; `ARIA2_SFTP` will select `PACKAGE_libssh2`
- curl will build libssh2 only if `LIBCURL_SSH2` is selected; `LIBCURL_SSH2` selects `PACKAGE_libssh2`
- mc will build libssh2 only if `MC_VFS` is selected; `MC_VFS` selects `PACKAGE_libssh2`.

The problem with nmap is that any package that's part of `feeds/packages/nmap/compile` will call `feeds/packages/libssh2/compile`--this is what `tmp/.packagedeps` is defining.  However, only `nmap-full` is selecting `PACKAGE_libssh2`.  When `PACKAGE_libssh2` is built without being selected, no configuration symbols are defined.  So both `LIBSSH2_MBEDTLS` and `LIBSSH2_OPENSSL` are undefined.  
Another, IMO better, option to make it more fail proof is to change the condition to compile libmbedtls, from `+LIBSSH2_MBEDTLS:libmbedtls` to `+!LIBSSH2_OPENSSL:libmbedtls`.  This way, no unnecessary packages are built.

micmac1, 2020-05-09:
+!LIBSSH2_OPENSSL:libmbedtls seems best. Thanks!

PR #12118 (2020-05-12): libssh2: add mbedtls to PKG_BUILD_DEPENDS
This adds to PKG_BUILD_DEPENDS to prevent built failures. A comment is
left in case anybody stumbles over this in the future.

Closes #12108

Maintainer: @jslachta 
Compile tested: ath79 master
Run tested: No change to package

Description:
Hi all,

With this change mbedtls gets pulled in when needed (fixes build failures like described in the referenced issue). In case the user selects the symbol LIBSSH2_OPENSSL mbedtls doesn't get pulled in, because it's not needed.

I did not bump the revision because package remains unchanged.

Kind regards,
Seb
commit c0da37b1 (2020-05-12): libssh2: switch default to OpenSSL and fix depends

All packages in the tree that depend on libssh2 _and_ have their own
dependency on a TLS library, depend on OpenSSL by default. It makes
sense for the libssh2 package to default to OpenSSL as well. This way it
does not pull in a second TLS library.

Also, the OpenSSL dependency is changed to
"+!LIBSSH2_MBEDTLS:libopenssl", which fixes build issues in case libssh2
is not selected but built anyway.

And last, Config.in file is removed and its content added to the
Makefile instead.

Closes #12108
closed 2020-05-12 by jslachta