openwrt/packages #12120: <ddns-scripts_cloudflare.com-v4>: cURL Error: '127'

dreamer2q, 2020-05-09:
Maintainer: @cherrot 

Environment: 
```bash
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r2506-b80d00ee"
OPENWRT_BOARD="ipq40xx/generic"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r2506-b80d00ee"
```

Description:

```bash
090803       : verbose mode  : 0 - run normal, NO console output
 090803       : check interval: 600 seconds
 090803       : force interval: 259200 seconds
 090804       : retry interval: 60 seconds
 090804       : retry counter : 3 times
 090804       : No old process
 090804       : last update: never
 090804       : Verify DNS server '1.1.1.1'
 090805       : #> timeout 2 -- /usr/bin/nc 1.1.1.1 53 </dev/null >/var/run/ddns/netwan.dat 2>/var/run/ddns/netwan.err
 090805       : Detect registered/public IP
 090806       : #> /usr/bin/nslookup net.myjj.tech 1.1.1.1 >/var/run/ddns/netwan.dat 2>/var/run/ddns/netwan.err
 090806       : Registered IP '60.176.40.144' detected
 090807  info : Starting main loop at 2020-05-09 09:08
 090807       : Detect local IP on 'interface'
 090807       : #> ip -o addr show dev pppoe-netwan scope global >/var/run/ddns/netwan.dat 2>/var/run/ddns/netwan.err
 090808       : Local IP '60.176.40.144' detected on interface 'pppoe-netwan'
 090808       : Forced Update - L: '60.176.40.xxx' == R: '60.176.40.xxx'
 090809       : parsing script '/usr/lib/ddns/update_cloudflare_com_v4.sh'
 090809       : debug path: /sbin:/usr/sbin:/bin:/usr/bin
 090810       : #> /usr/bin/curl -RsS -o /var/run/ddns/netwan.dat --stderr /var/run/ddns/netwan.err --noproxy '*' --header 'X-Auth-Email: xxxxxxx@qq.com'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=net'
 090811 ERROR : cURL Error: '127'
```
I am user I install curl in /usr/bin/curl

```bash
root@OpenWrt:~# file /usr/bin/curl
/usr/bin/curl: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-armhf.so.1, no section header
```

I do not know why `command not found` will occur but when I run `/usr/bin/curl -RsS -o /var/run/ddns/netwan.dat --stderr /var/run/ddns/netwan.err --noproxy '*' --header 'X-Auth-Email: xxxxxxx@qq.com'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=net'` from bash, there is nothing about `command not found` error

cherrot, 2020-05-09:
Does you stderr file `/var/run/ddns/netwan.err` have any content?

dreamer2q, 2020-05-09:
Actually file `netwan.err` did not exist since command `curl` did not get executed

```bash
root@OpenWrt:~# ls /var/run/ddns/
myddns_ipv4.ip   myddns_ipv4.pid  myddns_ipv6.ip   netwan.ip        netwan.pid
```