openwrt/packages #12125: collectd-mod-iwinfo: no stations value reported

aqwserf, 2020-05-09:
Maintainer: @jow- @hnyman 
Environment: OpenWrt 19.07.2 r10947-65030d81f3 on Linksys WRT3200ACM (mvebu/cortexa9 / arm_cortex-a9_vfpv3)
Description:
After following [this tuto](https://blog.christophersmart.com/2019/09/09/monitoring-openwrt-with-collectd-influxdb-and-grafana/), the package `collectd-mod-iwinfo` (or another one, but which one?) is supposed to send the value `stations`, according to `types.db`:
```
root@linksys:~# cat /usr/share/collectd/types.db | grep station
stations                value:GAUGE:0:256
```
but in InfluxDB, there's nothing:

```
root@grafana:~# influx
Connected to http://localhost:8086 version 1.8.0
InfluxDB shell version: 1.8.0
> use collectd
Using database collectd
> select * from iwinfo_value GROUP BY * ORDER BY ASC LIMIT 1
name: iwinfo_value
tags: host=linksys, instance=wlan0, type=bitrate
time                value
----                -----
1588802371153919913 0

name: iwinfo_value
tags: host=linksys, instance=wlan0, type=signal_noise
time                value
----                -----
1588802371184091977 -96

name: iwinfo_value
tags: host=linksys, instance=wlan0, type=signal_power
time                value
----                -----
1588802371153983736 0

name: iwinfo_value
tags: host=linksys, instance=wlan0, type=signal_quality
time                value
----                -----
1588802371184236038 0

name: iwinfo_value
tags: host=linksys, instance=wlan1, type=bitrate
time                value
----                -----
1588802371093965557 0
....
```

According to [this patch](https://github.com/openwrt/packages/blob/openwrt-19.07/utils/collectd/patches/900-add-iwinfo-plugin.patch#L203), published a month ago, there are some additions precisely concerning the value `stations`, and it fits my `collectd` version (5.11.0-1), but I don't what's the purpose of this patch, and if it applies to my router.

Does anyone has the same issue ? Is there another package to install to provide this value ?

hnyman, 2020-05-09:
No idea about using "influx", but collectd collects the data quite normally,  as shown here in the luci statistics:
[image]

The gets normally stored in the rrd database, where it is fetchable.

```
root@router1:~# rrdtool dump /tmp/rrd/router1/iwinfo-wlan0/signal_quality.rrd |
head -n 50
<!-- Round Robin Database Dump -->
<rrd>
        <version> 0001 </version>
        <step> 30 </step> <!-- Seconds -->
        <lastupdate> 1589036615 </lastupdate> <!-- 2020-05-09 18:03:35 EEST -->

        <ds>
                <name> value </name>
                <type> GAUGE </type>
                <minimal_heartbeat> 60 </minimal_heartbeat>
                <min> 0.0000000000e+00 </min>
                <max> NaN </max>

                <!-- PDP Status -->
                <last_ds> UNKN </last_ds>
                <value> 2.8500000000e+02 </value>
                <unknown_sec> 0 </unknown_sec>
        </ds>

<!-- Round Robin Archives -->
        <rra>
                <cf> AVERAGE </cf>
                <pdp_per_row> 1 </pdp_per_row> <!-- 30 seconds -->
                <xff> 1.0000000000e-01 </xff>

                <cdp_prep>
                        <ds><value> NaN </value>  <unknown_datapoints> 0 </unknown_datapoints></ds>
                </cdp_prep>
                <database>
                        <!-- 2020-05-09 17:04:00 EEST / 1589033040 --> <row><v> 5.9833333333e+01 </v></row>
                        <!-- 2020-05-09 17:04:30 EEST / 1589033070 --> <row><v> 6.0000000000e+01 </v></row>
                        <!-- 2020-05-09 17:05:00 EEST / 1589033100 --> <row><v> 6.0000000000e+01 </v></row>
                        <!-- 2020-05-09 17:05:30 EEST / 1589033130 --> <row><v> 6.0833333333e+01 </v></row>
                        <!-- 2020-05-09 17:06:00 EEST / 1589033160 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:06:30 EEST / 1589033190 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:07:00 EEST / 1589033220 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:07:30 EEST / 1589033250 --> <row><v> 5.9333333333e+01 </v></row>
                        <!-- 2020-05-09 17:08:00 EEST / 1589033280 --> <row><v> 5.9000000000e+01 </v></row>
                        <!-- 2020-05-09 17:08:30 EEST / 1589033310 --> <row><v> 6.0666666667e+01 </v></row>
                        <!-- 2020-05-09 17:09:00 EEST / 1589033340 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:09:30 EEST / 1589033370 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:10:00 EEST / 1589033400 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:10:30 EEST / 1589033430 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:11:00 EEST / 1589033460 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:11:30 EEST / 1589033490 --> <row><v> 6.1000000000e+01 </v></row>
                        <!-- 2020-05-09 17:12:00 EEST / 1589033520 --> <row><v> 6.0166666667e+01 </v></row>
                        <!-- 2020-05-09 17:12:30 EEST / 1589033550 --> <row><v> 5.9166666667e+01 </v></row>
                        <!-- 2020-05-09 17:13:00 EEST / 1589033580 --> <row><v> 5.9000000000e+01 </v></row>
                        <!-- 2020-05-09 17:13:30 EEST / 1589033610 --> <row><v> 5.9000000000e+01 </v></row>
                        <!-- 2020-05-09 17:14:00 EEST / 1589033640 --> <row><v> 5.9000000000e+01 </v></row>
```

I close this issue, as the data is collected  normally.

I have no experience with using influx in connection with collectd, but your problem is not in the collectd iwinfo module.

closed 2020-05-09 by hnyman

aqwserf, 2020-05-10:
Thanks for the quick feedback @hnyman .
I was able to display the stations too on my OpenWRT, so I indeed guess that collectd works fine.
The file you try to dump is `signal_quality.rrd`, not `stations.rrd`, but I'm not sure about the value. For example, on mine:

```
root@linksys:/tmp/linksys/iwinfo-wlan0# rrdtool dump stations.rrd | head -50
<!-- Round Robin Database Dump -->
<rrd>
	<version> 0001 </version>
	<step> 60 </step> <!-- Seconds -->
	<lastupdate> 1589102951 </lastupdate> <!-- 2020-05-10 11:29:11 CEST -->

	<ds>
		<name> value </name>
		<type> GAUGE </type>
		<minimal_heartbeat> 120 </minimal_heartbeat>
		<min> 0.0000000000e+00 </min>
		<max> 2.5600000000e+02 </max>

		<!-- PDP Status -->
		<last_ds> UNKN </last_ds>
		<value> 1.1000000000e+01 </value>
		<unknown_sec> 0 </unknown_sec>
	</ds>

<!-- Round Robin Archives -->
	<rra>
		<cf> AVERAGE </cf>
		<pdp_per_row> 1 </pdp_per_row> <!-- 60 seconds -->
		<xff> 1.0000000000e-01 </xff>

		<cdp_prep>
			<ds><value> NaN </value>  <unknown_datapoints> 0 </unknown_datapoints></ds>
		</cdp_prep>
		<database>
			<!-- 2020-05-09 15:30:00 CEST / 1589031000 --> <row><v> NaN </v></row>
			<!-- 2020-05-09 15:31:00 CEST / 1589031060 --> <row><v> NaN </v></row>
			<!-- 2020-05-09 15:32:00 CEST / 1589031120 --> <row><v> NaN </v></row>
			<!-- 2020-05-09 15:33:00 CEST / 1589031180 --> <row><v> NaN </v></row>
```
Is `<value> 1.1000000000e+01 </value>` corresponding to the actual number of stations connected ?
Anyway, I'm trying to understand why the values are not exported. Do you know if there are additional parameters to put in `/etc/collectd.conf`? Is `collectd-mod-iwinfo` really responsible for the value `stations` ?

hnyman, 2020-05-10:
If your question was specifically about the stations data, it gets read and stored normally for me in my WRT3200ACM currently running 19.07:
```
 OpenWrt 19.07-SNAPSHOT, r11050-a8c92e9eda
 -----------------------------------------------------
root@router3:~# rrdtool dump /tmp/rrd/router3/iwinfo-wlan1/stations.rrd | head -n 50
<!-- Round Robin Database Dump -->
<rrd>
        <version> 0001 </version>
        <step> 30 </step> <!-- Seconds -->
        <lastupdate> 1589113960 </lastupdate> <!-- 2020-05-10 15:32:40 EEST -->

        <ds>
                <name> value </name>
                <type> GAUGE </type>
                <minimal_heartbeat> 60 </minimal_heartbeat>
                <min> 0.0000000000e+00 </min>
                <max> 2.5600000000e+02 </max>

                <!-- PDP Status -->
                <last_ds> UNKN </last_ds>
                <value> 2.0000000000e+01 </value>
                <unknown_sec> 0 </unknown_sec>
        </ds>

<!-- Round Robin Archives -->
        <rra>
                <cf> AVERAGE </cf>
                <pdp_per_row> 1 </pdp_per_row> <!-- 30 seconds -->
                <xff> 1.0000000000e-01 </xff>

                <cdp_prep>
                        <ds><value> NaN </value>  <unknown_datapoints> 0 </unknown_datapoints></ds>
                </cdp_prep>
                <database>
                        <!-- 2020-05-10 14:33:00 EEST / 1589110380 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:33:30 EEST / 1589110410 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:34:00 EEST / 1589110440 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:34:30 EEST / 1589110470 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:35:00 EEST / 1589110500 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:35:30 EEST / 1589110530 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:36:00 EEST / 1589110560 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:36:30 EEST / 1589110590 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:37:00 EEST / 1589110620 --> <row><v> 2.3333333333e+00 </v></row>
                        <!-- 2020-05-10 14:37:30 EEST / 1589110650 --> <row><v> 2.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:38:00 EEST / 1589110680 --> <row><v> 2.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:38:30 EEST / 1589110710 --> <row><v> 2.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:39:00 EEST / 1589110740 --> <row><v> 2.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:39:30 EEST / 1589110770 --> <row><v> 2.6666666667e+00 </v></row>
                        <!-- 2020-05-10 14:40:00 EEST / 1589110800 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:40:30 EEST / 1589110830 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:41:00 EEST / 1589110860 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:41:30 EEST / 1589110890 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:42:00 EEST / 1589110920 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:42:30 EEST / 1589110950 --> <row><v> 3.0000000000e+00 </v></row>
                        <!-- 2020-05-10 14:43:00 EEST / 1589110980 --> <row><v> 3.0000000000e+00 </v></row>
```

Like you see, I had most of the time 2-3 stations active (2.0e+00 or 3.0e+00).

There aren't any parameters. (or you can specify some interfaces like wlan0, but that should be unnecessary). The available wifi stats may depend on the wifi driver, but the normal data is available in WRT3200ACM.

>  Is collectd-mod-iwinfo really responsible for the value stations ?

Yes. Note that I read the values from the database identified as "iwinfo-wlan1/stations"...
`root@router3:~# rrdtool dump /tmp/rrd/router3/iwinfo-wlan1/stations.rrd`

But I noticed that your database path was unusual, so you have configured the luci_statistics to have a non-standard location, right?   And also the time interval was not the normal luci_statistics 30 seconds, but one minute. So,, you have changes also other stuff in the luci_statistics (and/or directly collectd) config.

As long as you are unsure that you actually can collect data, it might be better to stick to the defaults.

aqwserf, 2020-05-10:
Yes, I was just making some tests, and I don't collectd to store any data on the router, just sending them to influxdb, and I find that 30s is a bit short so...

Anyway, I found where the problem was: in `types.db` on the server. There was no definition of `stations` in it, unlike the one on my OpenWRT, but I don't know why... After adding it and restarting, eveything works fine !

Sorry for opening a wrong issue...

hnyman, 2020-05-10:
>  in types.db on the server. There was no definition of stations in it, unlike the one on my OpenWRT, but I don't know why...

because that data type is OpenWrt specific, added by the "this patch" that you referred...
https://github.com/openwrt/packages/blob/openwrt-19.07/utils/collectd/patches/900-add-iwinfo-plugin.patch#L254

The whole iwinfo wifi data thing is added to collectd by OpenWrt. So, if you use the global collectd data definition types.db, that addition is naturally missing.

aqwserf, 2020-05-10:
Oh, I see. That make sense. Thanks for the explanation !