openwrt/packages #12129: samba4: compile error

gw826943555, 2020-05-10:
Maintainer: @Andy2244
Environment: OpenWrt master snapshot
Description:
building an image for my ipq40xx device fails, due to samba4 compile error.
error log:
```
2020-05-09T04:58:45.3956600Z Waf: Leaving directory `/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/bin/default'
2020-05-09T04:58:45.3958450Z 'install' finished successfully (8m14.072s)
2020-05-09T04:58:45.6888432Z touch /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/.built
2020-05-09T04:58:45.6954372Z mkdir -p /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/targets/ipq40xx/generic/packages /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/ipkg-arm_cortex-a7_neon-vfpv4/samba4-libs/CONTROL /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo
2020-05-09T04:58:45.6982794Z install -d -m0755 /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/ipkg-arm_cortex-a7_neon-vfpv4/samba4-libs/usr/lib
2020-05-09T04:58:45.7010539Z cp -fpR /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/ipkg-install/usr/lib/*.so* /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/ipkg-arm_cortex-a7_neon-vfpv4/samba4-libs/usr/lib/
2020-05-09T04:58:45.7099614Z # rpath-install
2020-05-09T04:58:45.7118372Z cp -fpR /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/ipkg-install/usr/lib/samba /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/ipkg-arm_cortex-a7_neon-vfpv4/samba4-libs/usr/lib/
2020-05-09T04:58:45.7312937Z find /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/samba-4.11.6/ipkg-arm_cortex-a7_neon-vfpv4/samba4-libs -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
2020-05-09T04:58:46.2096373Z Package samba4-libs is missing dependencies for the following libraries:
2020-05-09T04:58:46.2102301Z libpam.so.0
2020-05-09T04:58:46.2109899Z make[3]: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/packages/arm_cortex-a7_neon-vfpv4/packages/samba4-libs_4.11.6-1_arm_cortex-a7_neon-vfpv4.ipk] Error 1
2020-05-09T04:58:46.2110405Z Makefile:431: recipe for target '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/packages/arm_cortex-a7_neon-vfpv4/packages/samba4-libs_4.11.6-1_arm_cortex-a7_neon-vfpv4.ipk' failed
2020-05-09T04:58:46.2115718Z make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/feeds/packages/net/samba4'
2020-05-09T04:58:46.2116895Z time: package/feeds/packages/samba4/compile#562.35#108.36#548.47
2020-05-09T04:58:46.2127849Z make[2]: *** [package/feeds/packages/samba4/compile] Error 2
2020-05-09T04:58:46.2128314Z package/Makefile:111: recipe for target 'package/feeds/packages/samba4/compile' failed
2020-05-09T04:58:46.2128669Z make[2]: *** Waiting for unfinished jobs....
```
It can be solved by adding 'CONFIG_PACKAGE_libpam=y' in .config file

Andy2244, 2020-06-09:
Only happens on a "dirty" build, where libpam was somehow build/installed previously and than deselected later on. On a fresh/clean checkout this should not happen.

gw826943555, 2020-06-12:
> Only happens on a "dirty" build, where libpam was somehow build/installed previously and than deselected later on. On a fresh/clean checkout this should not happen.

I removed most packages and this didn't happen.

closed 2020-06-12 by gw826943555