openwrt/packages #12135: iptgeoip: Download script defunct

Kufat, 2020-05-10:
Maintainer: @jow- 
Environment: Any arch/model, v19.07 (also appears to apply to trunk)

Description: As of 2019/12/31, Maxmind discontinued their publicly-available geoIP downloads. The upstream package has [a fix](https://sourceforge.net/p/xtables-addons/xtables-addons/ci/4b43df69f52589931ef8f1b390620eb1ae7ddd68/tree/geoip/xt_geoip_dl?diff=4c68c774c884bcaf15e256e894e250595987c2b1&diformat=sidebyside), but it hasn't incorporated in a released version yet. 

I'd suggest either getting a new version from their git repo or getting the latest released version and patching xt_geoip_dl with the new address.

pprindeville, 2020-05-11:
Working on getting `xtables-addons-3.10` released.

dl12345, 2020-05-27:
> Maxmind discontinued their publicly-available geoIP downloads. 

The latest versions of xtables-addons have switched to using the dbip database from db-ip.com. The version in master is working fine with the new database (and new `/usr/share/xt_geoip` layout), although the download script needs tweaking to get it to work properly. 

```
#!/bin/sh
  
DATE=$(/bin/date +'%Y-%m')
XT_GEOIP_DATA="/usr/share/xt_geoip"
XT_GEOIP_SCRIPTS="/usr/lib/xtables-addons"
XT_GEOIP_DBIP="dbip-country-lite"
XT_GEOIP_URL="https://download.db-ip.com/free/${XT_GEOIP_DBIP}-${DATE}.csv.gz"

[ ! -d ${XT_GEOIP_DATA} ] && mkdir -p ${XT_GEOIP_DATA}
/usr/bin/wget -q ${XT_GEOIP_URL} -O /tmp/${XT_GEOIP_DBIP}.csv.gz > /dev/null 2>&1
/usr/bin/gzip -f -d /tmp/${XT_GEOIP_DBIP}.csv.gz
${XT_GEOIP_SCRIPTS}/xt_geoip_build -S /tmp -D ${XT_GEOIP_DATA} > /dev/null 2>&1
/bin/rm -f /tmp/${XT_GEOIP_DBIP}.csv
```

closed 2025-11-14 by GeorgeSapkin