openwrt/packages #12138: collectd: complains about missing /etc/collectd/conf.d

dl12345, 2020-05-11:
collectd logs error messages that it cannot find `/etc/collectd/conf.d`. Nothing critical of course, but it's ugly

```
May 11 00:32:48 openwrt collectd[7222]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory
```

feckert, 2020-05-11:
The default value for this `/tmp/collectd.d` https://github.com/openwrt/packages/blob/master/utils/collectd/files/collectd.init#L313

dl12345, 2020-05-11:
Well, as you can see from the log message, it's complaining about a non-existent `/etc/collectd/conf.d`. Any idea why?

feckert, 2020-05-11:
* Are you using your own collectd config?
If so its up to you to ensure this directory exists.
* Or do you use uci and changed the default option?
If so i think the line `mkdir -p "$Include"` should solve this issue and is already included in the master
https://github.com/openwrt/packages/blob/59b91fc342aa69c8aa107587fd29ff2161159c65/utils/collectd/files/collectd.init#L313-L318

dl12345, 2020-05-11:
Nope. This was a totally clean install, compiled from the latest master yesterday. I booted the install, enabled and setup collected using the luci app and that error message popped up.

feckert, 2020-05-11:
If you use luci-app-statistics then it should get fixed there. Please open a ticket in the luci feed it is not related to collectd itself. I think it must be fixed in the init file of luci_statistics https://github.com/openwrt/luci/blob/c32097d7d525d0c3e960e558d1c561c9c596bf96/applications/luci-app-statistics/root/etc/init.d/luci_statistics

closed 2020-05-11 by feckert

dl12345, 2020-05-11:
I'll do another clean install and check all the config files prior to and after enabling before filing another report...

feckert, 2020-05-11:
@dl12345 thanks for you report and your work
If something seems strange to you, you can always open an issue, that's what these are for.
:+1:

dl12345, 2020-05-11:
You're right that it seems to be something with luci_app_statistics

I just did a clean install. Since I use an x86_64, I mounted the root before I even booted it and I can see the following, which explains the problem. 

installed version of `/etc/collectd.conf`

```
BaseDir "/var/run/collectd"
Include "/etc/collectd/conf.d"
PIDFile "/var/run/collectd.pid"
PluginDir "/usr/lib/collectd"
TypesDB "/usr/share/collectd/types.db"
Interval 30
ReadThreads 2

LoadPlugin iwinfo
<Plugin iwinfo>
>---IgnoreSelected false
</Plugin>

LoadPlugin memory
<Plugin memory>
>---ValuesPercentage false
>---ValuesAbsolute true
</Plugin>

LoadPlugin cpu
<Plugin cpu>
>---ValuesPercentage false
>---ReportByCpu false
>---ReportByState false
</Plugin>

LoadPlugin rrdtool
<Plugin rrdtool>
>---DataDir "/tmp/rrd"
>---RRARows 100
>---RRASingle true
>---RRATimespan 3600
>---RRATimespan 86400
>---RRATimespan 604800
>---RRATimespan 2678400
>---RRATimespan 31622400
</Plugin>

LoadPlugin load

LoadPlugin interface
<Plugin interface>
>---IgnoreSelected false
>---Interface "br-lan"
</Plugin>
```

installed version of `/etc/config/collectd`

```
config globals 'globals'
>---option alt_config_file "/etc/collectd.conf"
#>--option BaseDir '/var/run/collectd'
#>--option Include '/tmp/collectd.d'
#>--option PIDFile '/var/run/collectd.pid'
#>--option PluginDir '/usr/lib/collectd'
#>--option TypesDB '/usr/share/collectd/types.db'
#>--option Interval '30'
#>--option ReadThreads '2'
```

installed version of `/etc/config/luci_statistics`

```
# general settings
  
config statistics 'collectd'
>---option BaseDir '/var/run/collectd'
>---option Include '/etc/collectd/conf.d'
>---option PIDFile '/var/run/collectd.pid'
>---option PluginDir '/usr/lib/collectd'
>---option TypesDB '/usr/share/collectd/types.db'
>---option Interval '30'
>---option ReadThreads '2'

config statistics 'rrdtool'
>---option default_timespan '1hour'
>---option image_width '600'
>---option image_path '/tmp/rrdimg'
```