openwrt/packages #12139: nginx: incorrect worker_processes invocation in init.d script

dl12345, 2020-05-11:
The init.d script tries to invoke nginx with a number of worker_processes equal to the number of cores. This is incorrect, as the master process is included in the maximum number of processes. The number of worker_processes should be $NCPUS-1

```
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
May 11 01:31:45 openwrt nginx[6063]: 2020/05/11 01:31:45 [alert] 6063#0: no more than 8 processes can be spawned
```

The following code
```
start_service() {
   ...
    procd_set_param command /usr/sbin/nginx -c "${NGINX_CONF}" \
        -g "daemon off; worker_processes $NCPUS;"
```

should read

```
start_service() {
   ...
    procd_set_param command /usr/sbin/nginx -c "${NGINX_CONF}" \
        -g "daemon off; worker_processes $(expr $NCPUS - 1);"
```

cc @peter-stadler

pprindeville, 2020-05-11:
The number of CPU's should be doubled if the cores are hyper-threaded...

dl12345, 2020-05-11:
Then surely the code needs to check for that? It just tries to blindly instantiate a number of worker processes equal to the number of cores. 

On a non-hyper threaded system, this behaviour results in the errors in my first comment. But even on a hyper threaded system, it would still need to spawn one less process than the total number of virtual cores due to the master process being counted in the maximum allowable total.

dl12345, 2020-05-11:
I've just been testing by invoking manually and specifying the number of processes on both my non hyper-threaded router and my hyper-threaded server. 

It seems that it can in fact spawn a number of workers equal to the number of cores, so my initial report is incorrect..

There's obviously another problem that's causing these errors in the logs relating to not being able to spawn more processes. I'll need to dig into it a bit further to see whether I can reproduce it consistently.

peter-stadler, 2020-05-11:
@dl12345 I think the message comes from [ngx_process.c](https://hg.nginx.org/nginx/file/tip/src/os/unix/ngx_process.c#l106), where it should output [NGX_MAX_PROCESSES](https://github.com/nginx/nginx/search?q=NGX_MAX_PROCESSES&unscoped_q=NGX_MAX_PROCESSES). But the linked search for this symbol shows that it is defined as `1024`. I do not know how it ends up with `8` …

@pprindeville The setting is based on the [official recomendation](https://nginx.org/en/docs/ngx_core_module.html#worker_processes), would it be better to use `worker_processes auto;` in the .conf file?

peter-stadler, 2020-05-11:
Silly me, I looked at the wrong directory on my system, there is a patch for [footprint optimization](https://github.com/openwrt/packages/pull/2101/commits/fae2d0c81c6634884b0f89676ab2eb0ddbb63831) used …

peter-stadler, 2020-05-11:
Hmm, can we get the number of CPUs at compile time?

champtar, 2020-05-11:
just change the optimization patch to 32 CPU then put auto ?

peter-stadler, 2020-05-11:
There are configs with [more](https://github.com/openwrt/openwrt/blob/master/target/linux/rockchip/config-default), thought about using `$(CONFIG_NR_CPUS)`, but maybe we should look for the maximum of these numbers or stop using that patch …
@dirkfeytons: Do you remember, how much difference this optimization makes?

dl12345, 2020-05-12:
@peter-stadler that footprint optimization patch explains why I'm getting the problem. 

My router is an 8-core Intel C3758.  So this code will cause the error every single time, since in my case that loop will exit and `s == NGX_MAX_PROCESSES`, since it always spawns a single master process and then a number of worker processes equal to the number of c
```
-#define NGX_MAX_PROCESSES         1024
+#define NGX_MAX_PROCESSES         8
```

```
ngx_pid_t
ngx_spawn_process(ngx_cycle_t *cycle, ngx_spawn_proc_pt proc, void *data,+
−
    char *name, ngx_int_t respawn)
{
    u_long     on;
    ngx_pid_t  pid;
    ngx_int_t  s;

    if (respawn >= 0) {
        s = respawn;

    } else {
        for (s = 0; s < ngx_last_process; s++) {
            if (ngx_processes[s].pid == -1) {
                break;
            }
        }

        if (s == NGX_MAX_PROCESSES) {
            ngx_log_error(NGX_LOG_ALERT, cycle->log, 0,
                          "no more than %d processes can be spawned",
                          NGX_MAX_PROCESSES);
            return NGX_INVALID_PID;
        }
    }
```

peter-stadler, 2020-05-12:
@dl12345 For now set in the `/etc/init.d/nginx` the command to:
```
	procd_set_param command /usr/sbin/nginx -c "${NGINX_CONF}" \
		-g "daemon off; worker_processes $((NCPUS>8 ? 8 : NCPUS));"
```
I will think about it and make a PR with a better solution the next days. I [asked upstream](https://mailman.nginx.org/pipermail/nginx-devel/2020-May/013154.html) if they want to NGX_MAX_PROCESSES for `worker_processes auto;` Then I would have used the advise of @champtar …

dl12345, 2020-05-12:
@peter-stadler - I've already removed that patch and put in the worker processes auto option in my build. I suspect this will be a more flexible long term solution that will accommodate more machines. I'm not sure that it's accomplishing much other than maybe reducing the size of a data structure. I don't really have memory issues though, as the router has 16GB of RAM

peter-stadler, 2020-05-12:
This would be my favourite solution for all systems. Therefore I would like to know if this is a big penalty (on some systems). In that case we could distinguish depending on the CONFIG_NR_CPUS, but I am not sure if that variable can (and should) be used for packages …

pprindeville, 2020-05-12:
> @pprindeville The setting is based on the official recomendation, would it be better to use worker_processes auto; in the .conf file?

@peter-stadler Without looking at the logic that handles that, I can't say.

> Hmm, can we get the number of CPUs at compile time?

No, by definition the number of CPU's is a run-time determination.

Within shell, you can test for hyper-threading as:

```
# egrep -q '^flags\t\t: .*\<ht\>' /proc/cpuinfo
```

dirkfeytons, 2020-05-12:
I had a quick look in the nginx code again to refresh my memory. It allocates `ngx_process_t ngx_processes[NGX_MAX_PROCESSES] `. A quick calculation tells me that struct is about 32 bytes (on 32-bit systems). That means that reducing `NGX_MAX_PROCESSES` from 1024 to 8 saves a bit less than 32KB of memory.

At the time that was an easy gain for the lowest end platforms I was working with. Nowadays I guess it doesn't really matter that much anymore. Most of that memory will not be touched in typical OpenWrt devices so it probably won't consume physical memory.

I haven't been directly working on OpenWrt devices since a number of years but my current advice would be to just drop this change. I suspect low-end OpenWrt devices won't be running nginx soon.

pprindeville, 2020-05-12:
Regarding:

```
-#define NGX_MAX_PROCESSES         1024
+#define NGX_MAX_PROCESSES         8
```

`1024` is too much and `8` seems like too little.

The current count of Xeon and ARM ThunderX seems to be about `64`.  On Xeon those would be HT, so `128` threads currently...

pprindeville, 2020-05-12:
> At the time that was an easy gain for the lowest end platforms I was working with. Nowadays I guess it doesn't really matter that much anymore. Most of that memory will not be touched in typical OpenWrt devices so it probably won't consume physical memory.

@dirkfeytons This seems like a topic for the `CONFIG_SKINNY` (etc) conversation...

peter-stadler, 2020-05-13:
@dirkfeytons Thank you very much.
So, I would remove the patch and use the upstream preset again :-)

Then I would add `worker_processes auto;` to the `nginx.conf` (resp. the `uci.conf.template` for the [new version](https://github.com/openwrt/packages/pull/11456)) and remove the option from the init. This sets the value to `ngx_ncpu` (so quite the same), but allows the user to change it permanently.

@pprindeville Btw, on my desktop system `/proc/cpuinfo` lists also the virtual CPUs (so it would take care of hyperthreading already), is this different on OpenWrt?

pprindeville, 2020-05-13:
> Btw, on my desktop system `/proc/cpuinfo` lists also the virtual CPUs (so it would take care of hyperthreading already), is this different on OpenWrt?

@peter-stadler Virtual CPU's?  Not following.  Are you on a VM?  Can you paste your output?

dl12345, 2020-05-13:
he means that `/proc/cpuinfo` will show a number of processors equal to the number of virtual cores if it has hyperthreading, rather than the number of physical cores. 

My Core i7-4790 does too: it's a 4-core with hyperthreading and it shows 8 cpus in `/proc/cpuinfo`

peter-stadler, 2020-05-14:
(This is right. I have a similar output.)

pprindeville, 2020-05-25:
Okay, so no special handling is required for hyper-threaded x86's.

dl12345, 2020-05-27:
Fixed by https://github.com/openwrt/packages/pull/12159

closed 2020-05-27 by dl12345