openwrt/packages #12151: Libvirtd: Add libvirtd support

peterwillcn, 2020-05-12:
Who can add [libvirtd]( https://libvirt.org/manpages/libvirtd.html) support

lucize, 2020-05-15:
is doable, but there are a lot of dependencies on it

taken from alpine:
_daemon_deps="bridge-utils dmidecode dnsmasq ip6tables iptables"
_client_deps="pm-utils gnutls-utils netcat-openbsd"
depends="lvm2"
depends_dev="libtirpc-dev"
makedepends="augeas-dev cyrus-sasl-dev device-mapper
	e2fsprogs-dev gnutls-dev libcap-ng-dev libgpg-error-dev
	netcf-dev libnl3-dev libxml2-dev libxslt-dev libtasn1-dev
	lvm2-dev lxc-dev libgcrypt-dev parted-dev perl pkgconfig
	eudev-dev zlib-dev yajl-dev libpcap-dev curl-dev libpciaccess-dev
	polkit-dev readline-dev linux-headers jansson-dev
	$depends_dev perl-xml-xpath
	fuse-dev util-linux libxml2-utils bash-completion $_daemon_deps $_client_deps
	automake autoconf libtool rpcsvc-proto py3-docutils

peterwillcn, 2020-09-19:
Anyone can add libvirtd support