openwrt/packages #12155: mpd-mini_0.21.20-1_armv5-3.2.ipk - not compiled with libid3tag

untiefe, 2020-05-12:
Maintainer: Ted Hess <thess@kitschensync.net>
Environment: QNAP NAS via Entware

Description:

Hello,

after contacting Ted directly (he pointed me here) I'd like to open the issue here:

I’m using the mpd-mini package on my QNAP Network storage and everything runs well, except that I can’t see the mp3 albums in my mpd app (but I can play them when I go via files). I think the reason is, that the mpd-mini is not compiled with libid3tag (mpd-full package is not working either). When I look into the control file of the ipk I cannot find libid3tag in the Depends section:

```Depends: libc, libssp, librt, libpthread, zlib, libcurl, libpthread, libmpdclient, libstdcpp, libiconv-full, alsa-lib, boost, libexpat, libflac, libmad, libvorbisidec```

Therefore I tried to create my own custom ipk file (manually) and only added libid3tag in the depends list and repackaged the stuff. But then I get the error message:

```Collected errors:
 * pkg_init_from_file: Malformed package file mpd-mini_0.21.20-1_armv5-3.2.ipk.```

I probably have to recompile it with this tag anyway, but couldn't - as meson seems not to be available for this platform.

Could you provide an updated package that supports libid3tag ? I think this is a must have for any mpd package!?

Thanks a lot for any help,

Michael

neheb, 2020-05-12:
Entware...

18.06 has 0.20.9
19.07 has 0.21.23

So you want id3tag for mpd-mini? mpd-full has it.

untiefe, 2020-05-12:
Thanks for pointing it out to me. I had other - pulseaudio - issues with mpd-full beforehand, and didn't see the libid3tag in the mpd-full package first.

But now, with the right mpd config it is working on my system - with libid3tag.

Anyhow, it would be great to include libid3tag into mpd-mini as well, as nearly everyone will have some mp3s to play...

Thanks a lot already :-)

neheb, 2020-05-13:
Seems that this was an oversight from https://github.com/openwrt/packages/commit/2cedae0aa9befe1135c83f007942d89ec906658e

libmpg123 got introduced without id3tag.

I'll put id3tag in mini next time I bump the version.

closed 2020-05-20 by neheb