openwrt/packages #12176: nginx: error log directive in nginx.conf

dl12345, 2020-05-15:
By design, luci calls `ubus call network get_proto_handlers` for all protocols and the browser attempts to fetch a matching javascript class for each one of these handler. However, there are no luci protocol handlers for batadv, batadv_vlan, batadv_hardif, bonding and potentially others, so the browser cannot load the files. As a result, the logs are flooded with error messages such as those shown below. 

I believe that we should have an error_log directive in nginx.conf to suppress these. Something along the lines of 

```
error_log /var/log/nginx/error.log crit;
```

```
May 14 23:48:07 openwrt nginx[6976]: 2020/05/14 23:48:07 [error] 12302#0: *26 open() "/www/luci-static/resources/protocol/batadv.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-static/   resources/protocol/batadv.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/"
May 14 23:48:07 openwrt nginx[6976]: 2020/05/14 23:48:07 [error] 12300#0: *10 open() "/www/luci-static/resources/protocol/batadv_hardif.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-   static/resources/protocol/batadv_hardif.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/"
May 14 23:48:07 openwrt nginx[6976]: 2020/05/14 23:48:07 [error] 12302#0: *21 open() "/www/luci-static/resources/protocol/batadv_vlan.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-     static/resources/protocol/batadv_vlan.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/"
May 14 23:48:07 openwrt nginx[6976]: 2020/05/14 23:48:07 [error] 12300#0: *28 open() "/www/luci-static/resources/protocol/bonding.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-static/  resources/protocol/bonding.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/"
May 14 23:48:11 openwrt nginx[6976]: 2020/05/14 23:48:11 [error] 12300#0: *4 open() "/www/luci-static/resources/protocol/batadv.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-static/    resources/protocol/batadv.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/network/network"
May 14 23:48:11 openwrt nginx[6976]: 2020/05/14 23:48:11 [error] 12300#0: *10 open() "/www/luci-static/resources/protocol/batadv_hardif.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-   static/resources/protocol/batadv_hardif.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/network/network"
May 14 23:48:11 openwrt nginx[6976]: 2020/05/14 23:48:11 [error] 12302#0: *21 open() "/www/luci-static/resources/protocol/batadv_vlan.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-     static/resources/protocol/batadv_vlan.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/network/network"
May 14 23:48:11 openwrt nginx[6976]: 2020/05/14 23:48:11 [error] 12302#0: *26 open() "/www/luci-static/resources/protocol/bonding.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-static/  resources/protocol/bonding.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/network/network"
May 14 23:48:14 openwrt nginx[6976]: 2020/05/14 23:48:14 [error] 12300#0: *10 open() "/www/luci-static/resources/protocol/batadv.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-static/   resources/protocol/batadv.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/status/overview"
May 14 23:48:14 openwrt nginx[6976]: 2020/05/14 23:48:14 [error] 12300#0: *28 open() "/www/luci-static/resources/protocol/batadv_hardif.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-   static/resources/protocol/batadv_hardif.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/status/overview"
May 14 23:48:14 openwrt nginx[6976]: 2020/05/14 23:48:14 [error] 12302#0: *21 open() "/www/luci-static/resources/protocol/batadv_vlan.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-     static/resources/protocol/batadv_vlan.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/status/overview"
May 14 23:48:14 openwrt nginx[6976]: 2020/05/14 23:48:14 [error] 12302#0: *26 open() "/www/luci-static/resources/protocol/bonding.js" failed (2: No such file or directory), client: 192.168.1.2, server: _lan, request: "GET /luci-static/  resources/protocol/bonding.js?v=git-20.135.61997-a7d1e25 HTTP/1.1", host: "192.168.1.1", referrer: "https://192.168.1.1/cgi-bin/luci/admin/status/overview"
```

Any thoughts on other ways to deal with this? I've got syslog-ng redirecting the syslog messages to a separate file, but that's not a way that will work for a default openwrt installation.

@Ansuel 
@peter-stadler

peter-stadler, 2020-05-15:
I would rather set `error_log stderr crit;` it in `/etc/nginx/conf.d/luci.locations` for `location /luci-static` only, what do you think?
In the long run, it would be good if those files would be installed: There is a `luci-proto-bonding` package, but I do not see one for `batadv` …

dl12345, 2020-05-15:
That would probably be cleaner. Happy to test it it....

dl12345, 2020-05-15:
Yep, that seems to work

dl12345, 2020-05-27:
Fixed by https://github.com/openwrt/packages/pull/12184

closed 2020-05-27 by dl12345