openwrt/packages #12185: [utils/sysstat] hard coded log path renders user conf option obsolete

n8v8R, 2020-05-16:
Maintainer: @ratkaj
Environment: Master, 19.07
___
#### Description:

with compiler CONFIGURE_VARS set to `sa_dir="/var/log/sysstat"` the log path is hardcoded into the binary and thereby rendering user conf option `SA_DIR=` in */etc/config/sysstat* obsolete.

neheb, 2020-05-16:
ping @lucize

lucize, 2020-05-17:
Hi, I'm not the maintainer but, maybe that config option can overwrite the binary hardcoded path ?

n8v8R, 2020-05-17:
The point of opening the issue is that the hardcoded path supersedes/obsoletes the user option specified in the conf file */etc/config/sysstat*.

If you initiate `/usr/lib/sysstat/sadc -S DISK -F -L -` manually and without `[ -d /var/log/sysstat ] || mkdir -p /var/log/sysstat` (as coded in the init script) it prints:

>Cannot open /var/log/sysstat/sa17: No such file or directory
___
on a marginal note the maintainer sited for the package does not appear actively maintaining the package for quite some time

closed 2020-05-30 by neheb