openwrt/packages #12189: libwebm: fails to build for mvebu-cortexa9

SvenRoederer, 2020-05-16:
Maintainer: nobody
Environment: mvebu-cortexa9 (Turin Omnia), OpenWrt-master (https://github.com/openwrt/openwrt/commit/8097fd4d5)

Description: The package fails to compile after switch to CMake (https://github.com/openwrt/packages/pull/12137) on mvebu-cortexa9.

```
[ 46%] Building C object CMakeFiles/webpencode.dir/src/enc/quant_enc.c.o
In file included from /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libwebp-1.1.0/src/dsp/quant.h:20,
                 from /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libwebp-1.1.0/src/enc/quant_enc.c:18:
/mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libwebp-1.1.0/src/dsp/quant.h: In function 'horizontal_add_uint32x4':
/mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/include/arm_neon.h:561:1: error: inlining failed in call to always_inline 'vadd_u32': target specific option mismatch
 vadd_u32 (uint32x2_t __a, uint32x2_t __b)
 ^~~~~~~~
In file included from /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libwebp-1.1.0/src/enc/quant_enc.c:18:
/mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libwebp-1.1.0/src/dsp/quant.h:26:10: note: called from here
   return vadd_u32(vreinterpret_u32_u64(vget_low_u64(b)),
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   vreinterpret_u32_u64(vget_high_u64(b)));
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libwebp-1.1.0/src/dsp/quant.h:20,
                 from /mnt/local/btrfs/build/freifunk/ffberlin/firmware/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/libwebp-1.1.0/src/enc/quant_enc.c:18:
```

It seems that the package tries to build NEON-specific extensions, which are likely not enabed for the target. ```src/dsp/quant.h``` has a ```#if defined(WEBP_USE_NEON)``` around the line causing the error. Looks like the same issue have been seen when the package was introduced initially (https://github.com/openwrt/packages/issues/9740#issuecomment-531458404) and reported by @rmandrad.

Before switching to CMake the Makefile supplied the option ```--disable-neon-rtcd``` to the configure script.

neheb, 2020-05-16:
I assume adding -DWEBP_USE_NEON=OFF to CMAKE_OPTIONS will fix it.

neheb, 2020-05-17:
ping @jefferyto 

I tried:

```
--- a/libs/libwebp/Makefile
+++ b/libs/libwebp/Makefile
@@ -39,9 +39,10 @@ CMAKE_OPTIONS += \
        -DWEBP_BUILD_VWEBP=OFF \
        -DWEBP_BUILD_WEBPINFO=OFF \
        -DWEBP_BUILD_WEBPMUX=OFF \
-       -DWEBP_BUILD_EXTRAS=OFF
+       -DWEBP_BUILD_EXTRAS=OFF \
+       -DWEBP_HAVE_NEON=OFF
 
-TARGET_CFLAGS += -flto
+TARGET_CFLAGS += -flto -UWEBP_USE_NEON -UWEBP_HAVE_NEON
 
 define Package/libwebp/install
        $(INSTALL_DIR) $(1)/usr/lib
```

Not working. Something else is enabling it...

neheb, 2020-05-17:
much simpler...

```
+ifneq ($(findstring arm,$(CONFIG_ARCH)),)
+ifeq ($(findstring neon,$(CONFIG_CPU_TYPE)),)
+CMAKE_OPTIONS += -DWEBP_ENABLE_SIMD=OFF
+endif
+endif
```

closed 2020-05-17 by neheb

SvenRoederer, 2020-05-18:
Thanks for fixing.