openwrt/packages #12209: python-openssl: Add dependency on ca-certificates

tiran, 2020-05-18:
# Issue template (remove lines from top till here)

Maintainer: @jefferyto
Environment: all

Description:

The ``python3-openssl`` package and ``python-openssl`` package should depend on  ``ca-certificates`` package. Since Python 2.7.9 and 3.4.0 the ssl module encourages developers to use ``ssl.create_default_context()`` to create a working and securely configured SSL context object. The implementation assumes that the platform has a correctly configured OpenSSL libssl that can load the default trust anchors (root CA certificates) with ``SSL_CTX_set_default_verify_paths()``.

Also see https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=960869 for my Debian related bug.

jefferyto, 2020-05-18:
@commodo Any opinion on this? This sounds reasonable to me though I'm far from an OpenSSL expert.

hnyman, 2020-05-18:
> should depend on ca-certificates 

Why should it be ca-certificates, and not the pseudo-package "ca-certs" ????

so that both ca-certificates and the newer ca-bundle can fulfill the dependency, and the user is not forced to install also ca-certificates...

If the dependency would be introduced, I suggest that the newish "ca-certs" is used.

https://github.com/openwrt/openwrt/commit/dd299805ad18472a8245b4524a25e4381e166057

tiran, 2020-05-18:
I didn't know about ``ca-certs``. A generic provider sounds even better to me.

commodo, 2020-05-19:
`ca-certs` sounds good to me as well;
i think we may need to open this as far back as OpenWrt 18.06?

jefferyto, 2020-05-19:
@commodo I think updating 18.06 for this makes sense - I'll open some PRs soon.

hnyman, 2020-05-19:
Note that the newish ca-certs may not exist in the relase branches.

commodo, 2020-05-20:
We can probably close this now.

closed 2020-05-20 by hnyman