openwrt/packages #12223: python3-setuptools: Required source file missing

vanrein, 2020-05-19:
Device problem occurs on
  - Root filesystem for AMD64
  - Presumably just as well on any other system

Software versions of OpenWrt/LEDE release, packages, etc.
  - OpenWrt 18.06.5 r7897-9d401013fc
  - `python3-setuptools` - 40.6.2-1 - Python 3.6 setuptools module
  - `python3-setuptools-src` - 40.6.2-1 - . (Contains the Python3 sources for this package).

Steps to reproduce
  - `PYTHONPATH=/tmp/xyz/lib/python python3 some/setup.py install –home=/tmp/xyz`

I was trying to install https://gitlab.com/arpa2/quick-der from it’s setup.py. This is a very compact library for DER parsing. DER is useful for LDAP, X.509, KIP; the latter is a key exchange tool with modest footprint I am trying to port to OpenWRT, https://gitlab.com/arpa2/kip/

The package `python3-setuptools-src` does not work, and misunderstands setuptools’ extensions because it uses distutils instead.

The package python3-setuptools cannot use the `python3 setup.py install –home` variant, which is used in the CMakeLists.txt for Quick DER (for bootstrapping reasons: need an ASN.1 parser written in Python to build some of the code). The `–home` variant requires a source file, which is stripped, namely `/usr/lib/python3.6/site-packages/setuptools/site-patch.py`

I cannot install `python3-setuptools` and `python3-setuptools-src` at the same time, due to clashing files (manifest).

When I took this file out of the `python3-setuptools-src` package and installed it in the dirtree of `python3-setuptools` which otherwise just has `.pyc`, it did work.  But that’s a crude hack and it's the incompatible result of stripping the code a bit too eagerly.

brada4, 2020-05-21:
The source archive in pypi is broken, missing some other files too. It does not work with any pip on a common Linux distro. Wheel does indeed.
https://pypi.org/project/arpa2.quickder-tools/#files

jefferyto, 2020-05-21:
Hello - first, thanks for the detailed information.

As a quick note: as only security fixes will go into 18.06, I'm afraid none of the issues you've identified will be fixed for that series. 19.07.3 is the current stable release; I suggest you upgrade your system if you can. (In fact, if you are working on porting software to OpenWrt, I suggest you run master / snapshots. 19.07 will only receive bug and security fixes; very few if any new software will be added to that branch.)

On the specific issues:

* Conflict between python3-setuptools and python3-setuptools-src - this has (somewhat inadvertently) already been fixed in 19.07 and master. If you want to backport the relevant changes to 18.06, you can take the `find` command changes in 85c805d0ada334c0b90c3b74bc4b7c5779627e11. (The issue is, because there are spaces in some file names in the setuptools package, the previous `find` commands were not deleting those files correctly.)

* `site-patch.py` missing from python3-setuptools - this has been fixed in #12245 and #12247. Again, if you want to backport this fix to 18.06, please take a look at those PRs.

I believe this covers everything mentioned in this issue - please let me know if I missed anything.

vanrein, 2020-05-24:
Yes, this answers the issues I had.  Sorry for not having the latest / greatest yet.

I thought about it a little more, and there's something I'm trying that is not OpenWRT-style at all ; that is to build on OpenWRT like it is a plain development platform.

Interestingly, the rootfs (for amd64) lulled me into this mode.  It is quite close to allowing this sort of thing.  But at some point there is a lack of `-dev` packages and I had to conclude that cross-building is the way to go for OpenWRT.

I will close the issue now.

closed 2020-05-24 by vanrein