openwrt/packages #12233: Cannot install package busybox in  OpenWrt-V2X-18.06.1

ngavrish, 2020-05-20:
HEAD of OpenWrt-V2X-18.06.1

	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/bober/openwrt2/OpenWrt-V2X/build_dir/target-mips_24kc_musl/root-ar71xx/usr/share/dnsmasq/rfc6761.conf
	But that file is already provided by package  * dnsmasq
 * opkg_install_cmd: Cannot install package dnsmasq-full.
 * check_data_file_clashes: Package busybox wants to install file /home/bober/openwrt2/OpenWrt-V2X/build_dir/target-mips_24kc_musl/root-ar71xx/usr/sbin/brctl
	But that file is already provided by package  * bridge
 * opkg_install_cmd: Cannot install package busybox.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/bober/openwrt2/OpenWrt-V2X'
package/Makefile:104: recipe for target '/home/bober/openwrt2/OpenWrt-V2X/staging_dir/target-mips_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/bober/openwrt2/OpenWrt-V2X/staging_dir/target-mips_24kc_musl/stamp/.package_install] Error 2

neheb, 2020-05-20:
ping @yousong is busybox missing alternatives for brctl?

yousong, 2020-05-21:
Couldn't reproduce it with 18.06.5, likely an issue in your fork.

```
root@OpenWrt:/# ls -l `which brctl`
lrwxrwxrwx    1 root     root            17 Nov  5  2019 /usr/sbin/brctl -> ../../bin/busybox
root@OpenWrt:/# opkg install bridge
Installing bridge (1.5-6) to root...
Downloading http://downloads.openwrt.org/releases/18.06.5/packages/x86_64/packages/bridge_1.5-6_x86_64.ipk
Configuring bridge.
root@OpenWrt:/# ls -l `which brctl`
lrwxrwxrwx    1 root     root            31 May 21 01:40 /usr/sbin/brctl -> /usr/libexec/bridge-utils-brctl
root@OpenWrt:/#
```

neheb, 2020-05-21:
It seems to be based on .1. Maybe that's why.

neheb, 2020-05-31:
Closing. This issue needs to be shown that it's an OpenWrt issue and not some random fork's.

closed 2020-05-31 by neheb