openwrt/packages #12238: dawn: 100% cpu

lucize, 2020-05-20:
maintainer @PolynomialDivision 
testing dawn on a single node I have always 100% cpu for one core and another issue is after one client leaves the ap area I get this in syslog, repeating over and over
the cpu usage is 100% even when no hostapd app is installed

```
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: xx:xx:xx:xx:xx:xx is not connected
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: Unknown measurement mode 255
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: xx:xx:xx:xx:xx:xx is not connected
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: Unknown measurement mode 255
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: xx:xx:xx:xx:xx:xx is not connected
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: Unknown measurement mode 255
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: xx:xx:xx:xx:xx:xx is not connected
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: Unknown measurement mode 255
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: xx:xx:xx:xx:xx:xx is not connected
Wed May 20 14:07:37 2020 daemon.notice hostapd: Beacon request: Unknown measurement mode 255

```

PolynomialDivision, 2020-05-20:
Can u post your dawn config under `/etc/config/dawn`?

This is 802.11k stuff.

lucize, 2020-05-20:
hi,
it's just the default script, the thing is that if no wifi available (removed, or general template for target with and without wifi) cpu also run on 100%

```
cat /etc/config/dawn
config network
    option broadcast_ip         '10.0.0.255'
    option broadcast_port       '1025'
    option tcp_port             '1026'
    option network_option       '2' # 0 udp broadcast, 1 mutlicast, 2 tcp
    option shared_key           'Niiiiiiiiiiiiiik'
    option iv                   'Niiiiiiiiiiiiiik'
    option use_symm_enc         '1'
    option collision_domain     '-1'     # enter here aps which are in the same collision domain
    option bandwidth            '-1'     # enter network bandwidth

config ordering
    option sort_order           'cbfs'

config hostapd
    option hostapd_dir          '/var/run/hostapd'

config times
    option update_client        '10'
    option denied_req_threshold '30'
    option remove_client        '15'
    option remove_probe         '30'
    option remove_ap            '460'
    option update_hostapd       '10'
    option update_tcp_con       '10'
    option update_chan_util     '5'

config metric
    option ap_weight            '0'
    option ht_support           '0'
    option vht_support          '0'
    option no_ht_support        '0'
    option no_vht_support       '0'
    option rssi                 '10'
    option low_rssi             '-500'
    option freq                 '100' # some kind of band-steering
    option chan_util            '0'
    option max_chan_util        '-500'
    option rssi_val             '-60'
    option low_rssi_val         '-80'
    option chan_util_val        '140'
    option max_chan_util_val    '170'
    option min_probe_count      '0'
    option bandwith_threshold   '6'
    option use_station_count    '1'
    option max_station_diff     '1'
    option eval_probe_req       '0'
    option eval_auth_req        '0'     # no real reasoncode...
    option eval_assoc_req       '0'     # just deny assocs...
    option kicking              '0'
    option deny_auth_reason     '1'     # unspecified
    option deny_assoc_reason    '17'    # assoc rejected can't handle new station
    option use_driver_recog     '1'
    option min_number_to_kick   '3'
    option chan_util_avg_period '3'

```
the wifi is 03:00.0 Network controller: Qualcomm Atheros AR9285 Wireless Network Adapter (PCI-Express) (rev 01)

DISTRIB_ARCH='i386_pentium4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r13292-5f82691'

PolynomialDivision, 2020-05-20:
You are missing that:
https://github.com/openwrt/packages/blob/master/net/dawn/files/dawn.config#L57-L61

And that:
https://github.com/openwrt/packages/blob/master/net/dawn/files/dawn.config#L27

lucize, 2020-05-20:
ah, ok maybe it didn't replaced it on sysupgrade

closed 2020-05-20 by lucize