openwrt/packages #12248: boinc: did not transition from snapshot to 19.07.3

smoe, 2020-05-21:
Maintainer: @smoe  @N30dG
Environment: all, all, 19.07.3

Description: boinc is not installable with opkg, missing from 19.07 branch. Is there something we can do to help its adoption, say for 19.07.4 and/or 20.01?

hnyman, 2020-05-21:
For 20.0x it will get automatically, as the 20.0x branch will be branched off the HEAD of master (with boinc).

For 19.07 branch, somebody should author and test the PR for backporting.

(note that 19.07 is mainly June 2019 code plus fixes since then. But nothing gets automatically backported, and new packages are rarely backported. But if a package does not interfere with others, it is possible.)

Also the 19.07 packages are constantly built by the buildbot, so once the package is in 19.07 sources, it will be automatically available by opkg after a few days.

smoe, 2020-06-24:
Thank you for your kind reply, hnyman. My access to OpenWrt is mostly via Gluon. Initially I had hoped that Gluon would jump on 20.0x with its experimental release but the Gluon firmware just proudly adopted 19.07 (for their 2020.1.x and 2020.2.x trees) and from what I understand it will stick with 19.07 for a while. So, I would very much like to see boinc backported to 19.07. Can I do something to help initiate/accelerate the backport?

hnyman, 2020-06-24:
> Can I do something to help initiate/accelerate the backport?

You can build and test it on 19.07 and then author a backport pull request, like I already said above:

> > For 19.07 branch, somebody should author and test the PR for backporting.

BKPepe, 2020-08-17:
Closing. It was explained how it gets to OpenWrt 19.07. It does not need to be open here and stalled until somebody test it on 19.07 and send a pull request with cherry-picked related commits from master.

closed 2020-08-17 by BKPepe