openwrt/packages #12251: lxc: lua bindings have been removed upstream

henryk, 2020-05-22:
Maintainer: @lperkov 
Environment: OpenWRT master branch (de8b88ce17c3e19cf1fe366be0de2e3c376762b0), x86_64 (APU3)

Description: lxc lua bindings have been removed upstream.

## Steps to reproduce:
 
- Enable lxc package
- Build

## Actual results

During configure
````
configure: WARNING: unrecognized options: --disable-nls, --disable-cgmanager, --disable-gnutls, --disable-python, --enable-lua, --with-lua-pc
````

During compile
````
install -d -m0755 /home/henryk/data/openwrt/build_dir/target-x86_64_musl/lxc-4.0.2/ipkg-x86_64/lxc-lua/usr/lib/lua
cp -fpR /home/henryk/data/openwrt/build_dir/target-x86_64_musl/lxc-4.0.2/ipkg-install/usr/share/lua/5.1/lxc.lua /home/henryk/data/openwrt/build_dir/target-x86_64_musl/lxc-4.0.2/ipkg-x86_64/lxc-lua/usr/lib/lua/
cp: cannot stat '/home/henryk/data/openwrt/build_dir/target-x86_64_musl/lxc-4.0.2/ipkg-install/usr/share/lua/5.1/lxc.lua': No such file or directory
Makefile:297: recipe for target '/home/henryk/data/openwrt/bin/packages/x86_64/packages/lxc-lua_4.0.2-1_x86_64.ipk' failed
make[2]: *** [/home/henryk/data/openwrt/bin/packages/x86_64/packages/lxc-lua_4.0.2-1_x86_64.ipk] Error 1
````

The upstream commit removing lua support is https://github.com/lxc/lxc/commit/b52a5bef6fa56481a6d817f4c73cb91ed012fe2c (starting in lxc 3.0.0)

This problem probably has been introduced by https://github.com/openwrt/packages/commit/9f43594e3a28deeea18cfa40acd0b1388e64f600 (bump lxc from 2.1.1 to 4.0.2)

henryk, 2020-05-22:
Apparently the lua-lxc bindings have moved here: https://github.com/lxc/lua-lxc Though there hasn't been a commit for 2 years.
Not sure if they're still active and an OpenWRT package should be created for them, or if they should just be dropped alltogether.
Does luci-app-lxc depend on lxc lua?

neheb, 2020-05-22:
https://github.com/openwrt/packages/pull/12241 removes them.

The whole version bump was a mess.

lua-lxc not being active indicates they should not be introduced.

neheb, 2020-05-31:
I merged that. Closing.

closed 2020-05-31 by neheb