openwrt/packages #12259: domoticz can't see Aqara wireless switch

Kheamu, 2020-05-23:
Hello!

Domoticz 4.9700-3_arm_cortex-a7_neon-vfpv4 don't work with Aqara wireless switch with 1 or 2 keys(manually try to add it - nothing). Tried this version of domoticz at two different routers(asus and Xiaomi 3g) with openwrt.
At the same time domoticz on RaspberryPi 3 correctly work with it.

neheb, 2020-05-24:
domoticz is unmaintained currently.

neheb, 2020-06-06:
@dwmw2 ping

dwmw2, 2020-06-06:
Is there anything in the logs when you attempt to add it? Precisely what hardware type are you choosing from the 'Type' dropdown list when you add the hardware?

dwmw2, 2020-06-08:
Also, are you sure you're comparing with 4.9700 elsewhere? I'd expect you to be using Domoticz 2020.2 elsewhere, which is what I'm running on OpenWRT locally.

Kheamu, 2020-06-08:
Hello! Thank you for answer. Aqara wireless switch adds in domoticz only manual(in windows domoticz after switch clicked, it appear in the system, but not in domoticz openWRT). My domoticz was installed from openwrt repository (opkg install domoticz). Where may a download Domoticz v.2020.2? Sorry for my English.

Kheamu, 2020-06-08:
OS Windows there is for comparison, everything works in it.

dwmw2, 2020-06-09:
I'll build you one. What version of OpenWRT are you using, on what hardware?

dwmw2, 2020-06-09:
You didn't say what version of Domoticz you are using on other platforms.

Also, please increase the loglevel:
```
uci set domoticz.@domoticz[0].loglevel=6
```
When you restart Domoticz you should get a lot more information in the logs (`logread -f`). What does it say when you try to add the switch?

dwmw2, 2020-06-09:
Domoticz 2020.1 and supporting packages for OpenWRT 19.07 arm_cortex-a7_neon-vfpv4 at 
 http://david.woodhou.se/domoticz-19.07/

2020.2 is slightly harder to build on 19.07 and shouldn't make much difference.

Kheamu, 2020-06-11:
Hello! Thank you for answer! I use Asus RT-AC58U router and OpenWRT 19.07.3. At windows i try to use latest version of domoticz - 2020.2.11995. For Openwrt version number 4.9700-3.

Kheamu, 2020-06-11:
When i try to add switch log says nothing. In web interfece (when i try to add switch manually) there is "timeout" like if i don't press the switch.

Kheamu, 2020-06-11:
Will try to install you version of domoticz. Thank you!

Kheamu, 2020-06-11:
Can't enter web enterface of dmtcz (x.x.x.x:8080) In system logs:

Thu Jun 11 13:43:27 2020 daemon.info procd: Instance domoticz::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

dwmw2, 2020-06-11:
Can you log in via SSH and run domoticz manually? What does it say?

Kheamu, 2020-06-11:
Sorry, i forget to install last .ipk from your repository. Test in process) Thank you!

Kheamu, 2020-06-11:
2020-06-11 14:28:35.416 Status: Domoticz V2020.1 (c)2012-1969 GizMoCuz
2020-06-11 14:28:35.416 Status: Build Hash: 0, Date: 1969-12-31 19:00:00
2020-06-11 14:28:35.417 Status: Startup Path: /usr/share/domoticz/
2020-06-11 14:28:35.479 Status: PluginSystem: Failed dynamic library load, install the latest libpython3.x library that is available for your platform.
2020-06-11 14:28:35.500 Status: WebServer(HTTP) started on address: :: with port 8080
2020-06-11 14:28:35.508 Status: TCPServer: shared server started...
2020-06-11 14:28:35.509 Status: RxQueue: queue worker started...
2020-06-11 14:28:37.555 Status: XiaomiGateway (ID=2): Delaying worker startup...
2020-06-11 14:28:42.565 Status: XiaomiGateway (ID=2): Selected as main Gateway
2020-06-11 14:28:42.574 Status: NotificationSystem: thread started...
2020-06-11 14:28:42.575 Status: XiaomiGateway (ID=2): Worker started...
2020-06-11 14:28:42.582 Status: EventSystem: reset all events...
2020-06-11 14:28:42.583 Status: XiaomiGateway (ID=2): Using 192.168.31.220 for local IP address.
2020-06-11 14:28:42.592 Status: EventSystem: reset all device statuses...
2020-06-11 14:28:42.594 Status: EventSystem - Python: Failed dynamic library load, install the latest libpython3.x library that is available for your platform.
2020-06-11 14:28:42.595 Status: EventSystem: Started
2020-06-11 14:28:42.595 Status: EventSystem: Queue thread started...
2020-06-11 14:28:42.606 Status: XiaomiGateway: RGB Gateway Detected
2020-06-11 14:28:42.918 Status: XiaomiGateway (ID=2): unhandled model: , name: Xiaomi Switch

dwmw2, 2020-06-11:
If you run Domoticz 2020.1 on a platform where this works, what do you see in the corresponding log output?

dwmw2, 2020-06-11:
I also built a new package with the Xiaomi changes from 2020.2, at http://david.woodhou.se/domoticz-19.07/domoticz_2020.1-3_arm_cortex-a7_neon-vfpv4.ipk

Kheamu, 2020-06-11:
2020-06-12 00:45:57.111 (G) Light/Switch () <----before i manually adde switch

2020-06-12 00:45:57.086 Status: XiaomiGateway: New Xiaomi Wireless Dual Wall Switch Found (158d0002cb179f)
2020-06-12 00:46:22.705 (G) Temp (Xiaomi Temperature)
2020-06-12 00:46:22.724 (G) Humidity (Xiaomi Humidity)
2020-06-12 00:46:22.147 Status: EventSystem: reset all device statuses...
2020-06-12 00:46:24.051 (G) General/kWh (Xiaomi Smart Plug Usage)
2020-06-12 00:46:42.597 (G) Light/Switch (Aqara wireless)
2020-06-12 00:46:44.624 (G) Light/Switch (Aqara wireless)  <--- After i manually added swich(name enter manually too).

Kheamu, 2020-06-11:
Updated new version. Nothing changed.

But in first 5-10 seconds after command "service domoticz restart" if you manage to add manually at this time, it will appear, but after it doesn't work. (Error sending switch command, check device/hardware (idx=9) !   <---error in web interface after switching icon)

Log:
2020-06-11 15:12:59.096 Status: Domoticz V2020.1 (c)2012-1969 GizMoCuz
2020-06-11 15:12:59.097 Status: Build Hash: 0, Date: 1969-12-31 19:00:00
2020-06-11 15:12:59.097 Status: Startup Path: /usr/share/domoticz/
2020-06-11 15:12:59.160 Status: PluginSystem: Failed dynamic library load, install the latest libpython3.x library that is available for your platform.
2020-06-11 15:12:59.181 Status: WebServer(HTTP) started on address: :: with port 8080
2020-06-11 15:12:59.189 Status: TCPServer: shared server started...
2020-06-11 15:12:59.190 Status: RxQueue: queue worker started...
2020-06-11 15:13:01.234 Status: XiaomiGateway (ID=2): Delaying worker startup...
2020-06-11 15:13:05.186 Status: New sensors allowed for 5 minutes...
2020-06-11 15:13:06.245 Status: XiaomiGateway (ID=2): Selected as main Gateway
2020-06-11 15:13:06.251 Status: XiaomiGateway (ID=2): Worker started...
2020-06-11 15:13:06.254 Status: NotificationSystem: thread started...
2020-06-11 15:13:06.255 Status: XiaomiGateway (ID=2): Using 192.168.31.220 for local IP address.
2020-06-11 15:13:06.255 Status: EventSystem: reset all events...
2020-06-11 15:13:06.258 Status: EventSystem: reset all device statuses...
2020-06-11 15:13:06.260 Status: EventSystem - Python: Failed dynamic library load, install the latest libpython3.x library that is available for your platform.
2020-06-11 15:13:06.261 Status: EventSystem: Started
2020-06-11 15:13:06.261 Status: EventSystem: Queue thread started...
2020-06-11 15:13:06.325 Status: XiaomiGateway: RGB Gateway Detected
2020-06-11 15:13:06.629 Status: XiaomiGateway (ID=2): unhandled model: '', name: 'Unknown Xiaomi device'
2020-06-11 15:13:22.613 Status: EventSystem: reset all device statuses...
2020-06-11 15:13:26.116 Status: User: root initiated a switch command (9/Switch/Off)
2020-06-11 15:13:26.325 Status: XiaomiGateway: SendMessageToGateway failed on first attempt, will try again
2020-06-11 15:13:26.319 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-11 15:13:26.599 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-11 15:13:38.219 Status: New sensors allowed for 5 minutes...
2020-06-11 15:20:48.282 Status: User: root initiated a switch command (9/Switch/Off)
2020-06-11 15:20:48.447 Status: XiaomiGateway: SendMessageToGateway failed on first attempt, will try again
2020-06-11 15:20:48.446 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-11 15:20:48.714 Error: XiaomiGateway: unable to write command - Invalid key

dwmw2, 2020-06-11:
Try removing the hardware in Domoticz and then re-adding it?

Kheamu, 2020-06-11:
Re-add it. Nothig.

2020-06-11 15:51:48.653 Status: XiaomiGateway (ID=2): unhandled model: '', name: 'Unknown Xiaomi device'

Huupke, 2020-06-12:
Which switch model are you trying to add? It seems the gateway reports an empty model.

Kheamu, 2020-06-12:
Huupke, model Aqara wireless switch. In windows domoticz it is defined as "2020-06-12 00:45:57.086 Status: XiaomiGateway: New Xiaomi Wireless Dual Wall Switch Found (158d0002cb179f)"

Kheamu, 2020-06-12:
2020-06-11 15:12:59.160 Status: PluginSystem: Failed dynamic library load, install the latest libpython3.x library that is available for your platform.

Is it necessarily to install libpython for my ordinary system?

Huupke, 2020-06-12:
Are you sure it's the same device that is not recognized on Windows but is the dual wireless switch on OpenWRT?
It probably has to do with the communication with the gateway. Are you able to use a packet sniffer?

Kheamu, 2020-06-12:
Amendment: Xiaomi Aqara dual wireless switch is not recognized in OpenWRT, but simply and good work in windows domoticz and mihome. I'm sure 100%.
About sniffer - if you give brief instructions then I’ll figure it out.

Kheamu, 2020-06-12:
Found an interesting feature. At different time(I can not set the frequency) i can add my switch in OpenWRT domoticz and can work with it. But only for about 10 seconds.

Log at this moment and after:

2020-06-12 13:27:50.421 Status: User: Admin initiated a switch command (9/Switch2 Guestroom/Off)
2020-06-12 13:27:53.842 Status: User: Admin initiated a switch command (9/Switch2 Guestroom/On)
2020-06-12 13:27:57.956 Status: User: Admin initiated a switch command (9/Switch2 Guestroom/Off)
2020-06-12 13:27:58.124 Status: XiaomiGateway: SendMessageToGateway failed on first attempt, will try again
2020-06-12 13:27:58.116 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-12 13:27:58.397 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-12 13:28:01.886 Status: User: Admin initiated a switch command (9/Switch2 Guestroom/Off)
2020-06-12 13:28:02.054 Status: XiaomiGateway: SendMessageToGateway failed on first attempt, will try again
2020-06-12 13:28:02.051 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-12 13:28:02.325 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-12 13:28:21.846 Status: User: Admin initiated a switch command (9/Switch2 Guestroom/Off)
2020-06-12 13:28:22.014 Status: XiaomiGateway: SendMessageToGateway failed on first attempt, will try again
2020-06-12 13:28:22.014 Error: XiaomiGateway: unable to write command - Invalid key
2020-06-12 13:28:22.290 Error: XiaomiGateway: unable to write command - Invalid key

Huupke, 2020-06-12:
You can use tcpdump, TShark or Wireshark. Start recording before you reboot Domoticz. After Domoticz has started stop the capture and share it

Kheamu, 2020-06-14:
No.	Time	Source	Destination	Protocol	Length	Info
1	0.000000	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
2	0.066788	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.1? Tell 192.168.31.220
3	0.186669	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.116? Tell 192.168.31.220
4	0.283135	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.116 (Request)
5	0.283458	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.116 (Reply)
6	0.882669	192.168.31.222	192.168.31.220	SSH	118	Client: Encrypted packet (len=64)
7	0.894200	192.168.31.220	192.168.31.222	SSH	118	Server: Encrypted packet (len=64)
8	0.939538	192.168.31.222	192.168.31.220	TCP	54	59040 → 22 [ACK] Seq=65 Ack=65 Win=1025 Len=0
9	1.070046	192.168.31.220	192.168.31.222	SSH	150	Server: Encrypted packet (len=96)
10	1.075543	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.1? Tell 192.168.31.220
11	1.111415	192.168.31.222	192.168.31.220	TCP	54	59040 → 22 [ACK] Seq=65 Ack=161 Win=1025 Len=0
12	2.002305	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
13	2.086707	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.1? Tell 192.168.31.220
14	2.365961	192.168.31.220	224.0.0.22	IGMPv3	60	Membership Report / Leave group 224.0.0.50
15	2.546773	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.160? Tell 192.168.31.220
16	2.550099	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.160 (Request)
17	2.550411	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.160 (Reply)
18	3.235424	192.168.31.220	224.0.0.22	IGMPv3	60	Membership Report / Leave group 224.0.0.50
19	4.011684	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
20	4.576682	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.158? Tell 192.168.31.220
21	4.731826	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.158 (Request)
22	4.732154	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.158 (Reply)
23	4.866345	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
24	5.546708	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.122? Tell 192.168.31.220
25	5.594804	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.122 (Request)
26	5.595131	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.122 (Reply)
27	6.054721	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
28	6.095796	192.168.31.222	192.168.31.1	NBSS	55	NBSS Continuation Message
29	6.101241	192.168.31.1	192.168.31.222	TCP	66	445 → 49668 [ACK] Seq=1 Ack=2 Win=691 Len=0 SLE=1 SRE=2
30	7.395763	ASUSTekC_56:06:38	IPv4mcast_16	ARP	60	Who has 224.0.0.22? Tell 192.168.31.220
31	8.060143	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
32	8.435779	ASUSTekC_56:06:38	IPv4mcast_16	ARP	60	Who has 224.0.0.22? Tell 192.168.31.220
33	9.475738	ASUSTekC_56:06:38	IPv4mcast_16	ARP	60	Who has 224.0.0.22? Tell 192.168.31.220
34	9.720897	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.1 (Request)
35	9.721240	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.1 (Reply)
36	9.945353	192.168.31.220	224.0.0.22	IGMPv3	60	Membership Report / Join group 224.0.0.50 for any sources
37	10.004142	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.101? Tell 192.168.31.174
38	10.064224	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
39	10.836271	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 224.0.0.22? Tell 192.168.31.220
40	10.886291	ASUSTekC_56:06:38	Broadcast	ARP	60	ARP Announcement for 192.168.31.160
41	11.875768	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 224.0.0.22? Tell 192.168.31.220
42	12.068185	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
43	12.120191	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.222? Tell 192.168.31.1
44	12.120218	ASUSTekC_9c:6d:86	ASUSTekC_56:06:38	ARP	42	192.168.31.222 is at 74:d0:2b:9c:6d:86
45	12.915715	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 224.0.0.22? Tell 192.168.31.220
46	14.118127	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
47	14.764400	192.168.31.174	224.0.0.50	UDP	154	4321 → 9898 Len=112
48	14.774495	192.168.31.174	224.0.0.50	UDP	151	4321 → 9898 Len=109
49	14.826462	192.168.31.174	224.0.0.50	UDP	152	4321 → 9898 Len=110
50	14.847791	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
51	15.080730	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.174 (Request)
52	15.081050	ASUSTekC_56:06:38	Broadcast	ARP	60	Gratuitous ARP for 192.168.31.174 (Reply)
53	16.120071	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
54	18.124401	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
55	20.129231	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
56	22.176275	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
57	22.352769	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.125? Tell 192.168.31.1
58	24.181012	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
59	24.356473	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.125? Tell 192.168.31.1
60	24.829058	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
61	25.907213	ASUSTekC_56:06:38	Broadcast	ARP	60	ARP Announcement for 192.168.31.160
62	26.184038	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
[... cut ...]
153	81.125446	fe80::42b0:76ff:fe56:638	ff02::1	ICMPv6	174	Router Advertisement from 40:b0:76:56:06:38
154	81.573810	192.168.31.174	224.0.0.50	UDP	160	4321 → 9898 Len=118
155	82.293103	192.168.31.174	224.0.0.50	UDP	153	4321 → 9898 Len=111
156	82.604667	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
157	83.291963	192.168.31.174	224.0.0.50	UDP	153	4321 → 9898 Len=111
158	84.608270	ASUSTekC_56:06:38	Broadcast	ARP	60	Who has 192.168.31.165? Tell 192.168.31.1
159	84.717831	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137

Kheamu, 2020-06-14:
Where 31.174 - Gateway Xiaomi
            31.220 - router Asus with OpenWRT and Domoticz
            31.222 - PC
            31.1 - second router with PPPoE internet.

Huupke, 2020-06-14:
I need the actual content of the package comming from the gateway.

Kheamu, 2020-06-14:
Set filter ip.addr==192.168.31.174  <--(Mi Gateway)

184	7.228367	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
245	14.906297	192.168.31.174	224.0.0.50	UDP	154	4321 → 9898 Len=112
246	14.908213	192.168.31.174	224.0.0.50	UDP	151	4321 → 9898 Len=109
267	17.157515	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
294	21.764642	192.168.31.174	224.0.0.50	UDP	152	4321 → 9898 Len=110
362	26.782371	192.168.31.174	224.0.0.50	UDP	152	4321 → 9898 Len=110
363	27.192105	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
441	30.778905	192.168.31.174	224.0.0.50	UDP	152	4321 → 9898 Len=110
489	32.725723	192.168.31.174	224.0.0.50	UDP	234	4321 → 9898 Len=192
507	35.793561	192.168.31.174	224.0.0.50	UDP	152	4321 → 9898 Len=110
518	36.714847	192.168.31.174	224.0.0.50	UDP	152	4321 → 9898 Len=110
519	37.128071	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
523	37.738988	192.168.31.174	224.0.0.50	UDP	152	4321 → 9898 Len=110
669	47.159783	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
790	57.096088	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
1021	67.129803	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
1114	77.064558	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
1127	80.137084	192.168.31.174	224.0.0.50	UDP	154	4321 → 9898 Len=112
1130	80.237655	192.168.31.174	224.0.0.50	UDP	151	4321 → 9898 Len=109
1131	80.239673	192.168.31.174	224.0.0.50	UDP	153	4321 → 9898 Len=111

Write command wia ssh service domoticz restart.....10sec.....service domoticz stop...10sec....service domoticz start/

dwmw2, 2020-06-14:
To show the content of the packets try adding `-x` to the tcpdump command line.

Or `-w /tmp/capture.cap` and send us the resulting capture.cap file.

Kheamu, 2020-06-14:
Use  tcpdump host 192.168.31.174 -x -w /tmp/capture.cap.
Then "service domoticz restart"...20sec..."service domoticz stop"...20sec..."service domoticz start".

1	0.000000	192.168.31.174	224.0.0.50	UDP	140	4321 → 9898 Len=98
2	4.403487	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
3	14.337034	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
4	19.077875	ASUSTekC_56:06:38	Broadcast	ARP	42	Who has 192.168.31.174? Tell 192.168.31.220
5	19.179311	ASUSTekC_56:06:38	Broadcast	ARP	42	Gratuitous ARP for 192.168.31.174 (Request)
6	19.179409	ASUSTekC_56:06:38	Broadcast	ARP	42	Gratuitous ARP for 192.168.31.174 (Reply)
7	24.370151	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
8	34.302260	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
9	36.455662	192.168.31.174	224.0.0.50	UDP	154	4321 → 9898 Len=112
10	36.457234	192.168.31.174	224.0.0.50	UDP	151	4321 → 9898 Len=109
11	36.458833	192.168.31.174	224.0.0.50	UDP	153	4321 → 9898 Len=111
12	43.213343	192.168.31.174	224.0.0.50	UDP	144	4321 → 9898 Len=102
13	43.214544	192.168.31.174	224.0.0.50	UDP	162	4321 → 9898 Len=120
14	44.337093	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
15	49.189171	ASUSTekC_56:06:38	Broadcast	ARP	42	Who has 192.168.31.174? Tell 192.168.31.220
16	49.286667	ASUSTekC_56:06:38	Broadcast	ARP	42	Gratuitous ARP for 192.168.31.174 (Request)
17	49.288135	ASUSTekC_56:06:38	Broadcast	ARP	42	Gratuitous ARP for 192.168.31.174 (Reply)
18	53.247994	192.168.31.174	224.0.0.50	UDP	156	4321 → 9898 Len=114
19	54.173677	192.168.31.174	224.0.0.50	UDP	178	4321 → 9898 Len=136
20	54.273422	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
21	64.202708	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
22	74.238084	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137
23	77.002832	192.168.31.174	224.0.0.50	UDP	154	4321 → 9898 Len=112
24	77.003259	192.168.31.174	224.0.0.50	UDP	151	4321 → 9898 Len=109
25	77.109135	192.168.31.174	224.0.0.50	UDP	153	4321 → 9898 Len=111
26	79.289034	ASUSTekC_56:06:38	Broadcast	ARP	42	Who has 192.168.31.174? Tell 192.168.31.220
27	79.392494	ASUSTekC_56:06:38	Broadcast	ARP	42	Gratuitous ARP for 192.168.31.174 (Request)
28	79.393412	ASUSTekC_56:06:38	Broadcast	ARP	42	Gratuitous ARP for 192.168.31.174 (Reply)
29	84.174154	192.168.31.174	224.0.0.50	UDP	179	4321 → 9898 Len=137

Where 31.174 - Xiaomi gateway
           31.220-router with openwrt and domoticz.

Kheamu, 2020-06-14:
How may i attach file capture.cap? It contains much more information.

Kheamu, 2020-06-14:
[captureMI.txt](https://github.com/openwrt/packages/files/4776581/captureMI.txt)

Kheamu, 2020-06-14:
[capture.zip](https://github.com/openwrt/packages/files/4776589/capture.zip)

dwmw2, 2020-06-14:
There's only 1½ seconds of capture there. Can you capture the full time that Domoticz starts up until it sees (or fails to see) the offending device?

Kheamu, 2020-06-15:
There is about 3 minutes of my active attemts to add switches in domoticz and restarting it
[capture.zip](https://github.com/openwrt/packages/files/4778137/capture.zip)

Kheamu, 2020-06-16:
I did something wrong?