openwrt/packages #12262: perl: host compile failed with segmentation fault in 19.07

dimich-dmb, 2020-05-23:
Environment: Arch Linux x86_64, OpenWRT-19.07, last commit: 6a8909893553eba2c24b2dc8b99abddbd0188a94
feeds.conf:
```
src-git packages https://git.openwrt.org/feed/packages.git;openwrt-19.07
```
feeds/packages last commit: a41e068db81d7ecc6d0c0d57f62ddfd49acd68d9

Perl 5.28.1 is disabled in target config:
```
$ grep CONFIG_PACKAGE_perl .config
# CONFIG_PACKAGE_perl is not set
```
but tries to compile for host and fails with segmentation fault:
```
make[3]: Entering directory '/home/dimich/work/musicbox/openwrt/feeds/packages/lang/perl'
CFLAGS="-O2 -I/home/dimich/work/musicbox/openwrt/staging_dir/host/include -I/home/dimich/work/musicbox/openwrt/staging_dir/hostpkg/include -I/home/dimich/work/musicbox/openwrt/staging_dir/target-mipsel_24kc_musl/host/include" CPPFLAGS="-I/home/dimich/work/musicbox/openwrt/staging_dir/host/include -I/home/dimich/work/musicbox/openwrt/staging_dir/hostpkg/include -I/home/dimich/work/musicbox/openwrt/staging_dir/target-mipsel_24kc_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/dimich/work/musicbox/openwrt/staging_dir/host/lib -L/home/dimich/work/musicbox/openwrt/staging_dir/hostpkg/lib -L/home/dimich/work/musicbox/openwrt/staging_dir/target-mipsel_24kc_musl/host/lib" make  -C /home/dimich/work/musicbox/openwrt/build_dir/hostpkg/perl/perl-5.28.1  
make[4]: Entering directory '/home/dimich/work/musicbox/openwrt/build_dir/hostpkg/perl/perl-5.28.1'
makefile:258: warning: ignoring prerequisites on suffix rule definition
makefile:254: warning: ignoring prerequisites on suffix rule definition
makefile:258: warning: ignoring prerequisites on suffix rule definition
makefile:258: warning: ignoring prerequisites on suffix rule definition
makefile:254: warning: ignoring prerequisites on suffix rule definition
makefile:258: warning: ignoring prerequisites on suffix rule definition
cc -fstack-protector-strong -L/usr/local/lib -o miniperl \
    opmini.o perlmini.o  gv.o toke.o perly.o pad.o regcomp.o dump.o util.o mg.o reentr.o mro_core.o keywords.o hv.o av.o run.o pp_hot.o sv.o pp.o scope.o pp_ctl.o pp_sys.o doop.o doio.o regexec.o utf8.o taint.o deb.o universal.o globals.o perlio.o perlapi.o numeric.o mathoms.o locale.o pp_pack.o pp_sort.o caretx.o dquote.o time64.o  miniperlmain.o  -lpthread -lnsl -lgdbm -ldb -ldl -lm -lcrypt -lutil -lc -lgdbm_compat 
./miniperl -w -Ilib -I. -Idist/Exporter/lib -MExporter -e '<?>' || sh -c 'echo >&2 Failed to build miniperl.  Please run make minitest; exit 1'
Attempt to free unreferenced scalar: SV 0x56318228e5b8.
/bin/sh: line 1: 35143 Segmentation fault      (core dumped) ./miniperl -w -Ilib -I. -Idist/Exporter/lib -MExporter -e '<?>'
Failed to build miniperl. Please run make minitest
make[4]: *** [makefile:365: lib/buildcustomize.pl] Error 1
make[4]: Leaving directory '/home/dimich/work/musicbox/openwrt/build_dir/hostpkg/perl/perl-5.28.1'
make[3]: *** [Makefile:149: /home/dimich/work/musicbox/openwrt/build_dir/hostpkg/perl/perl-5.28.1/.built] Error 2
make[3]: Leaving directory '/home/dimich/work/musicbox/openwrt/feeds/packages/lang/perl'
time: package/feeds/packages/perl/host-compile#0.32#0.08#0.39
make[2]: *** [package/Makefile:111: package/feeds/packages/perl/host/compile] Error 2
make[2]: Leaving directory '/home/dimich/work/musicbox/openwrt'
make[1]: *** [package/Makefile:107: /home/dimich/work/musicbox/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dimich/work/musicbox/openwrt'
make: *** [/home/dimich/work/musicbox/openwrt/include/toplevel.mk:227: world] Error 2
```
If package/feeds/packages/perl symlink is removed manually, then build succeed.

lynxis, 2020-05-23:
@dimich-dmb perl might be still a build depends of an other package. 

Can you supply your config?

dimich-dmb, 2020-05-23:
> Can you supply your config?

Sure. Attached .config
[config.txt](https://github.com/openwrt/packages/files/4672357/config.txt)

Legogris, 2020-05-24:
Same behavior here.

```
$ cat .config | grep -i perl
# Perl
# CONFIG_PACKAGE_perl is not set
# CONFIG_PACKAGE_python-hyperlink is not set
# CONFIG_PACKAGE_python-hyperlink-src is not set
# CONFIG_PACKAGE_python3-hyperlink is not set
# CONFIG_PACKAGE_python3-hyperlink-src is not set
```

The perl make messages start appearing right after `libtool` (running with `-j1`), so maybe that's a hint.

Selecting perl for inclusion in menuconfig seems to resolve this.

xinxijishuwyq, 2020-05-26:
#12178
try it

dimich-dmb, 2020-05-26:
> #12178
> try it
This patch works for me. Thanks.

xnoreq, 2020-05-27:
Why is perl being built in the first place?

dimich-dmb, 2020-05-28:
> Why is perl being built in the first place?

I found there is dependency in samba4 package:
`HOST_BUILD_DEPENDS:=python3/host perl/host`
And samba4 is enabled in my config.

xnoreq, 2020-05-28:
I have neither any perl packages nor any samba packages enabled and it still builds perl.

arnej, 2020-06-07:
I get the same error.

I manually fetched the patch from [12391](https://github.com/openwrt/packages/pull/12391), tried again an it fixed the issue for me.

I think the patch should be backported to 19.07.

lipnitsk, 2020-06-13:
Opened PR #12509 with the backport

neheb, 2020-06-14:
Merged.

closed 2020-06-14 by neheb