openwrt/packages #12266: Python 3.8+ update

Mikami382, 2020-05-23:
Hey there! When would there be a update to Python 3.8+?

neheb, 2020-05-24:
For 19.07?

Mikami382, 2020-05-24:
Yes, OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.057.55219-13dd17f to be precise.

neheb, 2020-05-24:
I don't think there is such a plan. ping @jefferyto

jefferyto, 2020-05-24:
That's right - as a stable release, 19.07 will receive bug and security fixes only. Updating past 3.7 would be too big a jump.

@neheb Is there a page on the wiki that describes the support/release policy for releases? I don't remember seeing a page like that but it may help to have one.

neheb, 2020-05-24:
I don’t think there’s an explicit policy. I think the goal is to have something like Debian or Ubuntu where a release stays with old packages that only get bug fixes.

jefferyto, 2020-05-25:
I think whether the policy is strict or flexible can be discussed and decided by project members/contributors/etc., but I also think having something written down is good so that everyone (developers and users) know what to expect.

neheb, 2020-05-31:
I personally would like to see the packages feed transition to rolling release, but that's beyond the scope of this Issue.

Bottom line: Python 3.8 will not be coming to 19.07.

closed 2020-05-31 by neheb