openwrt/packages #12267: make: compile failure on x86_64

dl12345, 2020-05-23:
```
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/9.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: glob/libglob.a(glob.o): in function `glob_in_dir':
glob.c:(.text+0x2b3): undefined reference to `__alloca'
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/9.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: glob.c:(.text+0x3f5): undefined reference to `__alloca'
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/9.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: glob.c:(.text+0x595): undefined reference to `__alloca'
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/9.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: glob.c:(.text+0x629): undefined reference to `__alloca'
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/9.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: glob/libglob.a(glob.o): in function `glob':
glob.c:(.text+0xad8): undefined reference to `__alloca'
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/9.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: glob/libglob.a(glob.o):glob.c:(.text+0x11a2): more undefined references to `__alloca' follow
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:656: make] Error 1
```    

Can't find the github handle of the maintainer listed in the package Makefile

A simple bump to version 4.3 fixes the compile failure.

neheb, 2020-05-24:
I had a PR that updated make to this version. Was rejected on the basis that some old packages might break.

I should probably bring it back...

dl12345, 2020-05-24:
I've got a patch handy too if you've deleted the code...

neheb, 2020-05-24:
https://github.com/openwrt/packages/pull/11343

pprindeville, 2020-05-25:
> I had a PR that updated make to this version. Was rejected on the basis that some old packages might break.

That's a specious argument that keeps getting trotted out. Any sort of progress risks breaking something: should we not have progress?

neheb, 2020-05-25:
¯\_(ツ)_/¯

closed 2020-05-28 by neheb