openwrt/packages #12273: nginx-util failing with clang-10

neheb, 2020-05-24:
Maintainer: @peter-stadler 
Environment: host build - clang-10 - libcxx-10

```
In file included from /home/mangix/devstuff/packages/net/nginx-util/src/nginx-util.cpp:4:
/home/mangix/devstuff/packages/net/nginx-util/src/nginx-ssl-util.hpp:150:16: error: no previous extern declaration for non-static variable '_capture<'\x00'>' [-Werror,-Wmissing-variable-declarations]
constexpr auto _capture = _Line{
               ^
/home/mangix/devstuff/packages/net/nginx-util/src/nginx-ssl-util.hpp:215:9: note: in instantiation of variable template specialization '_capture' requested here
        _capture<>, _newline >();
        ^
/home/mangix/devstuff/packages/net/nginx-util/src/nginx-ssl-util.hpp:150:11: note: declare 'static' if the variable is not intended to be used outside of this translation unit
constexpr auto _capture = _Line{
          ^
/home/mangix/devstuff/packages/net/nginx-util/src/nginx-ssl-util.hpp:150:16: error: no previous extern declaration for non-static variable '_capture<';'>' [-Werror,-Wmissing-variable-declarations]
constexpr auto _capture = _Line{
               ^
/home/mangix/devstuff/packages/net/nginx-util/src/nginx-ssl-util.hpp:218:56: note: in instantiation of variable template specialization '_capture' requested here
    Line::build<_begin, _escape<_server_name>, _space, _capture<';'>, _end>();
                                                       ^
/home/mangix/devstuff/packages/net/nginx-util/src/nginx-ssl-util.hpp:150:11: note: declare 'static' if the variable is not intended to be used outside of this translation unit
constexpr auto _capture = _Line{
          ^
/home/mangix/devstuff/packages/net/nginx-util/src/nginx-util.cpp:124:17: error: 'basic_string_view' may not intend to support class template argument deduction [-Werror,-Wctad-maybe-unsupported]
    auto args = std::basic_string_view{argv, static_cast<size_t>(argc)};
                ^
/usr/bin/../lib/gcc/x86_64-redhat-linux/10/../../../../include/c++/10/string_view:94:11: note: add a deduction guide to suppress this warning
    class basic_string_view
          ^
3 errors generated.
```

First two are easily fixed with static. No idea about the third.

neheb, 2020-05-24:
px5g-openssl.hpp is also missing an stdexcept include.

peter-stadler, 2020-05-24:
Thank you very much :-) I will look into it …

peter-stadler, 2020-05-26:
fixed these issues together with the upcoming fixes in the RFC …

for the 3. error: using `basic_string_view<char *>` solves it. Actually, `basic_string_view` is a placeholder for `span`, which would have a similar functionality but a better name (argv is actually an array of strings not a view on one "string" with base type `char *` instead of `char`).

neheb, 2020-05-26:
I use -Weverything and add -Wno- for all the pointless ones.

peter-stadler, 2020-05-26:
I see, thank you :-)

closed 2020-05-26 by neheb