openwrt/packages #12278: vpnbypass

muhammad-naderi, 2020-05-24:
Hello, 

Great job here, thanks!

Just wanted to know is there a way to pass a list of ip/ranges to /etc/config/vpnbypass to get bypassed from another file.
reason is I need to bypass like 11M IP addresses (local country ips, downloaded from https://www.nirsoft.net/countryip/ir.html) and leave all my international ips to go through vpn

thanks in advance

jamesmacwhite, 2020-06-11:
Potentially you would be looking at something like ipset to do this, you'd need to be able to ingest the IP addresses (probably best as ranges) and then load them into an ipset chain as `hash:net` which you can then use to conditionally route. However you'd have to be careful with the memory for the amount of ranges.

stangri, 2025-11-18:
@muhammad-naderi you need to migrate to the current release of OpenWrt and use `pbr` package instead, it can read the content of local or remote file for entries.