openwrt/packages #12291: wolfssl not compiling under multiple platforms

neheb, 2020-05-25:
ping @cotequeiroz I'm guessing the wolfssl updates broke this:

https://downloads.openwrt.org/snapshots/faillogs/i386_pentium/packages/wifidog/tls/compile.txt

Maintainer: @mhass
Environment: multiple

cotequeiroz, 2020-05-25:
I'll start looking at it when able--hopefully in a couple of hours.  Do you have a list of platforms other than i386--I'll start with it--that I should look into?

neheb, 2020-05-25:
Hrm it seems it's just i386. Edited the description.

cotequeiroz, 2020-05-25:
Wifidog is failing because wolfssl itself is failing.
See: https://downloads.openwrt.org/snapshots/faillogs/i386_pentium/base/wolfssl/compile.txt
We can certainly leave @mhaas out of this, and possibly close the issue here.

neheb, 2020-05-25:
OK

closed 2020-05-25 by neheb

cotequeiroz, 2020-05-26:
Just for the record. WolfSSL needs to be compiled with -fomit-frame-pointer, otherwise the frame pointer will occupy a register, and that error will show up. It’s been that way for a long time. CFLAGS perhaps have changed. I’ll include it in TARGET_CFLAGS and it’ll build.