openwrt/packages #12307: elektra not compiling with gcc10

neheb, 2020-05-25:
```
[ 98%] Linking C executable ../bin/benchmark_large
/usr/bin/ld: CMakeFiles/benchmark_large.dir/large.c.o:(.bss+0x0): multiple definition of `large'; CMakeFiles/benchmark_large.dir/benchmarks.c.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
```

This is the host build. Tested with:

```
make package/elektra/host/compile V=s
```

Maintainer: @haraldg
Environment: fedora 32, gcc10

haraldg, 2020-05-26:
Thanks, I will have a look.

haraldg, 2020-05-27:
@markus2330 Is this issue fixed in any released version?

(We probably should disable building benchmarks, but might be a nice excuse to update the package anyway.)

mpranj, 2020-05-27:
@haraldg we have fixed this problem, and the fix is released with [elektra 0.9.2](https://www.libelektra.org/news/0.9.2-release).

PR #12415 (2020-06-08): elektra: update to 0.9.2
Notable changes:
* Move some plugins from libelektra-plugins to libelektra-cpp because
  they got reimplemented in C++.
* New package libelektra-zmq
* Move the crypto plugin from libopenssl to libgcrypt
* Disable host build: elektra doesn't need it anymore and nobody uses
  kdb during package builds for now.

Closes: #12307

Maintainer: me
Compile tested: openwrt-sdk-mxs_gcc-8.4.0_musl_eabi.Linux-x86_64 from 2020-05-28
Run tested: none
commit 82e34754 (2020-06-08): elektra: update to 0.9.2

Notable changes:
* Move some plugins from libelektra-plugins to libelektra-cpp because
  they got reimplemented in C++.
* New package libelektra-zmq
* Move the crypto plugin from libopenssl to libgcrypt
* Disable host build: elektra doesn't need it anymore and nobody uses
  kdb during package builds for now.

Closes: #12307
closed 2020-06-08 by neheb