openwrt/packages #12317: Transmission is not enabled by default

BKPepe, 2020-05-26:
Maintainer: @neheb 
Environment: all OpenWrt versions

Description:

If I had enabled seccomp and it that case, there was a missing syscalls (will be added later), it is such pain to debug what's going on when you are trying to run transmission via `procd`, because in logs there isn't anything and transmission does not start. I tried to disable seccomp and transmission still does not start, however, it started when I run it manually!

This is because of this row https://github.com/openwrt/packages/blob/master/net/transmission/files/transmission.config#L2 , nobody touched that for 6 years.

**If you install any software, it should start by default no matter how**. If you are not satisfied with defaults, you might want to change it in UI (e.g. transmission-web), but when it does not start, you can not do it.

At least it should write to syslog something like this: `If you are trying to start transmission for the first time, you need to edit /etc/config/transmission`

neheb, 2020-05-26:
I do that for gerbera. Would makes sense to do the same here.

I don’t like starting by default. It defaults to using /tmp, which can potentially cause OOM.

BKPepe, 2020-05-26:
Yes, it can lead to OOM if you didn't setup it correctly, but it will be the user's fault that he started downloading a file to /tmp. The transmission in the first place should start.

neheb, 2020-05-26:
OTOH, there is already precedent in other packages that use an enabled variable:

```
~/d/packages (master)> git grep option\ enabled | wc -l
96
```

closed 2020-06-16 by neheb