openwrt/packages #12320: ffmpeg: libffmpeg-custom not handled properly

pablocool, 2020-05-26:
Maintainer: @BKPepe 
Environment: (ar71xx, wdr3600-v1, openwrt-19.07)

Description:
During libffmpeg-custom preparation with only x264 decoder enabled I faced issue
```
cp -fpR /home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-install/usr/bin/ffmpeg /home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-mips_24kc/ffmpeg-custom/usr/bin/
cp: cannot stat '/home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-install/usr/bin/ffmpeg': No such file or directory
make[3]: *** [Makefile:800: /home/pablocool/openwrt/bin/packages/mips_24kc/packages/ffmpeg-custom_3.4.7-3_mips_24kc.ipk] Error 1
make[3]: Leaving directory '/home/pablocool/openwrt/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/custom/compile#0.19#0.12#0.27
make[2]: *** [package/Makefile:113: package/feeds/packages/ffmpeg/compile] Error 2
make[2]: Leaving directory '/home/pablocool/openwrt'
make[1]: *** [package/Makefile:107: /home/pablocool/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pablocool/openwrt'
make: *** [/home/pablocool/openwrt/include/toplevel.mk:227: world] Błąd 2
```
Used config:
```
pablocool@PC ~/openwrt $ grep -i ffmpeg .config | grep -v \#
CONFIG_PACKAGE_libffmpeg-custom=y
CONFIG_FFMPEG_CUSTOM_GPL=y
CONFIG_FFMPEG_CUSTOM_PATENTED=y
CONFIG_FFMPEG_CUSTOM_DECODER_h264=y
CONFIG_PACKAGE_ffmpeg-custom=y
```

After few hours of digging with help of other openwrt developer urned out that build procedure was trying to copy not existing binary of ffmpeg, because of missing:
```
CONFIG_FFMPEG_CUSTOM_PROGRAMS=y
``` 
What passed --disable-programs to configure.

After adding above option I faced:
```
rstrip.sh: /home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-mips_24kc/libffmpeg-custom/usr/lib/libavcodec.so.57.107.100: shared object
rstrip.sh: /home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-mips_24kc/libffmpeg-custom/usr/lib/libavfilter.so.6.107.100: shared object
rstrip.sh: /home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-mips_24kc/libffmpeg-custom/usr/lib/libavformat.so.57.83.100: shared object
rstrip.sh: /home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-mips_24kc/libffmpeg-custom/usr/lib/libavdevice.so.57.10.100: shared object
rstrip.sh: /home/pablocool/openwrt/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-3.4.7/ipkg-mips_24kc/libffmpeg-custom/usr/lib/libswresample.so.2.9.100: shared object
```
So added:
```
CONFIG_FFMPEG_CUSTOM_AUDIO_DEC_SUPPORT=y
```

When trying to fix this issue I met several posts regarding this problem. People were "solving" this by using libffmpeg-full. Maybe it would be possible to fix it in repo?

pablocool, 2020-05-26:
One thing worth to mention. After adding
`CONFIG_FFMPEG_CUSTOM_AUDIO_DEC_SUPPORT=y
`
.config looks as follows:
```
pablocool@PC ~/openwrt $ grep -i ffmpeg .config | grep -v \#
CONFIG_PACKAGE_libffmpeg-custom=y
CONFIG_FFMPEG_CUSTOM_GPL=y
CONFIG_FFMPEG_CUSTOM_PATENTED=y
CONFIG_FFMPEG_CUSTOM_AUDIO_DEC_SUPPORT=y
CONFIG_FFMPEG_CUSTOM_PROGRAMS=y
CONFIG_FFMPEG_CUSTOM_DECODER_aac=y
CONFIG_FFMPEG_CUSTOM_SELECT_adpcm=y
CONFIG_FFMPEG_CUSTOM_DECODER_ac3=y
CONFIG_FFMPEG_CUSTOM_DECODER_alac=y
CONFIG_FFMPEG_CUSTOM_DECODER_amrnb=y
CONFIG_FFMPEG_CUSTOM_DECODER_amrwb=y
CONFIG_FFMPEG_CUSTOM_DECODER_ape=y
CONFIG_FFMPEG_CUSTOM_DECODER_atrac3=y
CONFIG_FFMPEG_CUSTOM_DECODER_flac=y
CONFIG_FFMPEG_CUSTOM_DECODER_h264=y
CONFIG_FFMPEG_CUSTOM_DECODER_mp2=y
CONFIG_FFMPEG_CUSTOM_DECODER_mp3=y
CONFIG_FFMPEG_CUSTOM_DECODER_mpc7=y
CONFIG_FFMPEG_CUSTOM_DECODER_mpc8=y
CONFIG_FFMPEG_CUSTOM_DECODER_opus=y
CONFIG_FFMPEG_CUSTOM_DECODER_pcm_s16be=y
CONFIG_FFMPEG_CUSTOM_DECODER_pcm_s16le=y
CONFIG_FFMPEG_CUSTOM_DECODER_vorbis=y
CONFIG_FFMPEG_CUSTOM_DECODER_wavpack=y
CONFIG_FFMPEG_CUSTOM_DECODER_wmav1=y
CONFIG_FFMPEG_CUSTOM_DECODER_wmav2=y
CONFIG_FFMPEG_CUSTOM_DECODER_zlib=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_aac=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_avi=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ac3=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_amr=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ape=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_flac=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ffm=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_matroska=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mov=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mp3=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mpegts=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mpc=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mpc8=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ogg=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_rm=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_rtsp=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_sdp=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_rtp=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_wav=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_wv=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_file=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_http=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_rtp=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_tcp=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_udp=y
CONFIG_PACKAGE_ffmpeg-custom=y
```

However after deleting this option from menuconfig. .config looks as follows:

```
pablocool@PC ~/openwrt $ grep -i ffmpeg .config | grep -v \#
CONFIG_PACKAGE_libffmpeg-custom=y
CONFIG_FFMPEG_CUSTOM_GPL=y
CONFIG_FFMPEG_CUSTOM_PATENTED=y
CONFIG_FFMPEG_CUSTOM_PROGRAMS=y
CONFIG_FFMPEG_CUSTOM_DECODER_aac=y
CONFIG_FFMPEG_CUSTOM_SELECT_adpcm=y
CONFIG_FFMPEG_CUSTOM_DECODER_ac3=y
CONFIG_FFMPEG_CUSTOM_DECODER_alac=y
CONFIG_FFMPEG_CUSTOM_DECODER_amrnb=y
CONFIG_FFMPEG_CUSTOM_DECODER_amrwb=y
CONFIG_FFMPEG_CUSTOM_DECODER_ape=y
CONFIG_FFMPEG_CUSTOM_DECODER_atrac3=y
CONFIG_FFMPEG_CUSTOM_DECODER_flac=y
CONFIG_FFMPEG_CUSTOM_DECODER_h264=y
CONFIG_FFMPEG_CUSTOM_DECODER_mp2=y
CONFIG_FFMPEG_CUSTOM_DECODER_mp3=y
CONFIG_FFMPEG_CUSTOM_DECODER_mpc7=y
CONFIG_FFMPEG_CUSTOM_DECODER_mpc8=y
CONFIG_FFMPEG_CUSTOM_DECODER_opus=y
CONFIG_FFMPEG_CUSTOM_DECODER_pcm_s16be=y
CONFIG_FFMPEG_CUSTOM_DECODER_pcm_s16le=y
CONFIG_FFMPEG_CUSTOM_DECODER_vorbis=y
CONFIG_FFMPEG_CUSTOM_DECODER_wavpack=y
CONFIG_FFMPEG_CUSTOM_DECODER_wmav1=y
CONFIG_FFMPEG_CUSTOM_DECODER_wmav2=y
CONFIG_FFMPEG_CUSTOM_DECODER_zlib=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_aac=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_avi=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ac3=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_amr=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ape=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_flac=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ffm=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_matroska=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mov=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mp3=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mpegts=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mpc=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_mpc8=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_ogg=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_rm=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_rtsp=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_sdp=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_rtp=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_wav=y
CONFIG_FFMPEG_CUSTOM_DEMUXER_wv=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_file=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_http=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_rtp=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_tcp=y
CONFIG_FFMPEG_CUSTOM_PROTOCOL_udp=y
CONFIG_PACKAGE_ffmpeg-custom=y
```

So looks also removing this option from menuconfig does not work correctly and leave stuff in .config

BKPepe, 2020-05-26:
I am not maintainer of ffmpeg package.
https://github.com/openwrt/packages/blob/master/multimedia/ffmpeg/Makefile#L18

Maintainers are @antonlacon and @thess

pablocool, 2020-05-26:
Sorry my mistake. I checked git log for Makefile while should look inside it. Thanks. I updated ticket.

neheb, 2020-05-27:
Best course of action is to post a PR :).

diizzyy, 2020-05-27:
This is one of many reasons why I suggested to "unify" ffmpeg options.

PR #22276 (2023-10-05): ffmpeg: fix compilation error of ffmpeg-custom
Fixes: #12320

Maintainer: @graysky2
Compile tested: default arch, v22.03.5
Run tested: openwrt-22.03.5-ramips-mt7621-zbtlink_zbt-wg3526-32m
Description:

After plain enable of libffmpeg-custom and ffmpeg-custom which adds:
- CONFIG_PACKAGE_libbz2=m
- CONFIG_PACKAGE_libffmpeg-custom=m
- CONFIG_FFMPEG_CUSTOM_GPL=y
- CONFIG_PACKAGE_zlib=m
- CONFIG_PACKAGE_ffmpeg-custom=m

I get on v22.03.5 compilation error:
```
cp -fpR /.../openwrt-git/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-5.1/ipkg-install/usr/bin/ffmpeg /.../openwrt-git/build_dir/ta [... cut]
commit e429c11b (2023-10-05): ffmpeg: fix compilation error of ffmpeg-custom

Fixes: #12320

After plain enable of libffmpeg-custom and ffmpeg-custom which adds:
- CONFIG_PACKAGE_libbz2=m
- CONFIG_PACKAGE_libffmpeg-custom=m
- CONFIG_FFMPEG_CUSTOM_GPL=y
- CONFIG_PACKAGE_zlib=m
- CONFIG_PACKAGE_ffmpeg-custom=m

I get on v22.03.5 compilation error:
```
cp -fpR /.../openwrt-git/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-5.1/ipkg-install/usr/bin/ffmpeg /.../openwrt-git/build_dir/target-mips_24kc_musl/ffmpeg-custom/ffmpeg-5.1/ipkg-mips_24kc/ffmpeg-custom/usr/bin/
cp: cannot stat '/.../openwrt-git/build_dir/target-mips_ [... cut]
closed 2023-10-05 by neheb