openwrt/packages #12332: [19.07]python-dateutil not compiling sometimes

neheb, 2020-05-26:
Maintainer: @commodo 
Environment: various

https://downloads.openwrt.org/releases/faillogs-19.07/arc_arc700/packages/python-dateutil/python/compile.txt
https://downloads.openwrt.org/releases/faillogs-19.07/arm_cortex-a8_neon/packages/python-dateutil/python/compile.txt
https://downloads.openwrt.org/releases/faillogs-19.07/arm_cortex-a8_vfpv3/packages/python-dateutil/python/compile.txt
https://downloads.openwrt.org/releases/faillogs-19.07/arm_cortex-a9_vfpv3-d16/packages/python-dateutil/python/compile.txt
https://downloads.openwrt.org/releases/faillogs-19.07/arm_mpcore/packages/python-dateutil/python/compile.txt
https://downloads.openwrt.org/releases/faillogs-19.07/mipsel_74kc/packages/python-dateutil/python/compile.txt

It's not everywhere. Maybe a missing build dependency.

commodo, 2020-05-27:
so, those links are 404 now;
i did see some of these weird failures from time to time;
but i can't understand them; and they are sporadic;

i found one just now on https://downloads.openwrt.org/releases/faillogs-19.07/arm_arm926ej-s/packages/python-dateutil/python3/  and it went away after a few minutes;

i check the dependencies;
the only thing left is to remove the conditional dependency stuff
other than that i don't know;

commodo, 2020-05-27:
> the only thing left is to remove the conditional dependency stuff
> other than that i don't know;

wait; that can only be done on master; on 19.07 we can't; there's still python2

neheb, 2020-05-27:
Ehhh I’ll close this. I also see mpd failures relating to its AUDIO_SUPPORT dependency.

closed 2020-05-27 by neheb

commodo, 2020-05-27:
these sporadic failures aren't new;
they are a few years old, but i could never make sense from the logs, as the messages didn't contain any errors