openwrt/packages #12353: docker-ce: issues when compiling with glibc

dl12345, 2020-05-29:
I can't get docker-ce and tini to compile with glibc unless I add the following to tini's Makefile

```
TARGET_LDFLAGS += $(if $(CONFIG_USE_GLIBC),-static -lc -lgcc_eh,)
```
and the following to docker-ce's Makefile

```
TARGET_LDFLAGS += $(if $(CONFIG_USE_GLIBC),-static -lpthread -lc -lgcc_eh,)
```

Failure to do this results in the following errors (many of them) for both tini and docker-ce

```
/opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/gcc/x86_64-openwrt-linux-gnu/9.3.0/../../../../x86_64-openwrt-linux-gnu/bin/ld: /opt/openwrt/staging_dir/toolchain-x86_64_gcc-9.3.0_glibc/lib/libc.a(iofclose.o): in function `_IO_new_fclose.cold':
iofclose.c:(.text.unlikely+0x37): undefined reference to `_Unwind_Resume'
```

I surmise this has something to do with the -fpic flag passed to the compiler coupled with the static linking being done, since the -static flag overrides the -fpic. Not sure that the flags above are the optimal ones to use though - there may be a more efficient way of doing it, since the tini binary that is produced is anything but tini

neheb, 2020-05-29:
Related: https://github.com/openwrt/packages/issues/11840

neheb, 2020-05-31:
Can you make a PR?

dl12345, 2020-05-31:
> Can you make a PR?

Sure

dl12345, 2020-05-31:
@neheb : I was fiddling with the flags to try and reduce the size of the statically linked binary. I can get about a 10% reduction in size by changing the TARGET_CFLAGS thus:

```
TARGET_CFLAGS +=  -ffunction-sections -fdata-sections -Wl,--build-id -Wl,--gc-sections,--as-needed
```

I actually went as far as compiling the entire libc with `-ffunction-sections -fdata-sections` as well, but it did not make any further improvement.

Worthwhile modification? I doubt anyone using docker-ce is going to be running it on a resource constrained system where a 10% drop in size of a 700k binary is going to make much of a difference...

neheb, 2020-05-31:
I don't see why not.

G-M0N3Y-2503, 2020-06-01:
Interesting.  

If I recall correctly, I was able to get the dynamically linked version to work, but it would only work with a limited number of containers. I think this was because docker essentially copies in `docker-init` into the container which meant the containers required musl to run with `--init`. So for the best compatibility, I went with static linking.

So it should be possible to add an option for using a dynamically linked version too if that better suits your needs.

dl12345, 2020-06-01:
I think statically linking is probably the right thing to do in that case. The two lines I added cause the static linking to work properly for glibc, so there's no reason to use a potentially broken dynamic link.

dl12345, 2020-06-01:
@G-M0N3Y-2503 do you want to do the PR, or you happy for me to make on?

G-M0N3Y-2503, 2020-06-04:
I haven't tested a glibc build before and my dev environment is a bit messed up right now, so go for it.

dl12345, 2020-06-04:
Ok, I'm just trying to get to the bottom of the reboot issue when I run docker then once I've resolved it I'll do a PR

neheb, 2020-07-23:
@dl12345 any progress on this?

dl12345, 2020-07-25:
@neheb I didn't bother to do this as docker-ce is pretty much unusable on x86_64 at the moment and I can't imagine that many people would be building with glibc on any other platform. 

Docker-ce crashes my router pretty much every time I use it. I suspect the problem is very likely this one that @jmarcet [reported on x64_64](https://bugzilla.kernel.org/show_bug.cgi?id=207403) 

I can make a PR for this if you still want it?

neheb, 2020-07-25:
I would like to close this issue if possible.

dl12345, 2020-07-25:
Ok..I'll look at making a PR tomorrow

neheb, 2020-09-19:
What happened?

jmarcet, 2020-09-19:
> @neheb I didn't bother to do this as docker-ce is pretty much unusable on x86_64 at the moment and I can't imagine that many people would be building with glibc on any other platform.
>
> Docker-ce [...]

docker has its issues on x86_64 indeed on openwrt, but here are a few workarounds I use to have a 100% stable station.

I haven't been able to solve the issue around netns_clone, but other than when rebooting, or updating containers, I manage to have a system with no backtrace at all, running 24/7 with 26 containers. If anyone wants to dig further on the problem, I am quite sure none of the openwrt patches over the mainline kernel have to do with it, although it looks crearly like a kernel problem an it doesn't happen on other distros.

Of course a proper fix is needed, but until then I can share what I do if anyone is interested. It revolves around stopping properly all the containers before shutdown, then starting then again after booting. That way I keep my system working 100% stable.

closed 2021-01-08 by neheb