openwrt/packages #12354: pdns/pdns-recursor: best practice for generating sample config

James-TR, 2020-05-29:
Maintainer: me
Environment: Any

Description:
Currently I'm manually generating sample config for pdns/pdns-recursor on another of my installations. Normally, this gets generated by calling the compiled binary and had to be patched out since this isn't compatible with a cross-compiler.

I was looking into postinst hooks but not sure if that's the correct place for it or not.

The command that used to be run by the build is below.
```
./pdns_server --no-config --config 2>/dev/null > $wherever
```

Thanks.

neheb, 2020-05-31:
With gerbera I generate it at runtime: https://github.com/openwrt/packages/blob/master/multimedia/gerbera/files/gerbera.init

Not ideal but works.

James-TR, 2020-06-01:
I like that idea. I'm thinking of combining with a post-install hook so that it's generated on first install and updated with current version whenever they upgrade.

closed 2020-06-02 by James-TR