openwrt/packages #12358: docker-ce: firewall rules not flushed on "/etc/init.d/docker stop"

dl12345, 2020-05-29:
The title of the issue says it all. Probably makes sense to do a `fw3 restart` when the service is stopped. 

@G-M0N3Y-2503

feckert, 2020-06-23:
This raises the question of who is now responsible for this to clean up the mess

- dockerd binary ?
or
- dockerd init script ?

I would expect the this is the task for dockerd binary. But the way I see it, that's not the case. Then we have no choice but to do this in the [init script](https://github.com/openwrt/packages/blob/master/utils/docker-ce/files/dockerd.init) 

I think the whole fw3  firewall restart is dangerous.

feckert, 2020-06-24:
I have updated the dockerd init script in this pullrequest [commit](https://github.com/openwrt/packages/pull/12556/commits/2c33d39a5e0f69d6de295ea7aa279b8f3b2a2545) to cleanup all docker related rules that have been created on my system.

@dl12345 does this solve your issue?

dl12345, 2020-06-24:
Yeah, that should work. Definitely better than doing a `fw3 restart`

G-M0N3Y-2503, 2020-06-25:
FYI it looks like [libnetwork](https://github.com/moby/libnetwork/blob/master/drivers/bridge/setup_ip_tables.go) is the one responsible for modifying the firewall, Though I'm not familiar with `docker-proxy`s interactions with `dockerd` so I'm not sure where a binary fix would go.

feckert, 2020-06-26:
I think we should start by getting the firewall rules out on service stop first (https://github.com/openwrt/packages/pull/12556)

We could set the iptables key to false in the Docker engine’s configuration file at /etc/docker/daemon.json.
And do everything ourselves with the fw3?

G-M0N3Y-2503, 2020-06-27:
> 1. We could set the iptables key to false in the Docker engine’s configuration file at /etc/docker/daemon.json.
>    And do everything ourselves with the fw3?

To me, this seems like a lot of ongoing maintenance, and I'm not sure what benefits we would gain by doing this...
I imagine most people will either have `dockerd` installed and running, in which case the inserted rules are being used, or just not have it installed. I don't think we'd have many users with `dockerd` installed run once and then not have it running for the majority of the time, and even if we do I think https://github.com/openwrt/packages/pull/12556 is enough.

> 2\. Another solution is to patch/update the docker-ce go program to delete the firewall rules on binary stop.
>     But I don't think I'm gonna be able to do it that fast.

I think this is the ideal solution, figuring out how we can `deinit` the [init](https://github.com/moby/libnetwork/blob/bump_19.03/drivers/bridge/bridge.go#L159) too, but I'm not all that familiar with golang either.

feckert, 2020-07-13:
Fixes with pull request https://github.com/openwrt/packages/pull/12556

closed 2020-07-13 by feckert