openwrt/packages #12364: Is libfuse missing?

qiuzi, 2020-05-30:
`WARNING: Makefile 'package/feeds/packages/davfs2/Makefile' has a dependency on 'libfuse', which does not exist
WARNING: Makefile 'package/feeds/packages/gnunet-fuse/Makefile' has a dependency on 'libfuse', which does not exist
WARNING: Makefile 'package/feeds/packages/ntfs-3g/Makefile' has a dependency on 'libfuse', which does not exist
WARNING: Makefile 'package/feeds/packages/owfs/Makefile' has a dependency on 'libfuse', which does not exist
WARNING: Makefile 'package/feeds/packages/owfs/Makefile' has a dependency on 'fuse-utils', which does not exist
WARNING: Makefile 'package/feeds/packages/seafile-server/Makefile' has a dependency on 'libfuse', which does not exist
WARNING: Makefile 'package/feeds/packages/sshfs/Makefile' has a dependency on 'libfuse', which does not exist
WARNING: Makefile 'package/feeds/packages/sshfs/Makefile' has a dependency on 'fuse-utils', which does not exist`

lucize, 2020-05-30:
manual add the link from feeds where it was moved, it seems sometimes the link will not be added, maybe because it was present in the base repo ?

neheb, 2020-05-31:
./scripts/feeds/update -a -f

neheb, 2020-05-31:
My goal was to update sshfs to the latest version but it seems that it requires udev...

lucize, 2020-05-31:
https://git.alpinelinux.org/aports/tree/main/sshfs/APKBUILD doesn't list udev, the libudev-fbsd  is not ok ?

neheb, 2020-05-31:
Yeah but if you look at the fuse3 package in alpine, it uses eudev.

The problem is that the build system uses pkgconfig to find udev. libudev-fbsd does not contain a pkgconfig file.

neheb, 2020-06-06:
Clear your tmp directory and try again.

closed 2020-06-06 by neheb