openwrt/packages #12365: OCSERV - MFA config

Gustavo-FS, 2020-05-30:
Hello everyone

I'm trying to enable MFA with Google Authentication on my openvpn server but not work
I found several references on the subject but they are very old

https://forum.archive.openwrt.org/viewtopic.php?id=38905
https://github.com/openconnect/recipes/blob/master/ocserv-2fa.md#OATH1

can someone help me with a Configuration How-TO?
in all the configurations that I tried to make the ocserv does not go up when changing the method to PAM I need some additional package installed?

brada4, 2020-05-31:
Could you describe the steps you made? As much as I checked installing ocserv and libpam and changing ocserv authenticator to pam still permits it to start. So you did something on top of the guides you linked that broke the stuff.

Gustavo-FS, 2020-06-02:
well, i tried with de 2 orientation in link's create ocserv in /etc/pam.d and modify for google authentication or OATH no one works
did you can help me to configure?

nmav, 2020-09-22:
`auth = "plain, pam"` is simply wrong configuration. Check the upstream site on how to enable pam:
https://ocserv.gitlab.io/www/recipes-ocserv-authentication-pam.html

closed 2020-09-23 by nmav