openwrt/packages #12366: [Request] add easylist FR list to default package

OhSoGood, 2020-05-31:
Hi

I have been manually adding since several versions the French version of Easylist, similar to all existing ones in openwrt's adblock:

> config source 'reg_fr'
> 	option enabled '1'
> 	option adb_src 'https://easylist-downloads.adblockplus.org/liste_fr.txt'
> 	option adb_src_rset 'BEGIN{FS=\"[|^]\"}\$0~/^\|\|([[:alnum:]_-]+\.){1,}[[:alpha:]]+\^("\\\$third-party")?$/{print tolower(\$3)}'
> 	option adb_src_desc 'focus on French ads plus generic easylist additions'
> 

With the latest version of adblock, it's less easier to add them manually (/etc/config/adblock has changed). Nervertheless, could it be added to the default package ?

Thank you very much for your time and effort on adblock for openwrt !

dibdot, 2020-05-31:
No problem, will be included with the next update

closed 2020-05-31 by dibdot

dibdot, 2020-05-31:
Included in https://github.com/openwrt/packages/commit/ab44d8146af2abcfd96f08aba99358e960049bd3