openwrt/packages #12367: conflict between CLI and LUCI about adblock sources

OhSoGood, 2020-05-31:
On adblock 4.0.5 on Openwrt 19.7.3, the command line and LUCI can be in conflict:

1. If I add a list via /etc/init.d/adblock list add, it doesn't appear in Luci Adblock. 
2. If later I change something on luci adblock, the manually added listed gets removed.

Cheers

dibdot, 2020-05-31:
When you change something on the CLI (e.g. add/remove lists) then you have to manually reload the LuCI app (e.g. click on the Overview tab to get the latest uci config) ... there is no automatism for that.

closed 2020-05-31 by dibdot

dibdot, 2020-05-31:
https://github.com/openwrt/packages/commit/ab44d8146af2abcfd96f08aba99358e960049bd3 should behave better ... anyway you need a page reload, e.g. press "Overview" after CLI actions ...