openwrt/packages #12377: irqbalance: Daemon couldn't be bound to the file-based socket

nbk15lk, 2020-06-01:
The start of irqbalance throws an error:  "/usr/sbin/irqbalance:  Daemon couldn't be bound to the file-based socket.

strace shows the program is trying to create a socket in /run/irqbalance which does not exist:

Package:  irqbalance - 1.6.0-7
OpenWRT version:  	OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.150.62258-217d331
Hardware:  	Linksys WRT32X

```
# strace /usr/sbin/irqbalance -f -t 10
...
getpid()                                = 13522
bind(5, {sa_family=AF_UNIX, sun_path="/run/irqbalance//irqbalance13522.sock"}, 110) = -1 ENOENT (No such file or directory)
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 6
connect(6, {sa_family=AF_UNIX, sun_path="/dev/log"}, 12) = 0

```

nbk15lk, 2020-06-01:
Defined on line 165 here:  https://github.com/Irqbalance/irqbalance/blob/master/irqbalance.h

neheb, 2020-06-06:
No. patch out the path from irqbalance.

neheb, 2020-06-16:
ping @hnyman

hnyman, 2020-06-16:
Sounds like something that should be taken care by the main OpenWrt devs like @jow- or @adrianschmutzler in the base-files, so that we would create also /run (and symlink that to /var/run, or vice versa)

Looks like irqbalance has been modified to follow the current Linux standard ( /run being the main dir and /var/run a symlink), while we are still following the old way.

Irqbalance people have discussed their approach and decided to use /run:
* https://github.com/Irqbalance/irqbalance/commit/f5ca2eb481ac76b799c8efef5a7b7469bb368fd5
* https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=915834
* https://github.com/Irqbalance/irqbalance/issues/85

Current Linux standard:

https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard
> In FHS 3.0, /var/run is replaced by /run; a system should either continue to provide a /var/run directory, or provide a symbolic link from /var/run to /run, for backwards compatibility

We should likely add a symlink from /run to /var/run here:
 https://github.com/openwrt/openwrt/blob/master/package/base-files/files/etc/init.d/boot#L25

I guess that some other packages will start using that /run  sooner or later.

neheb, 2020-06-16:
> Sounds like something that should be taken care by the main OpenWrt devs like @jow- or @adrianschmutzler in the base-files, so that we would create also /run (and symlink that to /var/run, or vice versa)
> 
> Looks like irqbalance has been modified to follow the current Linux standard ( /run being the main dir and /var/run a symlink), while we are still following the old way.
> 
> Irqbalance people have discussed their approach and decided to use /run:
> 
>     * [Irqbalance/irqbalance@f5ca2eb](https://github.com/Irqbalance/irqbalance/commit/f5ca2eb481ac76b799c8efef5a7b7469bb368fd5)
> 
>     * https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=915834
> 
>     * [Irqbalance/irqbalance#85](https://github.com/Irqbalance/irqbalance/issues/85)
> 
> 
> Current Linux standard:
> 
> https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard
> 
> > In FHS 3.0, /var/run is replaced by /run; a system should either continue to provide a /var/run directory, or provide a symbolic link from /var/run to /run, for backwards compatibility
> 
> We should likely add a symlink from /run to /var/run here:
> https://github.com/openwrt/openwrt/blob/master/package/base-files/files/etc/init.d/boot#L25
> 
> I guess that some other packages will start using that /run sooner or later.

I was thinking that irqbalance should be patched for now.

hnyman, 2020-06-17:
Apparently it would not be enough to create /run, as the /run/irqbalance dir is assumed to exist by .h and .c, but is actually created by a systemd init script that we do not use.

Fixed by 4f0c847828

> Irqbalance defines /run/irqbalance dir for its socket communication between irqbalance and its UI. /run does not exist in OpenWrt (although it is defined by the Linux FHS), so the socket creation fails. Although we do not compile UI and thus the issue is not critical to us, fix the directory location.
>
> Additionally, the creation is originally handled by a systemd init script that we do not use.
>
> * patch source to define dir as /var/run/irqbalance
> * create the dir in the procd init script.

closed 2020-06-17 by hnyman