openwrt/packages #12379: linuxptp: doesn't compile with 19.07.3

vercas, 2020-06-02:
Maintainer: Wojciech Dubowik <Wojciech.Dubowik@neratec.com> (maybe @wodu I think?)
Environment: x86_64 generic 19.07.3

Description:

Package does not compile:
```
make[3]: Entering directory '/path/to/openwrt/feeds/packages/net/linuxptp'
rm -f /path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0/.built
touch /path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0:linuxptp-2.0 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC -I/path/to/openwrt/staging_dir/target-x86_64_musl/usr/include -I/path/to/openwrt/staging_dir/target-x86_64_musl/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0:linuxptp-2.0 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC -I/path/to/openwrt/staging_dir/target-x86_64_musl/usr/include -I/path/to/openwrt/staging_dir/target-x86_64_musl/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include " LDFLAGS="-L/path/to/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/path/to/openwrt/staging_dir/target-x86_64_musl/lib -L/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -fpic -specs=/path/to/openwrt/include/hardened-ld-pie.specs -znow -zrelro " EXTRA_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0:linuxptp-2.0 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/path/to/openwrt/staging_dir/target-x86_64_musl/usr/include -I/path/to/openwrt/staging_dir/target-x86_64_musl/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC" make -j1 -C /path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0:linuxptp-2.0 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0'
x86_64-openwrt-linux-musl-gcc -Wall -DVER=2.0  -D_GNU_SOURCE -DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC -DHAVE_ONESTEP_P2P  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/path/to/openwrt/build_dir/target-x86_64_musl/linuxptp-2.0:linuxptp-2.0 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/path/to/openwrt/staging_dir/target-x86_64_musl/usr/include -I/path/to/openwrt/staging_dir/target-x86_64_musl/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/path/to/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -DHAVE_CLOCK_ADJTIME -DHAVE_POSIX_SPAWN -DHAVE_ONESTEP_SYNC   -c -o sk.o sk.c
sk.c: In function 'sk_timestamping_init':
sk.c:436:14: error: 'HWTSTAMP_TX_ONESTEP_P2P' undeclared (first use in this function); did you mean 'HWTSTAMP_TX_ONESTEP_SYNC'?
    tx_type = HWTSTAMP_TX_ONESTEP_P2P;
              ^~~~~~~~~~~~~~~~~~~~~~~
              HWTSTAMP_TX_ONESTEP_SYNC
sk.c:436:14: note: each undeclared identifier is reported only once for each function it appears in
<builtin>: recipe for target 'sk.o' failed
make[4]: *** [sk.o] Error 1
```

wodu, 2020-06-04:
We would most likely need a fix from https://github.com/openwrt/packages/pull/12298 as it's build system dependent i.e. linuxptp takes some include files from a host instead of a target. Can you try it out as I can't reproduce it on my host (mint 19.3 with kernel 4.15).

neheb, 2020-06-06:
I backported it.

neheb, 2020-06-16:
Closing. I backported it. The buildbot also builds this fine.

closed 2020-06-16 by neheb