openwrt/packages #12380: docker-ce:  missing kernel configuration symbol

dl12345, 2020-06-02:
On startup

```
WARN[2020-06-02T19:13:22.619792818+01:00] Your kernel does not support cgroup blkio weight 
WARN[2020-06-02T19:13:22.619998687+01:00] Your kernel does not support cgroup blkio weight_device
```

After which my entire system hangs and reboots, although to be fair, I'm not sure it's this that's causing the reboot. I'll update once I've compiled a kernel with the relevant symbols enabled. 

The symbol that needs to be enabled is `CONFIG_BLK_CGROUP_IOCOST=y`

```
CONFIG_BLK_CGROUP_IOCOST:                                                                                                                                                                                                
  Enabling this option enables the .weight interface for 
  cost  model based proportional IO control.  The IO controller
  distributes IO capacity between different groups based on
  their share of the overall weight distribution.
```

```
diff --git a/utils/docker-ce/Config.in b/utils/docker-ce/Config.in
index dba32c4..d9184d0 100644
--- a/utils/docker-ce/Config.in
+++ b/utils/docker-ce/Config.in
@@ -31,6 +31,7 @@ menu "Kernel features for Docker"
                select KERNEL_MEMCG_SWAP
                select KERNEL_MEMCG_SWAP_ENABLED
                select KERNEL_BLK_DEV_THROTTLING
+               select KERNEL_BLK_CGROUP_IOCOST
                select KERNEL_CFQ_GROUP_IOSCHED
                select KERNEL_CGROUP_PERF
                select KERNEL_FAIR_GROUP_SCHED
@@ -83,3 +84,7 @@ menu "Kernel features for Docker"
                        select KERNEL_BTRFS_FS_POSIX_ACL
        endmenu
 endmenu
+
+config KERNEL_BLK_CGROUP_IOCOST
+       bool
+       default n
```

Platform is x86_64

@G-M0N3Y-2503

G-M0N3Y-2503, 2020-06-04:
You should be able to run docker-ce even with this warning.

G-M0N3Y-2503, 2020-06-27:
For this issue, I think we'd be better off adding something more similar to the [Kernel definition of `BLK_CGROUP_IOCOST`](https://github.com/torvalds/linux/blob/v5.4/block/Kconfig#L138) to the [OpenWrt `/config/Config-kernel.in`](https://github.com/openwrt/openwrt/blob/master/config/Config-kernel.in#L743) then selecting it in `docker-ce/Config.in` like above.

dl12345, 2020-06-28:
> For this issue, I think we'd be better off adding something more similar to the [Kernel definition of `BLK_CGROUP_IOCOST`](https://github.com/torvalds/linux/blob/v5.4/block/Kconfig#L138) to the [...]

Indeed, although this then assumes a change to the main Openwrt repo in order to accommodate a feeds-based package. Tidier in the absolute, but there may or may not be a policy regarding that.

G-M0N3Y-2503, 2020-06-28:
> but there may or may not be a policy regarding that

There may be, but [it's not unprecedented to have kernel options for a specific package](https://github.com/openwrt/openwrt/commit/4f94a331e172fd23a21b74d3e552abad25248326#diff-a9af7e4a79e28e1a7b2e85d7e8150bf8), so it's not unlikely to be accepted one way or another.